sinara-hw/sinara #496: RTM 1V2 and 2V0 failure

sbourdeauducq, 2018-01-30:
On one RTM I have, the P1V2 and P2V0 LEDs are off. This persists after power cycles. This is all the information I have for now.

gkasprow, 2018-01-30:
please make sure that the transistor that is on the board edge, close to LEDs, that switches these LEDs on is not broken mechanically. I managed to break it by inserting the board to the chassis.
It is simply too close to the board edge.

sbourdeauducq, 2018-01-30:
Yes. T7 is broken on that board.

closed 2018-01-30 by hartytp

jbqubit, 2018-01-31:
So the power hasn't actually failed in this case, right? Its just the transistor for the LEDs? @gkasprow did you update layout?

gkasprow, 2018-01-31:
It is on todo list.

jbqubit, 2018-01-31:
Just looked at my RTMs too. Those LEDs are off and transistor is broken. Glad to hear this is easy to fix in Sayma 2.