sinara-hw/sinara #497: Urukul LDO PSRR

hartytp, 2018-01-30:
@gkasprow Looking at figure 11 of the TPS74901 datasheet, the PSRR of regulators IC7, IC11 and IC18 is very low with a dropout voltage of 0.2V. Can we change the outputs of IC15 to give an extra couple of hundred mV headroom on these LDOs?

jordens, 2018-01-30:
What are you after? The 1 MHz of the SMPS were already down 85 dBc and comparable to the DDS spurs. And at 1 MHz the LDO PSRR doesn't change much with dropout. Give it another 100 mV. But 300 mV would add another 10% in power consumption.

hartytp, 2018-01-30:
More worried about general broadband crud coming through from the SMPS.

We're planning to do a careful AM/PM noise measurement, hopefully tomorrow. From that, I'll know if this is actually needed...

jordens, 2018-01-31:
Except for the 1khz Spurs you'll measure the phase noise that I measured already.

hartytp, 2018-02-02:
With the PLL on, this won't help. Not sure about the noise when the DDS is clocked directly from a low noise 1GHz oscillator.

In any case, let's not do this until we have evidence that it will help.

closed 2018-02-02 by hartytp