sinara-hw/sinara #502: Sayma_RTM 3.3V regulator at Oxford

jbqubit, 2018-01-31:
From #472 @hartytp said
> I have two RTMs atm and one is not usable since one of the 3V3 regs doesn't seem to work, breaking the clock path. I haven't investigated why or tried replacing components yet. 

Would be good to know if this is an isolated issue or not.

hartytp, 2018-01-31:
I need to send Greg some stuff anyway, so I'll post the RTM to him at the same time for him to inspect. Probably, it's just a bad solder joint or something and not an issue with the design (it's pretty hard to see what else could cause this reg to fail). Once the cause of this issue is determined we'll know more.

hartytp, 2018-02-26:
@gkasprow Now has the RTM with this issue, so should be able to diagnose what went wrong.

gkasprow, 2018-02-26:
@hartytp I still didn't get your board...

hartytp, 2018-02-26:
@gkasprow Really, that's odd. I posted that at the same time as the RF PA chassis, which you did get.

Is it worth me getting the tracking number (UPS IIRC)? Or, is this another case of "lost in the polish postal system"? That's a lot of money on Saymas missing atm.

gkasprow, 2018-02-26:
@hartytp please give me the tracking number. At least I would be able to find who received the package :) It usually lands somewhere in administration department.

hartytp, 2018-02-26:
Will do!

hartytp, 2018-02-27:
@gkasprow I hope you didn't bin the packaging for the RF PA I sent you -- the RTM was in an anti-static bag in the same box!

gkasprow, 2018-02-28:
Oh. This explains a lot! :)

jbqubit, 2018-02-28:
It's OK if the PCB got lost-by-trashing but just want to be sure that I understand. Am I missing a --wink-wink-- emoji here?

gkasprow, 2018-02-28:
Nothing was lost. The package is in my lab but I was simply expecting
another one. I passed the box with RFPA to my coleague who extracted the
amplifier only and didnt pay attention what was left.I will check it
tomorrow once I get back from Copenhagen

gkasprow, 2018-03-05:
@hartytp do you remember which regulator was causing the clock path problem?

hartytp, 2018-03-06:
It was the LDO that powered the LTC sine-square converter by the RTM front-panel 100MHz SMA input (can't remember PN off the top of my head). You'll see that I removed the ferrite bead that connects the LDO to the clock IC during debugging.

gkasprow, 2018-03-06:
It was bad solder caused by poor wetting. Repaired. I also checked other LDOs and they are fine.

closed 2018-03-06 by gkasprow

hartytp, 2018-03-06:
Thanks Greg! I suspected something like that, but didn't have time to look into it.

In any case, we only paid for one Sayma RTM, so that one wasn't meant for us anyway.