sinara-hw/sinara #503: DIO SMA 1.2

gkasprow, 2018-02-01:
- [ ] reverse DIP switch channels to match front panel channel assignment

jordens, 2018-03-12:
The bottom SMA (channel 7) interferes with the handle. To use it that channel, the handle needs to be removed.

gkasprow, 2018-03-15:
hm, I connected the SMA a few times and didn't notice that. Can you take a pic ?

jordens, 2018-03-15:
Will do.

jordens, 2018-03-22:
[image]

gkasprow, 2018-03-22:
maybe on my board the SMA connector was assembled at slightly different angle or I have smaller SMA nut :)

gkasprow, 2018-03-22:
@jordens do you observe this issue on all boards you have?
Maybe some cut-out in the handle would do the job? Not very elegant but if we make the connectors placed denser, we limit access space to them.

jordens, 2018-03-22:
Yeah. I can wiggle it and file it off. It's on two out of three and the connector where the connector is at a slight angle. Could also be viewed as a QC issue.
8 SMA on 4 HP is too dense IMHO (but that's old news).

marmeladapk, 2018-06-13:
_Migrated to sinara-hw/DIO_SMA#1_

closed 2018-06-13 by marmeladapk