sinara-hw/sinara #504: Urukul v1.1 erata

hartytp, 2018-02-02:
- Consider adding LC filters between the SMPSs and the low noise analog voltage rails.

gkasprow, 2018-02-02:
there already LC filters at the output of the SMPS.
There is 1.5uH + 47uF ceramic and 22uF tantalium. I can add yet another one just at the LDO.

hartytp, 2018-02-02:
Aren't those inductors just the inductors required in the SMPS? Elsewhere, we've added a second inductor to help further reduce the ripple.

gkasprow, 2018-02-02:
no, I used low EMI DC/DC converter with integrated inductors. These are low pass filters.

hartytp, 2018-02-02:
Aaah, I missed that, sorry!

Okay, leave it as is. The noise is already good enough.

gkasprow, 2018-02-02:
they are placed close to the SMPS to limit noise propagation. But the inductors may still pickup some noise from SMPS. We can add some inductor  on existing Urukul close to the LDO to check if this helps.

hartytp, 2018-02-03:
Up to you Greg. The current situation is probably fine, but if the inductors are cheap and there is space for them then feel free to add them. In any case, basedon the data we have, this is a low priority.

hartytp, 2018-02-16:
@WeiDaZhang has finished looking at the Urukul AD9910 loop filters. Turns out that the reason that the performance didn't match the simulations is that there is 1k resistor and 1.1pF capacitor on-chip, which was lowering the phase margin. These aren't mentioned in the datasheet, but are taken into account in the PLL design spreadsheet ADI provide.

Optimised loop filter components for the AD9910 variant of Urukul are:
C112 - DNP
R90 - 3.3k
R92 - 0Ohm
C99 - 68pF
C72 - DNP

@gkasprow Where are we with the Urukul manufacturing process? Have the boards been populated yet? If not, can we sneak this change into v1.1? If that's a problem, let's include this into v1.2.

hartytp, 2018-02-16:
Measured PM and AM spectra using that loop filter...

[Urukul Phase Noise and Amplitude Modulation of 4dBm Output with Wenzel Referenced.pdf](https://github.com/m-labs/sinara/files/1733099/Urukul.Phase.Noise.and.Amplitude.Modulation.of.4dBm.Output.with.Wenzel.Referenced.pdf)

gkasprow, 2018-02-19:
@hartytp the Urukul boards are already assembled and are waiting for tests.

hartytp, 2018-02-19:
No problem, let's make this change next time we print a fresh batch of PCBs.

gkasprow, 2018-02-20:
@hartytp did you test these values with hardware?

hartytp, 2018-02-20:
@gkasprow yes, see the plot I posted. That was a measurement @WeiDaZhang made on one of our Urukul v1.0 boards using the new loop filter.

gkasprow, 2018-02-20:
OK, so let's make new issue for Kasli 1.3

hartytp, 2018-03-06:
@gkasprow pointed out in an email to me that the SMPS used in Urukul actually has integrated inductors, so the external inductors on the schematic are already a second stage of filtering. As there are no obvious performance issues in the current design related to SMPS spurs on the power lines, there is no good reason to change the design and add more inductors. So, I've removed that from the ToDo list at the top of this issue.

hartytp, 2018-03-07:
@gkasprow once the current batch of Urukul is sold, can we make sure that the next batch produced is 
3,  including these changes? Thanks!

gkasprow, 2018-03-07:
Sure.

closed 2018-03-28 by gkasprow