sinara-hw/sinara #507: Compiling MMC code at M-Labs

jbqubit, 2018-02-05:
As part of developing for Sayma_AMC [WUT is running ARTIQ](https://github.com/m-labs/sinara/issues/468), tweaking Migen/ARTIQ source and using the ARTIQ build toolchain. Likewise, M-Labs ought to be able to edit WUT's MMC source and use the NXP toolchain to generate binaries as part of debugging. The back-and-forth of MMC binaries is slowing M-Labs's ability to tinker and debug. @sbourdeauducq I know you prefer full control over a device under test. Please assert control over MMC to debug Ethernet (https://github.com/m-labs/artiq/issues/854).

gkasprow, 2018-02-05:
Hold on a while. Don't dig in my messy temporary code. We will publish full support for OpenMMC in a few days. There are just a few minor issues to solve...

jbqubit, 2018-02-06:
@gkasprow There's not time to wait for code beautification. Footprint of MAX23287-related code in MMC source is small. Aim is to get M-Labs empowered to debug Ethernet not rewrite your MMC firmware.

sbourdeauducq, 2018-02-06:
No. The code is difficult and a waste of time to compile, in addition to being a mess. OpenMMC compilation is more reasonable.

closed 2018-05-03 by hartytp