sinara-hw/sinara #510: Sinara collaborator permissions revoked for @jbqubit

jbqubit, 2018-02-06:
My github [collaborator permissions](https://help.github.com/articles/permission-levels-for-a-user-account-repository/#collaborator-access-on-a-repository-owned-by-a-user-account) were revoked last night for the Sinara repository. I can no longer commit to the repository nor assign or label Issues. Presently M-Labs (@jordens, @sbourdeauducq) has administrator privileges for the repository so I presume they made the change. If this is an error please revert.

marmeladapk, 2018-02-11:
@jbqubit I see that you have collaborator permissions now so I'll allow myself to close this issue.

jbqubit, 2018-02-14:
No, I don't have these permissions. I still can't assign Issues or add Issues to milestones. Please reopen.

marmeladapk, 2018-02-14:
Weird. Perhaps you have read only rights?

marmeladapk, 2018-02-20:
[image]

Perhaps you have read only rights?

jbqubit, 2018-02-27:
@sbourdeauducq @jordens Please stop this passive aggressive behavior. I can't edit wiki either.

jordens, 2018-02-27:
@jbqubit you'll need to talk to @sbourdeauducq

jbqubit, 2018-02-27:
@sbourdeauducq What gives?

hartytp, 2018-02-27:
I don't understand what's going on here, but I'm sick of wasting time on this kind of thing. Joe has been a main driver and funder of this work and it's not okay to do this kind of thing. Would it be easier if we moved all hw work not funded by m-labs to a different project?

jbqubit, 2018-02-27:
Thanks @hartytp. 

@sbourdeauducq decided to revoke my credentials the same day as I posted [this](https://github.com/m-labs/sinara/issues/507) Issue. I gather he found either the content or my choice to Assign [it](https://github.com/m-labs/sinara/issues/507) to him objectionable. Who knows? I didn't get an explanation.

hartytp, 2018-02-27:
Whatever the complaint I can't see this as being a professional way of handling it. If this project is going to be run in a childish fashion we can move our part of it elsewhere to more neutral teritory.

dhslichter, 2018-02-27:
Whatever has happened, @jbqubit clearly needs to have full collaborator permissions as he is funding the bulk of the work.  Disagreements over planning, frustration with actions taken by others, etc, need to be resolved by discussion (painful though it might be), and not by removing someone else's permissions.

dhslichter, 2018-02-27:
Someone looking to fund future development might look at this kind of thing and feel that they might commit money and then be locked out -- not an enticing prospect, and not a good advertisement.

jordens, 2018-02-27:
AFAICT the statements about platinum sponsor levels are neither correct nor at all helpful as a measure to assign access levels. This should be flat and open.
OTOH there is a lot askew that needs to change really soon and the time is ripe for writing down some etiquette, community, and contribution guidelines. Anyone want to volunteer?

hartytp, 2018-02-27:
>  This should be flat and open.

Yes.

> OTOH there is a lot askew that needs to change really soon and the time is ripe for writing down some etiquette, community, and contribution guidelines. Anyone want to volunteer?

In general, that's probably a good idea. But, let's not pretend that an etiquette manual is required to make it perfectly clear that this particular piece of behaviour is unhelpful and unacceptable.

> AFAICT the statements about platinum sponsor levels are neither correct nor at all helpful as a measure to assign access levels.

Possibly true. But, the level of frustration for a few of us is getting to the point where a fork might make sense.

dhslichter, 2018-02-27:
> AFAICT the statements about platinum sponsor levels are neither correct nor at all helpful as a measure to assign access levels. This should be flat and open.

Sure.  What I am saying is that people funding the work (and very involved in its development as well) are clearly an integral part of the team and as such should not have privileges summarily revoked, period.

jbqubit, 2018-02-27:
@jordens said
> This should be flat and open.

You've had 21 (93) days to revert sinara (artiq) permissions changes made by @sbourdeauducq or to speak up in support of "flat and open." At this juncture your words ring hollow. 

>  But, the level of frustration for a few of us is getting to the point where a fork might make sense.

A down side of a fork is that Issues, Wiki and other important metadata is lost. I make [backups](https://help.github.com/articles/backing-up-a-repository/) on occasion for contract purposes using [github-records-archiver](https://github.com/benbalter/github-records-archiver) which contain most metadata (excluding photos) but there's no automated restore-to-github. The highest degree of continuity would be provided by a [repository transfer](https://help.github.com/articles/about-repository-transfers/) out of M-Labs and into a different github organization.

jordens, 2018-02-27:
@hartytp That's not at all as clear and absolute as you make it sound. There needs to be a way to escalate and react if Joe's behavior is deemed wrong. (I do think the latter is the case but also disagree with the type of reaction.)
Forking a hardware project (as opposed to software) is a mere technicality to me. I don't get how that solves the governance issue. Is it more than a knee-jerk reaction?

@dhslichter I agree. But there was a lot more going on here than what you perceive and paint as privileges being summarily revoked. And no need to "period" me. That's a dull shot at preempting a response.

While everybody is allowed to vent freely: I really hope the constant battle of "summarizing" issues from somewhere else to stir it up, spraying assignees, priorities and milestones bursting of entitlement, filing issues in multiple copies here and there, and the attitude of carelessly expecting that everything goes because others will clean up ends soon. From where I stand, that makes this unmanageable and any attempt to improve meaningless.

jordens, 2018-02-27:
@jbqubit Don't use me to get to @sbourdeauducq You are old enough to not have to resort to publicly lobbying against individuals. Sort it out with him.
Your behavior on artiq was immediately destructive as well as directly ignorant and ridiculing the agreement. There is no way you dare bring that up here.

jordens, 2018-02-27:
And you were even so insolent as to make fun of what you did. Don't expect my support.

jbqubit, 2018-02-28:
I try to consistently use the strings "Sayma_AMC" and "Sayma_RTM" to make it easier to search for Issues involving the different PCBs. [This](https://github.com/m-labs/artiq/issues/847#issuecomment-346852339) was the edit that got me booted from the ARTIQ repository: 

[> @jbqubit jbqubit changed the title from Sayma won't boot from flash
to Sayma_AMC: booting from flash on Nov 24,
2017](https://github.com/m-labs/artiq/issues/847#issuecomment-346852339)

> @jbqubit Don't use me to get to @sbourdeauducq You are old enough to not have to resort to publicly lobbying against individuals. Sort it out with him.

My contracts are not with @sbourdeauducq or @jordens they're with M-Labs Ltd. My concern was addressed to M-Labs (see above). I don't care who made the call, it reflects poorly on the company. It is alarming that censure is M-labs's response to conflict.  

> And you were even so insolent as to make fun of what you did. 

I don't recall being insolent or even funny, which seems like a missed opportunity. 

The sinara Issue system is a shared resource. Please restore sinara access so I can make edits to the wiki, etc. and we can move forward. Let's return to these matters of governance for Sinara and debate how to use the Issue system once Sayma/EEMs are done. Right now this topic is just a distraction.

sbourdeauducq, 2018-02-28:
> I try to consistently use the strings "Sayma_AMC" and "Sayma_RTM" to make it easier to search for Issues involving the different PCBs. This was the edit that got me booted from the ARTIQ repository:

No. There were warnings before, plus an explanation when that happened.

hartytp, 2018-02-28:
> And you were even so insolent as to make fun of what you did. Don't expect my support.

What the hell is this? "Insolent"? Do you think you're a parent scolding a naughty client? Whatever your complaint, that's neither an acceptable or helpful way of addressing it.

Whatever you think Joe might have done, that kind of language puts you in the wrong in my books.

hartytp, 2018-02-28:
> Forking a hardware project (as opposed to software) is a mere technicality to me. I don't get how that solves the governance issue. Is it more than a knee-jerk reaction?

Not a knee-jerk reaction; an accumulation of frustrations to the point where I feel that change is needed. When Joe first posted this, I just rolled my eyes and decided not to get involved. The fact that this is *still* an issue is immensely frustrating.

For my mind, forking solves a few things:
1. Unrelated, I want to burn the repository anyway, since it's grown so bloated with binaries that it's a pain to work with. In principle, it would be better to sort through it and tidy up the history, but that's a lot of work and I don't think it's realistic to expect anyone to do that. In any case, I'm not sure how valuable keeping the history is for PCB designs; unlike code they're usually in binary format anyway so one can't do diffs and merges. I'd like to start a new project and copy all current designs across and just leave the history here.
2. While this is an M-Labs project they have administration rights. This issue has made me fundamentally uncomfortable with that. I don't want to contribute to and fund work and then find people being blocked in this way. Moving to a non-M-Labs project gives us a chance to change the allocation of privileges to users.
3. Call it "symbolic" if you want, but it's more than that as far as I'm concerned: having this as an M-Labs project gives the impression that they're leading it. I'm uncomfortable with that for multiple reasons. Apart from the above, the continued trash-talking of Sayma falls under this category. In my view, Sayma is still *the* flagship part of this hardware project -- when I talk to prospective users it's Sayma and shuttler which they cite as the deal-breakers that would make them switch to it. I do not want the organisation hosting this work to continually bad mouth Sayma when, as far as I can tell, the actual hardware issues discovered with it so far have been quite minor.

jordens, 2018-02-28:
@hartytp You are not in a position to judge whether that language is appropriate or not. You have no idea what Joe did and said.
Being accused of censorship is absurd. Joe doesn't seem to know what censorship is.
And being accused of rudeness, condescending behavior by you in the light of the constant abuse that we are facing by Joe is too much.

Speaking for @sbourdeauducq and me now: We've had it. After years of advice and warnings without any adequate response or remedy, we need to see the following corrected immediately:

* The outrageous claim of censorship. You don't seem to have the slightest idea what censorship is.
* The misrepresentation of and abuse of our agreement regarding the experiment of having you do management activities on artiq.
* The opportunistic choosing and switching of roles. On a given topic you engage with the grandiose laissez-faire attitude of a benefactor where you boast to have been responsible for support, funding, and enabling the projects that others are leading. Then when it suits you you put on the demanding and entitled attitude of a customer expecting others to execute on contracts and agreements and clean up your carelessly written ideas and uninformed and unqualified requirements. And then again you draw from the entitlement and use that as leverage permitting you to bulldoze over everything interfering and sabotaging the work of others. Those roles need to be separate. You can't mix and match what helps you most and seems convenient in any given instant.

jordens, 2018-02-28:
I also rolled my eyes and decided to not get involved. See where that got us.

hartytp, 2018-02-28:
> @jbqubit Don't use me to get to @sbourdeauducq You are old enough to not have to resort to publicly lobbying against individuals. Sort it out with him.

I agree with Joe on this. We have no contracts with you as an individual. All our contracts are with M-Labs so we expect to treat you as an organisation. I don't see that as being in any way unreasonable or unexpected. If you want to be treated as an individual then setup as a sole contractor and take contracts on that basis.

> Your behavior on artiq was immediately destructive as well as directly ignorant and ridiculing the agreement. There is no way you dare bring that up here.

Firstly, I cannot get behind the use of the word "dare" here. Again, do you think you're talking to a child? Actually, no I really hope you don't speak to children that way. Whatever Joe has done, you're unlikely to get much support from anyone else until you drop that kind of language.

sbourdeauducq, 2018-02-28:
@jbqubit I have restored your access, however, we would be *really* appreciative if you could use the issue system sensibly instead of barging into everything, assigning people in entitled ways, moving other peoples' comments around, making a mess with the labels, copy-pasting emails in inappropriate places, etc.

As for "blocking people": we currently have no plans to restrict write access to others, who (despite the occasional beef) have generally been reasonable and made a lot of positive contributions, especially you @hartytp.

Note that restricting write access is different from "censorship": Joe could still post issues, comments etc.

In the ARTIQ repository, only developers (not funders) have write access, and this has been working fine so far AFAICT. Joe had it briefly because he wanted his own dashboard, and we told him that his write access was only for that and "trivial changes" such as fixing typos, but he contined to overstep the limit despite warnings, which led us to revoke it.

hartytp, 2018-02-28:
> Your behavior on artiq was immediately destructive as well as directly ignorant and ridiculing the agreement. 

Finally, whatever has gone on between you all, it's worth remembering that all that Joe has done for this project. He got the ball rolling and galvanized people to take part. I/Oxford wouldn't be involved if it weren't for him. Would Greg? He also found the bulk of the funding for this.

The level that Sinara has come to is far beyond the original M-Labs white paper. Even look at the version of Kasli put out in the white paper compared with what we have. There was no idea of EEMs in the white paper or the IDCs, it was Joe and Greg who pushed for that and the project is much better of because of it.

So, while you may find his approach frustrating, I don't understand this level of vitriol. I just don't.

jordens, 2018-02-28:
With "dare" I meant to say that it will backfire and hurt his cause instead of regaining the trust that was lost.

jordens, 2018-02-28:
@hartytp You are trying to derive entitlement and leniency from alleged merits. That doesn't make sense. And if it would, it would be imbalanced.

jordens, 2018-02-28:
If you want to choose the role of customer, then the involvement stops there and you can't exercise micromanagement capabilities, assign issues and priorities, and walz all over the contractors work.

hartytp, 2018-02-28:
> If you want to choose the role of customer, then the involvement stops there and you can't exercise micromanagement capabilities, assign issues and priorities, and walz all over the contractors work.

Well, we find ourselves with blurred lines here. Equally, if you want to choose the role of the contractor then you can't go around publicly calling projects "trash fires" and talking about how they "screw you at every turn".

To be honest, some level of micromanagement seems to be needed atm. Take, for example, the 1V8 issue on Sayma. It turned up and @sbourdeauducq filed a bug. However, everyone was busy, so it got neglected. Rather than chasing Greg up and pushing him to fix it, a lot of time has been spent implementing a hacked work around (power cyclers etc) and moaning about trash fires (which, let's be honest we all tend to filter out and ignore). If a small fraction of that time had been spent on polite posts/emails to Greg to say "hey, this is really causing us problems, please can you look into it as a top priority asap" then I suspect that it would have been fixed ages ago -- it was just a couple of registers mis-set and it took half a day to debug and fix, just no one looked at it. Once I and others started posting repeatedly about this, it got fixed pretty quickly.

If you guys take on a more positive, professional attitude towards Sayma etc then I'll support you. But, with your current approach, I find that hard.

jordens, 2018-02-28:
@hartytp Yes. The complaining and whining about hardware (or software or gateware) is pure poison and utterly unacceptable. If you see or have seen me do it, please let me know. Please also check your own comments on that. I am working hard on getting people on the same page.

jordens, 2018-02-28:
And on the forking: I am all for having one repo per board. That's better from a technical, management and governance perspective. Wether they are under m-labs/quartiq/wut-ise/oxford/arl/nist/jqi or elsewhere doesn't mean fragmentation to me. On the contrary, it expresses the diverse nature of the project. Nor does the umbrella org mean leadership.

jordens, 2018-02-28:
And it's not hard to filter the git history for one board, even with the renames and pdfs. I had tried it for Kasli a while back. Now you only need volunteers to do it for Sayma etc.

jbqubit, 2018-03-02:
- [__censure__](https://www.google.com/search?site=async/dictw&q=Dictionary#dobs=censure) is the expression of formal disapproval.
- [__censorship__](https://www.google.com/search?site=async/dictw&q=Dictionary#dobs=censorship) is the suppression or prohibition of any parts of books, films, news, etc. that are considered obscene, politically unacceptable, or a threat to security.

closed 2018-03-05 by hartytp