sinara-hw/sinara #513: Characterise Si5324 + CDR (DRTIO clock) phase noise

hartytp, 2018-02-19:
Aim is to determine the noise floor that can be achieved by using the DRTIO clock for data converters etc. (e.g. Sayma), as well as the optimal Si5324 parameters.

- [ ] Measure phase noise of free-running Si5324 using evaluation board with Xtal
- [ ] Measure phase noise of free-running Si5324 with Xtal replaced by a decent ($10 ish) XO
- [ ] Measure phase noise of CDR clock output from FPGA. Measure phase stability (long term drift/very low frequency phase noise) of the CDR clock by beating the recovered clock agains the source clock.

hartytp, 2018-02-19:
Data from @WeiDaZhang, using the Si5324 evaluation board with its on-board Xtal reference. This should be the same as Kasli v1.1.

[image]

hartytp, 2018-02-19:
With the caveat that I in no way intend to suggest that it's worth the hassle of implementing this, it's worth noting that for Sayma RTM, we could replace the HMC830, Si5324 and HMC7043 all with the HMC7044. If we scrapped the Si5324 and used the 2-loop PLL in the HMC7044 then we'd actually use the majority of functionality in that chip and would probably get better overall performance -- the analog performance is certainly a *lot* better specified than the Si5324, which is appealing...

jbqubit, 2018-02-22:
Let's hold off on [revisiting](https://github.com/m-labs/sinara/issues/40#issuecomment-258821421) HMC7044 for now.

hartytp, 2018-02-22:
@jbqubit This is the right time to do this: it's something we can do in parallel with the current Sayma debugging; and we want to know the answer before we've finished getting Sayma v1.0 up and running, so it doesn't hold us up.

In any case, right now, it just boils down to characterising the components already on Sayma, which is important work to do.

closed 2018-04-17 by hartytp