sinara-hw/sinara #514: shipping to WUT

jbqubit, 2018-02-26:
Shipping items to WUT is a real problem.
 https://github.com/m-labs/sinara/issues/502#issuecomment-368553910 
https://github.com/m-labs/sinara/issues/358#issuecomment-367463958

@gkasprow What shipping options exist for inbound packages other than national mail service?

hartytp, 2018-02-26:
Currier? I'd love to visit Warsaw.

jbqubit, 2018-02-26:
@hartytp said 
> @gkasprow Really, that's odd. I posted that at the same time as the RF PA chassis, which you did get. Is it worth me getting the tracking number (UPS IIRC)? Or, is this another case of "lost in the polish postal system"? That's a lot of money on Saymas missing atm

gkasprow, 2018-02-26:
@hartytp personal visit would probably cost the same as DHL service:) Anyway any parcel service with tracking would work. DHL, TNT, Fedex they all work fine.

hartytp, 2018-02-26:
Tempting. The [RZD Paris to Moscow](https://www.seat61.com/paris-moscow-express.htm) service stops there, and has been on my travel list for a while!

gkasprow, 2018-02-26:
Spring/summer is much better time than winter, unless you like -15 degrees :)

closed 2018-02-27 by hartytp

jbqubit, 2018-02-27:
@gkasprow Please add to wiki your shipping address and instructions on how to ship without getting stuck in customs or routed via national mail service. [Team page](https://github.com/m-labs/sinara/wiki/Team) seems like a reasonable place.  I'd do it myself but for https://github.com/m-labs/sinara/issues/510.