sinara-hw/sinara #517: OpenMMC: compiling sayma RTM Board specification

jbqubit, 2018-03-01:
```
$ cmake .. -DMODULE_RTM=sayma
CMake Warning at CMakeLists.txt:19 (message):
  No toolchain configuration file specified.  Using default option!

-- Build type: Release
-- Selected Board: sayma
-- Selected Controller: LPC1776
-- Selected modules to compile: FRU;PAYLOAD;SDR;WATCHDOG;SCANSTA112;HOTSWAP_SENSOR;XR77129;RTM;ETHERNET;UART_DEBUG;SENSORS
CMake Error at modules/CMakeLists.txt:103 (message):
  RTM Module included but no RTM Board specified

-- Configuring incomplete, errors occurred!
See also "/home/britton/artiq-dev/mmc-firmware/build/CMakeFiles/CMakeOutput.log".
```

wizath, 2018-03-02:
Try using commands specified in the Makefile, e.g 
`-DBOARD=sayma -DBOARD_RTM=sayma -DTARGET_CONTROLLER=LPC1776 -DCMAKE_BUILD_TYPE=Release`

jbqubit, 2018-03-03:
Compilation successful. There are four files in the resulting 'out' directory. 

- Which is the firmware I should use? openMMC.axf or openMMC.bin?
- Using Flash Magic
    - LPC1766 is selected 
    - I read Device Signature: Device ID 27191F43 Bootloader Ver 8.1
    - when I flash/erase/read: "The device signature does not match the expected values."
    - when I write .bin or .afx I see: "Operation Failed. Nothing to program. Is the hex file inside the device memory range?"

jbqubit, 2018-03-03:
Can I ignore bootloader.axf and bootloader.bin?

gkasprow, 2018-03-03:
U need to generate special hex file for Flashmagic which need to contain checksum.
I have added post-compilation[ instructions](http://blog.ilektronx.com/2011/07/how-to-generate-hex-files-from-code.html) to my IDE that do the job.

jbqubit, 2018-03-04:
I followed the instructions for creating a build environment [here](https://github.com/m-labs/mmc-firmware/). The Debian package ```gcc-arm-none-eabi``` contains a tool called   ```arm-none-eabi-objcopy``` which is exposed as a command line tool called ``objcopy```. The following generates a ```.hex``` that FlashMagic accepts. 

```
$ objcopy -I binary openMMC.bin -O ihex openMMC.hex
```

jbqubit, 2018-03-04:
It appears DTR and RTS must be deasserted for MMC to boot. Several terminal programs I commonly use don't work (eg flterm, putty). The terminal in FlashMagic works as does [Terminalbpp](https://sites.google.com/site/terminalbpp/). Here's what I see upon booting.
```
![image](https://user-images.githubusercontent.com/20543958/36947693-2061eb6c-1f95-11e8-92b5-f79f1551d8e3.png)

The "AMC hotswap handle pressed" and "RTM hotswap handle pressed"  messages appear repeatedly. Please disable that for now. I don't see any change in behavior when I send "E" or "P".

jbqubit, 2018-03-04:
Current consume by AMC+RTM prior to updated MMC firmware was about 2.6 A on 12V. Following MMC firmware update it is 857 mA. @gkasprow is this as expected?

jbqubit, 2018-03-04:
Oh, I see. There's no power applied to RTM (all LEDs are dark). Creating new Issue.

jbqubit, 2018-03-04:
Thanks for the help @wizath. Closing.

henrique-silva, 2018-03-05:
@jbqubit just to clarify about the binaries (and .axf) that are generated:
- bootloader.bin(.axf) is a "standalone" firmware that runs first and checks if there's any image on the "Upgrade area" of the FLASH memory. If there is, it overwrites the "User area", which is where the main firmware (openMMC itself) runs.
- openMMC.bin(.axf) is the firmware itself, but since we have a large FLASH, we can store more than one version of it and leave the bootloader to manage it.

You need them both to run the firmware properly. The `booloader.bin` must be flash at location 0x0 and `openMMC.bin` at 0x2000.
If you'd like to program all at once, you can call `make full_binary`, and a `openMMC_full.bin` file will be created. This binary is just both bootloader and openMMC appended together with some padding in the middle, so you can flash it at 0x0, and both firmwares will be on place.

jbqubit, 2018-03-05:
I've been converting openMMC.bin to openMMC.hex and flashing to 0x0. I wonder why the MMC works at all??

----- 

Following your new instructions...

mmc-firmware/build$ cmake .. -DBOARD=sayma -DBOARD_RTM=sayma -DTARGET_CONTROLLER=LPC1776 -DCMAKE_BUILD_TYPE=Release -DBENCH_TEST=true
mmc-firmware/build/out$ make clean
mmc-firmware/build/out$ make full_binary
mmc-firmware/build/out$ objcopy -I binary openMMC_full.bin -O ihex openMMC_full.hex

When I flash openMMC_full.hex the MMC does not boot.

henrique-silva, 2018-03-05:
> I've been converting openMMC.bin to openMMC.hex and flashing to 0x0. I wonder why the MMC works at all??

Sorry, I wasn't very clear in the last message (:
It is supposed to work if you flash only the application bin (openMMC.bin/hex/axf) at 0x0 (this is the action that the bootloader mimics after verifying that there's no new firmware to update), but you lose the HPM functionality, as there's no way to overwrite the FLASH in the same space or even manage different boot locations.

> When I flash openMMC_full.hex the MMC does not boot.

Did you perform a full erase of the chip before flashing it? The bootloader is (very) simple and just checks if there's anything written at flash location 0xFFFC (last word on the defined section for the "user" firmware) other than 0xFFFFFFFF. If there's any garbage data there, it will trigger the bootloader to copy the whole data from 0x10000 up to 0x1E000 and (try to) boot it.

I'm finishing up a more robust version of the bootloader, asserting some checksums before rewriting the FLASH. It shouldn't take long to release it.

gkasprow, 2018-03-05:
@henrique-silva I'm not sure if we are using this bootloader. We use ROM-based bootloader placed there by the MCU manufacturer. @wizath please confirm.

jbqubit, 2018-03-07:
Reopening until this gets sorted.

wizath, 2018-03-07:
Ah, sorry but at this moment we are not using bootloader, base firmware flashes directly to address 0x00, so just flash openMMC.axf directly

hartytp, 2018-05-03:
Is there any outstanding issue here, or can we close this?

jbqubit, 2018-05-07:
@wizath Please review instructions for [building](https://github.com/sinara-hw/sinara/wiki/Sayma#building-openmmc-firmware-for-sayma) and [flashing](https://github.com/sinara-hw/sinara/wiki/Sayma#flashing-mmc) Sayma MMC. Then close.

marmeladapk, 2019-01-22:
Seems to be done: https://github.com/sinara-hw/sinara/wiki/Sayma#mmc-related

closed 2019-01-22 by jbqubit