sinara-hw/sinara #520: OpenMMC for Sayma_AMC no RTM power

jbqubit, 2018-03-04:
From https://github.com/m-labs/sinara/issues/517. There's no power applied to RTM (all LEDs are dark). Power was working fine prior to updating MMC firmware. I built my own firmware from master.

gkasprow, 2018-03-04:
@jbqubit Did you try it in the MTCA crate?

jbqubit, 2018-03-05:
No, the PCBs are not in the crate. This PCB pair worked fine prior to updating the MMC firmware.

wizath, 2018-03-05:
You have to add `#define BENCH_TEST` in order to run MMC without crate. It's not tested yet, but should work fine. You should see something like `BENCH Test active` on the UART log if the mode is activated.

jbqubit, 2018-03-05:
@wizath Which file should I edit? Please work closely with @filipswit to carefully document these things.

wizath, 2018-03-05:
You can edit `main.c` and add on top `#define BENCH_TEST 1`

henrique-silva, 2018-03-05:
@wizath Adding `-DBENCH_TEST=true` on the CMake call will have the same effect.
The CMake call will print in purple "Bench test mode is active!", but I haven't added any info com the UART log yet.

jbqubit, 2018-03-05:
I'm put my recently updated AMC into a uTCA crate along with an RTM. No power is applied to RTM. 

> -DBENCH_TEST=true
I will try this now.

jbqubit, 2018-03-05:
> -DBENCH_TEST=true

Does not work. Only P12V0 led illuminates on RTM.

wizath, 2018-03-05:
I'll get sayma tommorow and check that

jbqubit, 2018-03-05:
Where can I get an older .hex that work? Would like to continue debugging this afternoon...

gkasprow, 2018-03-05:
Try this. I didn't test it.
[lpc1776_ethernet_I2C.zip](https://github.com/m-labs/sinara/files/1782177/lpc1776_ethernet_I2C.zip)

jbqubit, 2018-03-05:
> lpc1776_ethernet_I2C.zip

Better but 1V2 and 2V0 on RTM are still off.

gkasprow, 2018-03-05:
Are you sure the transistor that turns them on is on its place ?

jbqubit, 2018-03-07:
The transistor was in the process of falling off on this RTM. I've fully removed the transistors. Using the .hex in your .zip the RTM powers on while sitting on bench top. I can boot AMC + RTM using gateware with no SAWG. 

When sitting for several minutes on bench top the current starts at 2.8 A, then steps to 3.5A and often ends at 3.9A. 

When AMC+RTM is inserted into uTCA crate, RTM is not powered.

wizath, 2018-03-07:
I've fixed the issue with RTM power, since at the initialization MMC tried to disable RTM power and in result blocked whole I2C interface. Everything should work fine now along with UART commands (#521).

jbqubit, 2018-03-07:
https://github.com/m-labs/mmc-firmware/commit/c18a7ecbb1b175b476c143f041592fe7b4083e4c I've compiled and reflashed. Now MMC does rolling reboot. No power is applied to AMC or RTM when inserted into the uTCA crate. Works fine on benchtop.

wizath, 2018-03-07:
Did you remove BENCH_TEST directive for uTCA crate? I'll test it in crate in a moment.

jbqubit, 2018-03-07:
I set -DBENCH_TEST=false. See attached for the .hex I'm using. The build script I'm using is [here](https://github.com/jbqubit/sinara-testing/blob/master/sayma/tools/build_mmc.bash). 

[mmchex.zip](https://github.com/m-labs/sinara/files/1789915/mmchex.zip)

wizath, 2018-03-07:
Try to remove whole BENCH_TEST declaration. As I can see, there are `#ifdef` declarations, so it does not matter if it's true or false.

`./modules/main.c:#ifdef BENCH_TEST
./modules/main.c:    printf("BENCH Test active\n");
./modules/sensors/hotswap.c:#ifdef BENCH_TEST
./modules/sensors/hotswap.c:#ifdef BENCH_TEST
./modules/rtm.c:#ifdef BENCH_TEST
./port/board/afc-timing/payload.c:#ifndef BENCH_TEST
./port/board/afc-bpm/v3_1/payload.c:#ifndef BENCH_TEST
./port/board/afc-bpm/v3_0/payload.c:#ifndef BENCH_TEST
`

jbqubit, 2018-03-07:
No. That doesn't help. 

Please test on your hardware using my build script. That way we are on the same page.

wizath, 2018-03-08:
I've replicated issue - apparently MMC enables hardware, but is not getting PWR level change message from MCH. Working on it atm.

jbqubit, 2018-03-09:
@wizath What did you find on this? Is there a hack you can implement that just makes cards work in rack?

wizath, 2018-03-09:
Quick way - comment the #ifdefs in rtm.c

```
//#ifdef BENCH_TEST
rtm_power_level = 0x01;
rtm_pwr_lvl_change = 0x00;
//#endif
```

That way, the RTM will start without getting message from MCH. It works on Vadatech MCH (I don't have NAT at this moment)

jbqubit, 2018-03-09:
In the crate the MMC applies power to neither AMC nor RTM. I don't see how commenting out BENCH_TEST will impact the AMC no-power problem.

gkasprow, 2018-03-09:
According to the MTCA specification, the power is enabled after MMC gets confirmation from MCH that it is capable delivering it (it asks for certain power capability). So if you don't plug it to the crate, the MCH won't communicate and give permission to MMC to turn the power on.
So to operate outside of the crate we can do two things
- flash the firmware that has negotiation disabled
- implement the mechanism that will enable payload power when MMC does not get ENABLE signal, after some timeout. I have foreseen that problem and there is a detector and the signal routed to the MMC called P12V0_OK that can be used together with ENABLE signal to detect off-crate operation.

jbqubit, 2018-03-09:
Powering boards on benchtop works fine. Powering boards /inside the crate/ does not work.

wizath, 2018-03-09:
This is strange, AMC is working fine for me on both MCH regardless of RTM status.

1. Is the RED led lit when you put sayma into crate?
2. Could you open serial monitor and toggle hotswap to see if there is any activity on console?

jbqubit, 2018-03-09:
On front of AMC red, blue and green lights are flashing in phase at about 3 Hz. On the serial terminal I see boot chatter repeated again and again. I suspect AMC is doing rolling reboot. I see same behavior in two different crates.

jbqubit, 2018-03-09:
Green indicator on MCH for corresponding AMC slot is flashing green.

wizath, 2018-03-09:
Could you show me this boot chatter?

jbqubit, 2018-03-09:
Here's an excerpt. 

[image]

[image]

wizath, 2018-03-09:
It looks like after enabling RTM system reboots. Can you try without RTM? I've configured the firmware to halt on any error to ease debugging so this bootloop seems strange to me.

jbqubit, 2018-03-09:
With RTM disconnected I see the following behavior.
- rolling display of "AMC Hotswap handle pressed!"
- front panel: blue led solid, green led flashing, red led off

[image]

jbqubit, 2018-03-09:
With RTM installed it looks like MMC reset happens in middle of printing "RTM Enable Hardwar"

wizath, 2018-03-09:
Anything happens if you toggle hotswap handle? AFAIK this rolling `AMC Hotswap handle pressed` could be an issue with IPMI commands, since the state is changed only if event is successfully sent to MCH `hotswap_send_event( hotswap_amc_sensor, new_state_amc ) == ipmb_error_success`

gkasprow, 2018-03-09:
Doesn't the MMC detect over-current event at that time?

jbqubit, 2018-03-09:
> Anything happens if you toggle hotswap handle?

With RTM removed... The rolling message changes between "AMC Hotswap handle pressed!" and "AMC Hotswap handle released!".

wizath, 2018-03-09:
Can you change line 72 in ipmi.c to `#if 1` and see if there are any IPMI messages received from the MCH?

jbqubit, 2018-03-09:
I've changed line 72 and reflashed. No IPMI messages. Rolling reboot of MMC observed.

wizath, 2018-03-09:
[mmc.tar.gz](https://github.com/m-labs/sinara/files/1798260/mmc.tar.gz)

Could you try these? Maybe there is something wrong with FlashMagic flashing, I'll try @ monday.

jbqubit, 2018-03-09:
I've converted your .bin to .hex and flashed. I see same rolling reboot.

gkasprow, 2018-03-09:
@wizath could you embed the build date to the message the MMC prints after the start? This would clearly indicate that the last version was loaded

jbqubit, 2018-03-12:
What's the next step on this? One approach would be to produce a simplified .hex that unconditionally turns on everything on AMC and RTM boards.

gkasprow, 2018-03-12:
I brought today to the lab the NAT MCH so finally @wizath can recreate the problem

wizath, 2018-03-13:
I've checked programming with FlashMagic under Windows.
[openMMC_fm_working.zip](https://github.com/m-labs/sinara/files/1807220/openMMC_fm_working.zip)

Please check this file.

jbqubit, 2018-03-13:
@wizath Is this intended to be used in crate or on bench top?

wizath, 2018-03-13:
Crate

gkasprow, 2018-03-13:
However I have an idea to make the MMC detect if it operates in the crate or on the bench. In the second case all the power supplies will be brutally on. This will simplify the things a lot.

jbqubit, 2018-03-13:
@gkasprow Agreed. Please make that the default for out of crate operation.

gkasprow, 2018-03-13:
@jbqubit did you manage to load the bit file @wizath posted above?

jbqubit, 2018-03-13:
Using lateset from @wizath on benchtop and in crate I see rolling reboot. 

[image]

[image]

wizath, 2018-03-14:
What FlashMagic settings are you using? This (again) looks like something happens after enabling RTM Hardware (it only initializes 3V3 MP nad LEDS). Can you try if AMC alone gets initialized?

I'm using these
[image]

jbqubit, 2018-03-14:
AMC+RTM installed in second AMC slot. Here's what I see from MCH console after cycling power on crate. 

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2018.03.14 10:23:59 =~=~=~=~=~=~=~=~=~=~=~=
 CU1 FRU 40 added
LSHM(0): CU1 front properties min 0 max 15 norm 3
LSHM(0): CU1 front now operational at level 11 (73%)
LSHM(0): CU1 rear properties min 0 max 15 norm 3
LSHM(0): CU1 rear now operational at level 11 (73%)
ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=12 SET_FRU_LED_S
TATE_REQ
AMC2(6): WARNING: entering Communication lost state !
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
Activation: modules are ready
Activation: all modules ready, Allowance Period (20 sec) stopped - continue with
 module startup !
Activation: starting AMC2
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=8 GET_DE
VICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
WARNING: usb: TX buffer(0) pending, dtdt=0x00008080, EPSR=0x000a0004, EPCR1=0x00
880000
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=15 GET_D
EVICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=32 GET_D
EVICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=43 GET_D
EVICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=57 GET_D
EVICE_ID_REQ
```

jbqubit, 2018-03-14:
@wizath... scratch that last MCH log... I was using older version of .hex from a week ago. Will repeat with your 3/13 .hex shortly.

jbqubit, 2018-03-14:
Recently I've been using FlashMagic with Interface "None (ISP)" which seems to work. 

Just now I've flashed openMMC_fm_working.hex using FlashMagic configuration as in your screenshot. 
RTM installed in second AMC slot. Here's what I see from MCH console after inserting AMC into crate. Rolling reboot again. 

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2018.03.14 10:36:48 =~=~=~=~=~=~=~=~=~=~=~=
: REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=15 READ_FRU_DATA_REQ

IpmiFru_ReadRecords(6): ERR - failed to communicate with EMMC
AMC2(6): WARNING: entering Communication lost state !
WARNING: usb: TX buffer(0) pending, dtdt=0x00008080, EPSR=0x000a0004, EPCR1=0x00880000
WARNING: usb: TX buffer(8) pending, dtdt=0x00088080, EPSR=0x000a0004, EPCR1=0x00880000
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2, result -11
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2, result -11
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2, result -11
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=15 GET_DEVICE_ID_REQ
AMC2(6): INFO - AMC removed in state M7
AMC2(6): stop management task 'AM74' for i2c 0x74
LSHM(0): CM sensor 254 LUN 2 <unknown> hotswap M0->M1
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=51 SET_FRU_LED_STATE_REQ
AMC2(6): WARNING: entering Communication lost state !
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=61 GET_DEVICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=4 GET_DEVICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=21 GET_DEVICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=32 GET_DEVICE_ID_REQ
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,1)ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)ipmi_SendFru(6): timeout - no response for REQ: 0x20->0x74, Seq=46 GET_DEVICE_ID_REQ

```

wizath, 2018-03-14:
Could you check with AMC alone?

jbqubit, 2018-03-14:
With AMC alone no rolling reboot of MCH. But AMC power LEDs still dark. As soon as I additionally plug in the RTM rolling reboot of MMC resumes. 

I advocate that we stop debugging this delicate MCH-IPMI-MMC dance for the moment. Is it possible to generate MMC source where MCH is 100% ignored. Unconditional power supplies on, peripherals configured.

wizath, 2018-03-14:
Ok, so i'll focus for a while to update EXAR configuration

jbqubit, 2018-03-14:
Before shifting to work on EXAR please generate source that unconditionally powers on and configures everything for Sayma AMC+RTM. Even better if this is a single (few) source solution. OpenMMC may be a good long term solution but with struggles to configure HMC830 and simply power AMC+RTM in crate the OpenMMC is a distraction.

gkasprow, 2018-03-14:
@jbqubit I've just tested. It works well in the crate and on the bench with same firmware loaded. Tomorrow will ship it.

[5 more replies]

closed 2018-05-24 by hartytp