sinara-hw/sinara #521: OpenMMC on Sayma_AMC doesn't respond to 'P' and 'E'

jbqubit, 2018-03-05:
I expect that sending 'P' to dump Exar registers and sending 'E' to send Ethernet PHY dump. I see neither in latest OpenMMC build.

jbqubit, 2018-03-07:
@wizath https://github.com/m-labs/mmc-firmware/commit/c18a7ecbb1b175b476c143f041592fe7b4083e4c didn't fix this problem.

wizath, 2018-03-07:
What serial console / settings are you using?

jbqubit, 2018-03-07:
It appears DTR and RTS must be deasserted for MMC to boot. Several terminal programs I commonly use don't give control over DTR/RTS (eg flterm, putty). But the terminal in FlashMagic works as does Terminalbpp. 

I use 115200 8N1 no handshaking DTR/RTS deasserted.

wizath, 2018-03-07:
Could you plug USB cable after MMC boot? I'm using standard configuration without any flow control, and it works fine.

jbqubit, 2018-03-07:
Steps.
1) apply power to uTCA crate
2) wait for MMC front panel lights to start flashing --- AMC/RTM PCB power LEDs are still off
3) attach micro USB cable to front panel -- front panel lights stop flashing, AMC/RTM PCB power LEDs are still off

Did you try using the .hex that I built?

gkasprow, 2018-03-14:
@jbqubit I found it! just type e and p :D 
@wizath will change the MMC code to react to capital letters.

jbqubit, 2018-03-27:
@wizath @gkasprow Using Sayma_AMC shipped from WUT to UMD 3/15/2018 (TS190717-8) with MMC code as-flashed by WUT. Success! Please post .hex of this firmware so I can apply to other AMC. :) 

```
openMMC Starting!
build date Mar 14 2018 18:29:50
CPU clock = 80 MHz
Selected board: Sayma
Could not find FRU information in EEPROM, building a runtime info...
>AMC FRU Information:
	-Board info area:
		-Language Code: 0
		-Manuf time: 10463040
		-Manufacturer: Creotech
		-Name: Sayma AMC
		-Serial Number: 1
		-Part Number: SAYMA
		-File ID: AFCFRU
No Chassis info area
No internal use area
	-Product info area:
		-Language Code: 0
		-Manufacturer: Creotech
		-Name: Sayma AMC
		-Part Number: 00001
		-Version: v1.0
		-Asset Tag: No tag
		-Serial Number: 1
		-File ID: AFCFRU
	-Multirecord Area: 
		-Module Current: 8 A
		-Zone3 Compatibility code: 0x11223344
>AMC FRU total size: 220 bytes
RTM Board detected!
PowerCould not find FRU information in EEPROM, building a runtime info...
RTM Board is compatible! Initializing...
RTM Enable Hardware
 override
AMC Hotswa31 0
GMIICR mod 18 0x0
PHY init done
nit done
p hand31 0
GMIICR mod 18 0x0
PHY init done
le pressed!
RTM Hotswap handle pr31 0
GMIICR mod 18 0x0
PHY init done
essed!
31 0
GMIICR mod 18 0x0
PHY init done
31 0
GMIICR mod 18 0x0
PHY init done
31 0
GMIICR mod 18 0x0
PHY init done
31 0
GMIICR mod 18 0x0
PHY init done
31 0
GMIICR mod 18 0x0
PHY init done
31 0
GMIICR mod 18 0x0
PHY init done
31 0
GMIICR mod 18 0x0
PHY iEnabling RTM Payload power...
RTM PWR Enable
DCDC en 146
------------ PHY Dump ----------
Register, ADDR DATA
BMCR 0 0x0
BMSR 1 0x0
ID1 2 0x0
ID2 3 0x0
AN_ADV 4 0x0
AN_RX 5 0x0
AN_EXP 6 0x0
EXT_STAT 15 0x0
page0
JIT_DIAG 16 0x0
PCSCR 17 0x0
GMIICR 18 0x0
CR 19 0x0
IR 20 0x0
page1
ID 16 0x0
GPIOCR1 17 0x0
GPIOCR2 18 0x0
GPIOSR 19 0x0
PTPCR1 20 0x0
page2
PTPCR1 16 0x0
?? 17 0x0
?? 18 0x0
?? 19 0x0
?? 20 0x0
page3
?? 16 0x0
?? 17 0x0
?? 18 0x0
?? 19 0x0
?? 20 0x0
------------ Exar Dump ----------
GET_HOST_STS 0x2 0x4 0x1
GET_FAULT_STS 0x5 0x0 0x0
PWR_GET_STATUS 0x9 0xf 0x0
PWR_READ_VOLTAGE_CH1 0x10 67 67 V
PWR_READ_VOLTAGE_CH2 0x11 220 220 V
PWR_READ_VOLTAGE_CH3 0x12 120 120 V
PWR_READ_VOLTAGE_CH4 0x13 100 100 V
PWR_READ_VOLTAGE_IN 0x14 160 160 V
```

closed 2018-03-27 by jbqubit