sinara-hw/sinara #523: Sayma Ethernet over uTCA backplane

jbqubit, 2018-03-07:
@wizath @gkasprow have you tried Ethernet on Port 0 of uTCA backplane? This is a parallel route to using SATA-SFP adapter and doesn't involve a dongle.

jbqubit, 2018-04-04:
@wizath @gkasprow Ping. Please do this while running ARTIQ. This is one of the last bits to be verified before we can move on with Metlino and updated Sayma AMC.

jbqubit, 2018-04-12:
@wizath @gkasprow Have you tried this?

jbqubit, 2018-04-13:
So you can't get it to work either with the SATA cable?? I thought this was working for you. Has Port0-SATA-SFP--->Ethernet ever worked at WUT for Sayma? https://github.com/m-labs/artiq/issues/854

gkasprow, 2018-04-13:
It was worked, but not with ARTIQ. Then we shipped away all boards and have only 2 that returned with some other issues.

jbqubit, 2018-04-13:
I've now setup SATA-SFP on the Sayma (s/n 7) that you sent me last week. Running ARTIQ from today's master. From panel LEDs are ```000G00G``` (0 dark, G green). So I understand that to mean that MII mode is active -- see [cheat sheet](https://github.com/sinara-hw/sinara/wiki/Sayma#front-panel-leds).

gkasprow, 2018-04-13:
Did you update your MMC with recent version ?

gkasprow, 2018-04-13:
Can you dump PHY registers (E command ) ?

jbqubit, 2018-04-18:
I'll hold off on ethernet-in-uTCA testing until you get this to work at WUT. I think there's too many Ethernet unknown for my skills to be useful.

gkasprow, 2018-05-10:
After one day of fighting with SFP modules here is what I discovered:
- I tool the SFP module Joe sent me (SGMII). Connected an Arduino board and run the I2C scanner tool. The only address the SFP responded was 0x50.
- I took working SFP module by default in 1000Base-x mode. Dumped all registers. 
- I loaded same registers to Joe module. No result
- then I realised that after the power off, the module config is maintained! I realised that I talked to the EEPROM chip instead.
- The Joe's module does not have 88E1111 connected to the I2C pins of the SFP module. Both modules have same PHY chip but my has bigger EEPROM (it occupied 4 addresses) and the PHY is located under 0x56 address
- it seems that the only way to play is to use correct SFP modules.
- I will check if the Sayma PHY can be switched to SGMII.

jbqubit, 2018-05-10:
I sent you 3 different types of SFP modules. Three worked with Kasli one did not. 

> it seems that the only way to play is to use correct SFP modules.

Sayma/Kasli don't have to work with all the SFP modules in the world. Rather, we need to specify the particular SFP module that end users should purchase. Once you complete your tests please update the [SFP wiki page](https://github.com/sinara-hw/sinara/wiki/SFP) with your recommendations. Better still if they can be purchased from the technosystem web store.

jbqubit, 2018-05-10:
> I will check if the Sayma PHY can be switched to SGMII.

Better to keep it simple and use just a single type of SFP module. If 1000Base-X works I don't see a need for adding SGMII support at this point. 

Have you tried have you tried Ethernet on Port 0 of uTCA backplane?

gkasprow, 2018-05-10:
@jbqubit Yes, you've sent me 2 types of fibre SFPs and one copper SFP. And this copper SFP does not work and probably did not work with Kasli.
Last success was establishing Ethernet connection on the boards I have in my lab. So far only one board does not want to cooperate and I'm debugging it.
I'm also trying to find why some 1000Base-x modules work and some don't.
I also tried to switch the Sayma PHY to SGMII, but the funniest thing is that even in SGMII mode, it works well with 1000Base-x SFP.
I also discovered that on my SFP-SATA boards the pull-down that enables the modules is too week, so some of them may not work at all.
I didn't try in the crate yet. First want to make sure the AMC boards work stand-alone.
So it seems that there are many reasons why we had such problems with Ethernet. PHY or MMC code was the least issue.

gkasprow, 2018-05-10:
I want to make sure there are no any unsolved issues with Ethernet before we produce Metlino and Sayma rev2

gkasprow, 2018-05-11:
The SFP modules Joe sent me (SFP-GE-T) did not responded to the I2C commands because the pulldown on TX_Disable was too weak. Once I replaced 10k with 470Ohm, the module started responding, it even established a link with PHY. But it does not work.
So we have following SFPs and media converters with status:

| model | type | comment | interface |
| ------------- | ------------    | ---- | ---- |
| SFP-GE-T | copper SFP |  | GBIC |
| MGBT1 | copper SFP |  | ? | 
| GLC-T | copper SFP | working | GBIC |
| GLC-T? | copper SFP | link on, not working | GBIC |
| MC220L | SFP<-> RJ45  converter | working | SFP |
| SFP-GE-BX | 1310/1490 10km |  | GBIC |
| SFP-GE-BX | 1490/1310 10km |  | GBIC |

gkasprow, 2018-05-11:
GLC-T registers content
ADDR  DAT page0 page1  
0 140 140  
1 141 141  
2 C01 C01  
3 F D  
4 4006 4006  
5 7800 7800  
6 0 0  
7 0 0  
8 78 78  
9 0 0  
10 C68 C68  
11 0 0  
12 4100 4100  
13 A A  
14 0 0  
15 0 0  
16 140 140  
17 141 141  
18 C01 C01  
19 D D  
20 4006 4006  
21 7800 7800  
22 0 0  
23 0 0  
24 78 78  
25 0 0  
26 C68 C68  
27 0 0  
28 4100 4100  
29 A A  
30 0 0  
31 0 0

gkasprow, 2018-05-11:
SFP-GE-T registers content:
ADDR  DAT page0 page1  
0 1140 1140  
1 7949 149  
2 141 141  
3 CC2 CC2  
4 DE1 801  
5 0 0  
6 4 4  
7 2801 2001  
8 0 0  
9 F00 F00  
10 4000 4000  
11 0 0  
12 0 0  
13 0 0  
14 0 0  
15 F000 F000  
16 378 378  
17 8150 8010  
18 0 0  
19 50 0  
20 C68 C68  
21 0 0  
22 0 1  
23 0 0  
24 4100 4100  
25 0 0  
26 2 2  
27 8084 8084  
28 0 0  
29 0 0  
30 0 0  
31 0 0

gkasprow, 2018-05-11:
I applied register update to non-working SFP plugins : 
 write16B(22, 0x00); //page0
      write16B(0, 0x140); 
      write16B(1, 0x149); 
      write16B(4, 0xC01);
      write16B(7, 0x2001);
      write16B(9, 0xE00);
      write16B(16, 0x78);
      write16B(17, 0x8140);
      write16B(19, 0x50);  
      write16B(26, 0xA);       
      write16B(27, 0x9088);
      
      write16B(22, 0x01); //page1
      write16B(0, 0x1140); 
      write16B(1, 0x149); 
      write16B(4, 0x0);
      write16B(7, 0x2001);
      write16B(9, 0xE00);
      write16B(16, 0x78);
      write16B(17, 0x8010);
      write16B(19, 0x0);  
      write16B(26, 0xA);       
      write16B(27, 0x9088);
   
      write16B(0x00, 0x8000 | (0x0F & read16B(00)));
      write16B(0x00,  (0x0F & read16B(00)));
And they started working !

jbqubit, 2018-05-11:
If there are SFP modules that work without tiny-CPU I advocate that we just use them. It would be nice to avoid adding another uP and development tool to the already large set of Sinara dependencies.

gkasprow, 2018-05-11:
I know. I wrote little program that detects if the SFP is in SGMII mode and if the chip is from Marvell. If you plug 1000base-x or any other SFP, it won't change any configuration. It also prints the SFP module ID  and current config mode on USB UART, so will help with debug. The CPU is just 2$ but I believe it's worth to have some diagnostic tool. If you don't want to use it, you don't have to. We can even make the CPU not programmed by default. When somebody wants to do some debugging, will connect Adruino IDE using USB cable, load the bitstream and communicate with it.

gkasprow, 2018-05-11:
@jbqubit I did all this investigation to make sure the Ethernet issue is caused by wrong SFP module interface, not some hardware problem. I managed to make all SFP modules I own running. If somebody has non-working media converters, I'd like to get them to verify what causes the problem and finally close the Ethernet issue.
I'd like to avoid situation that we select one working SFP module type, but it will cause problem on some Saymas or Kaslis from time to time because I did not dig deep enough.

hartytp, 2018-05-11:
@gkasprow thanks so much for doing a proper job on this. It's a pain but will be worth it when Sayma works reliably.

jbqubit, 2018-05-11:
I'm glad for all your work on this! Your solution does sound like it could work. Does anybody else have SPF adapters to send to @gkasprow?

hartytp, 2018-05-12:
@gkasprow Please can you summarise the things you've done to make ethernet work robustly? If we want to test with one of the non-fussy media converters, are there any HW/MMC changes that need to be made (e.g. the TRST pin)?

gkasprow, 2018-05-12:
The only thing that needs to e fixed is TRST pin. I also correctly connected LOS pin, but it is used only to generate interrupts which we don't use. I will post simple instruction with photos how to implement this fix.

gkasprow, 2018-05-12:
Not only SFPs, but any other media converters.
I use them in so many designs so finally wanted to know why some works and some don't :)

hartytp, 2018-05-12:
How will Ethernet on Metlino work? This same phy with rj45 on the fp or sfp + 1000BASEX?

gkasprow, 2018-05-12:
Metlino has it wired identical as Sayma.
There is connector to the MCH where Ethernet is routed to the main switch. it is also routed to SATA for debugging.
SFPs are routed to the GTH directly

hartytp, 2018-05-12:
So for artiq on Metlino the user connects Ethernet to a nat switch module which connects Ethernet to the uTCA BP, which connects to Metlino?

That seems quite complex compared with using a SFP directly on Metlino

gkasprow, 2018-05-12:
In MTCA Ethernet is connected to the MCH T1. It is distributed to all AMCs and MCH T3 where Metlino is installed. In this way we have access to all boards via Ethernet.

hartytp, 2018-05-12:
@gkasprow I understand that is how Ethernet is generally done in uTCA. But, my question is whether we will do that for ARTIQ/Sinara, or if we will try to do something cheaper/simpler?

AFAICT, we do not currently need to supply ethernet to any AMCs: all communication is via DRTIO over the BP; the FPGA and MMC firmware is updated by USB on the AMC FP.

So, would it be easier for Sinara if we didn't bother with the NAT Ethernet Switch on tongue 1, and use an SFP on Metlino's FP for ethernet the same way we do with Kasli. I guess that the downside is that we would need to port the M-Labs 1000BASEX code to KU, but other than that, it seems easier to use the SFP.

gkasprow, 2018-05-12:
We can do two things:
- route RGMII PHY to one of the SFPs. We have  3 of them but AFAIR their purpose was different. No code porting is needed. 
- keep PORT0 connection and use GTH with ported code.

gkasprow, 2018-05-16:
I made Ethernet running on the last AMC board we have. There was failed contact on Rx CLK. I asked Technosystem for X-ray inspection.
When I switched the rx clock to another clock input (DIO3), it works.

gkasprow, 2018-05-17:
2 AMC boards work in MTCA crate.

closed 2018-05-17 by gkasprow