sinara-hw/sinara #525: Kasli/v1.1 clock distribution

jordens, 2018-03-07:
The single ended clock outputs on board have gotten quite a bit noisier than in v1.0. Haven't checked AM vs PM. And the 440 kHz spur from before is now at 300 kHz. This is opticlock with free_run.

[image]

jordens, 2018-03-07:
I don't think I changed the Si5324 parameters w.r.t. the previous Kasli/v1.0 tests.

hartytp, 2018-03-07:
Ouch. What do the power rails look like?

hartytp, 2018-03-07:
Can you check am v pm by clocking urukul off it?

jordens, 2018-03-07:
## From Kasli/v1.1 125 MHz
### Urukul-AD9912/v1.1
[image]

### Urukul-AD9910/v1.1
[image]

## From Kasli/v1.0 125 MHz
### Urukul-AD9912/v1.1
[image]

But I probably won't have time to dig into this in the next couple of days anyway.

gkasprow, 2018-03-07:
Is the onboard oscillator turned off? There is dedicated CPLD pin for that purpose.

gkasprow, 2018-03-07:
Ups, I mixed it with Urukul :)

marmeladapk, 2018-03-07:
> The single ended clock outputs on board have gotten quite a bit noisier

All outputs look similar?

> What do the power rails look like?

I'm esp. interested in P2V5 LDO and P12V0 before and after filter. Unfortunately we're out of Kaslis at the moment so I can't measure that myself.

jordens, 2018-03-07:
I tested at least three of the four. And the 12v supply was a pretty clean linear one.

hartytp, 2018-03-08:
@marmeladapk Looking over the Kasli schematic again, it seems that all the filter inductors after the SMPS are actually ferrite beads. They are about 100R at 100MHz, but probably <1Ohm at a few hundred kHz (from a quick skim of the data sheet, I didn't see a Z versus frequency curve, so that's just a guess), so probably they're not actually doing much at all for the noise frequencies Robert is looking at. Also, the filter capacitors aren't that large (6V 22uF is probably <10uF with a 3.3V bias). 

Having said that, we have a good drop out for that LDO, so I would expect its PSRR to be excellent at a few hundred kHz.

The only think I can think of is that the PSRR of either the Si5324 of the clock buffer is *much* worse at Vcc=2V5 than at Vcc=3V3. As a temorary fix, we could try changing the 2V5 regulator set point to 3V3 and wiring it to run directly from the 12V supply. That's quick to do, and would confirm this as a PSRR issue (I'm happy to do that once my V1.1 Kasli arrives).

hartytp, 2018-03-08:
either way, good catch @jordens

marmeladapk, 2018-03-08:
@hartytp At 300 kHz LT3045 has 80dB PSSR.

> As a temorary fix, we could try changing the 2V5 regulator set point to 3V3

Could you try first setting it to 3,0 V without changing input? LDO has 250 mV dropout at 0,5 A (and we have lesser load) so it should manage it. Of course unless I get another Kasli first.

hartytp, 2018-03-08:
> Could you try first setting it to 3,0 V without changing input? LDO has 250 mV dropout at 0,5 A (and we have lesser load) so it should manage it. Of course unless I get another Kasli first.

Sure, we can try that. I'd expect the PSRR to be quite a bit worse at such a low dropout, so it's not clear how much that would help.

hartytp, 2018-03-08:
Having said that, previously the spur was at 440kHz which is consistent with the SMPSs frequency. Now, it's at 300kHz, so AFAICT can't be due to the SMPS. Maybe this is a coupling from some other noise source?

marmeladapk, 2018-03-08:
[image]

> I'd expect the PSRR to be quite a bit worse at such a low dropout

You're right.

jordens, 2018-03-08:
Na. The hardware is fine. I was using a bit too little grounding...
The dominant spurs are +-450 kHz from the SMPS at about -65 dBc.

hartytp, 2018-03-08:
> The dominant spurs are +-450 kHz from the SMPS at about -65 dBc.

That still seems very high, given how good the PSRR of the LDO is...

jordens, 2018-03-08:
Non-ideal layout and worse behavior at 2.5V maybe.

hartytp, 2018-03-08:
Well, I still think this needs further investigation, as that is *much* higher than it should be.

marmeladapk, 2018-03-08:
@hartytp Look at the graph I posted - 500 kHz at 0,8 V difference gives arround 65 dB PSSR.

jordens, 2018-03-08:
And it depends on what exactly is on the underside of Kasli. An ESD mat gives me -65dBc, upright and half inserted into a crate is -70dBc. And -- probably due to the somewhat radical grounding design -- if you are not careful, even just connecting to USB makes a big difference.

hartytp, 2018-03-08:
Right, so if @jordens's measurement was a reflection of power supply noise then it would seem that without the LDO we'd have approx 0dBc carrier sidebands on the clock. That's not very realistic, given the the SMPS noise isn't that high before any filtering!

Seems like something else is going on here...

> And -- probably due to the somewhat radical grounding design -- if you are not careful, even just connecting to USB makes a big difference.

:) Did you try replacing the grounding resistors/capacitors with solder blobs? When you do that, you have a standard ground situation (100MHz clock is too low a frequency for the small cut-outs to make that much difference once they're tied to the main ground in a few places with solder blobs).

jordens, 2018-03-08:
Don't make fun of the measurements. Why do you insist on assuming ideal layout and ideal conditions?

hartytp, 2018-03-08:
> Don't make fun of the measurements. Why do you insist on assuming ideal layout and ideal conditions?

I wasn't at all -- the measurements are useful and totally valid (the noise has to be low in realistic conditions, not under ideal conditions).

My point was that the spurs you're seeing can't be due to the limited PSRR of the LDO. The noise is far too high for that.

jordens, 2018-03-08:
Yes. I would have said that the PSRR of our LDO layout is unlikely to reproduce the datasheet PSRR. But it is ~20-30dB better than v1.0.

hartytp, 2018-03-08:
Well, I still think that if this is connected with the floated SMA grounds (assuming that's what you mean by "radical") then unfloating them by replacing the Rs and Cs with solder blobs should make a large difference.

I also think it's very unlikley to be power supply noise, since -65dBc is really quite big. But, again, we can test that by replacing the FB by the Si5324/ADCLK with a larger inductor and also adding a bigger decoupling capacitor. If that helps, it's an easy fix.

Given the symptoms you describe, it does sounds like a pickup/grounding issue, but it's not obvious to me where the problem lies.

jordens, 2018-03-08:
No. I didn't use the SMAs. I meant the fact that with a usual power bricks that have ~100 V AC capacitive at a few tens or hundreds of µA on their grounds (which tend to be touching the front panel), the rack being otherwise isolated, and the MMCX connected DC to the spectrum analyzer, and optionally the PC grounding Kasli as well, that is a tricky situation. If I isolate it completely (disconnecting USB and an UNUN on the MMCX), there is only the SMPS spur at -70dBc (Kasli half inserted into a crate).

a-shafir, 2018-03-08:
Currently most of the FPGA VCC I/O are tied together. Some of the I/O banks are quite noisy. It certainly makes sense to isolate power of some FPGA vcc i/o with inductors or LDO besides the other improvements.

hartytp, 2018-03-28:
Okay, so I finally found some time to look at this.

Mostly just reiterating what @jordens already found:
- Running Kasli off the OptiClock variant but with a 100MHz external reference from a synth
- Powering it from an isolated (floating) 12V supply (wall wart)
- Looking at 125MHz MMCX clock output on spectrum analyzer (E4405B)
- I see a horrific noise spectrum, qualitatively similar to the mess that @jordens posted above
- Grounding the shield of the coax for the MMCX *clock output* to the SFP cage removes the vast majority of the noise. This result doesn't seem particularly sensitive to how I do the grounding, e.g. I'm not using a careful low-impedance ground connection.
- So, it seems plausible that if Kasli is used to clock an EEM in the same rack then this may not be a problem so long as the PCB grounds are properly connected to the panels, which are in turn properly connected to the chassis.
- Similarly grounding the clock *input* ground to the PCB ground does not seem to make any difference, so the issue appears to be with the outputs/spectrum analyzer input, and not with the Kasli clock input.

hartytp, 2018-03-28:
However, a *much* more serious issue is that my synth popped up with the dreaded: Reverse Power Protection Tripped message (@dtcallcock reminds me of the bad days of killing synths with inside-outside DC blocks way back). 

So, I think it's fair to say that running Kasli from an isolated PSU and having all clock outputs floating is a recipe for disaster, as things just charge up way too much.

hartytp, 2018-03-28:
My recommendation is that the 100k resistors on the Kasli clock inputs/outputs should be replaced with 0R resistors, but that we leave the ground islands where they are. That way, the default option should work in all setups without crazy noise levels or dead instruments. But, if users find the need to remove ground loops they can replace the 0Rs with 100ks and take responsibility for ensuring that there is a ground connection between all instruments.

Thoughts @gkasprow @jordens?

hartytp, 2018-03-28:
Disclaimer: no synths were harmed in the taking of this data :)

hartytp, 2018-03-28:
The situation is presumably the same for clocker etc: we must either ensure that the PSU is always tied to mains ground (grounding bolt + warning, choice of PSU, etc) or short the output grounds to PCB ground by default and allow users to float them at their own risk.

hartytp, 2018-03-28:
Adding an explicit mains ground bond does not help noise issues.

I soldered a wire onto ground near the SFP cage and connected that to mains ground near my synth/spectrum analyzer. The DC potential difference between PCB ground and MMCX ground is now 5mV.

hartytp, 2018-03-28:
Okay, shorting across the 100k resistors on the SMA clock input and the MMCX clock outputs to tie all grounds together, I see no noise issues  at the -100dBc/Hz level when looking on a spectrum analyzer with a 100Hz span! 

This measurement does not depend on whether the mains grounding strap I added or the USB are connected.

[image]

So, *in this particular configuration* shorting across those 100k resistors seems to fix all issues.

Nice job @marmeladapk on a good design; that looks like a very clean clock.

hartytp, 2018-03-28:
@jordens can we close this issue now?

jordens, 2018-03-29:
Sure. My mind was made up when I closed it the first time.

closed 2018-03-29 by hartytp

hartytp, 2018-03-29:
@jordens ack.

I reopened the issue because you reported seeing -65dBc spurs from the SMPS, which alarmed me. But, if that was just a grounding issue, which can easily be fixed by de-floating the connectors then I'm happy that there is no issue here -- the clock distribution on Kasli seems to work beautifully in the configuration I tested.