sinara-hw/sinara #528: Sayma blink_led.py idle kernel silently stops

jbqubit, 2018-03-13:
Good news on my end. Using .bit's I compiled from last Friday (2edf65f5) with SAWG enabled I was able to get ARTIQ to run blink_led.py as idle kernel. This is the first I've successfully run a kernel on Sayma. 

Running ``artiq_flash start``` repeatedly, I see that boot proceeds to the stage where idle kernel can run about 25% of the time. Most common failures are SDRAM fail and AD9154-0 config attempt failed. Sometimes kernel panics happen after 30 seconds to 120 seconds of running. 

```
panic at src/libcore/result.rs:906: cannot initialize AD9154: "JESD ready timeout"
backtrace for software version 4.0.dev+722.g2edf65f5:
0x40023984
0x400436a0
0x40005ef0
0x40002fa4
0x400236f4
halting.
```

For blink_led.py, when the idle kernel does run it works for 5 to 20 seconds then TTL/LED output stops. No error is emitted on terminal.

jbqubit, 2018-03-14:
@sbourdeauducq does this work on your setup?

sbourdeauducq, 2018-03-14:
With flaky SDRAM, all sorts of crashes can happen (though the JESD failures are most likely another issue, that is already reported elsewhere). Yes, I can run idle kernels, and they don't crash - and my SDRAM is more stable than on your boards.

sbourdeauducq, 2018-03-14:
(I can also run regular kernels with Ethernet)

hartytp, 2018-03-14:
@jbqubit Let's shelve this for now, while M-Labs are working on the SDRAM.

jbqubit, 2018-03-14:
Ok. Closing.

closed 2018-03-14 by jbqubit