sinara-hw/sinara #529: Sayma blink_led.py idle kernel demo_2tone.py RTIO sequence error

jbqubit, 2018-03-13:
Using 2edf65f5 I built locally I see the following when running demo_2tone.py as idle kernel. 

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+722.g2edf65f5.dirty
Initializing SDRAM...
DQS initial delay: 97 taps
Write leveling scan:
Module 3:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001010001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101111110000010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
0000000000000000000000000000000000000000000000000000000000000000000000000001000011111001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
0000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000100010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 97 taps
Write leveling: 72 85 102 97 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000001000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000010000100111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000101101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101101000010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 3:75-207 2:62-198 1:40-177 0:33-168 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003889s]  INFO(runtime): software version 4.0.dev+722.g2edf65f5
[     0.010157s]  INFO(runtime): gateware version 4.0.dev+722.g2edf65f5.dirty
[     0.016962s]  INFO(runtime): log level set to INFO by default
[     0.022665s]  INFO(runtime): UART log level set to INFO by default
[     0.028809s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.701793s]  INFO(board_artiq::serwb): done.
[     0.704940s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+722.g2edf65f5.dirty
[     0.712912s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1.712006s]  INFO(runtime): continuing boot
[     1.715023s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     1.721293s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 configuration...
[     1.739737s]  INFO(board_artiq::ad9154): AD9154-0 found
[     1.743661s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     1.826585s]  INFO(board_artiq::ad9154): AD9154-0 PRBS test
[     2.841580s]  INFO(board_artiq::ad9154): AD9154-0 found
[     2.845497s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     2.938731s]  INFO(board_artiq::ad9154): AD9154-1 found
[     2.942653s]  INFO(board_artiq::ad9154): AD9154-1 configuration...
[     3.025578s]  INFO(board_artiq::ad9154): AD9154-1 PRBS test
[     4.040574s]  INFO(board_artiq::ad9154): AD9154-1 found
[     4.044492s]  INFO(board_artiq::ad9154): AD9154-1 configuration...
[     4.143295s]  INFO(runtime): using MAC address 00-0a-35-03-1e-75
[     4.148161s]  INFO(runtime): using IP address 192.168.1.75
[     4.153702s] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry, using internal RTIO clock
[     4.165223s]  INFO(runtime::mgmt): management interface active
[     4.178403s]  INFO(runtime::session): accepting network sessions
[     4.192571s]  INFO(runtime::session): running startup kernel
[     4.197097s]  INFO(runtime::session): no startup kernel found
[     4.202784s]  INFO(runtime::session): no connection, starting idle kernel
[     4.236614s]  INFO(runtime::kern_hwreq): resetting RTIO
[     4.240909s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 22
[     4.342486s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.362510s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.422528s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.442549s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.502567s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.522501s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.582519s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.602540s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.662558s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.682579s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.742511s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.762532s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.822550s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.842569s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.902588s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     4.922608s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     4.982626s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     5.002559s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     5.062581s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     5.082603s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     5.142626s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     5.162645s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
[     5.222577s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 78
[     5.242596s] ERROR(runtime::rtio_mgt): RTIO busy error involving channel 82
```

jbqubit, 2018-03-26:
Using build from master today I see the following running ```demo_2tone.py```. I can see TTL on scope but no RF. 

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+796.g5ca59467.dirty
Initializing SDRAM...
DQS initial delay: 97 taps
Write leveling scan:
Module 3:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011100010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001101111
Module 2:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001011011
Module 1:
0000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010001001111111111111111
Module 0:
0000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000100011111111111111111111
DQS initial delay: 97 taps
Write leveling: 62 67 88 82 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000110111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
11111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 98+-63 90+-65 71+-66 63+-63 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003889s]  INFO(runtime): software version 4.0.dev+796.g5ca59467
[     0.010157s]  INFO(runtime): gateware version 4.0.dev+796.g5ca59467.dirty
[     0.016962s]  INFO(runtime): log level set to INFO by default
[     0.022665s]  INFO(runtime): UART log level set to INFO by default
[     0.028809s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.186510s]  INFO(board_artiq::serwb): done.
[     0.189659s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+796.g5ca59467.dirty
[     0.197629s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1.197006s]  INFO(runtime): continuing boot
[     1.200026s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     1.206293s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 configuration...
[     1.224748s]  INFO(board_artiq::ad9154): AD9154-0 found
[     1.228673s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[... cut ...]
[     4.040171s] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry, using internal RTIO clock
[     4.051686s]  INFO(runtime::mgmt): management interface active
[     4.064869s]  INFO(runtime::session): accepting network sessions
[     4.079039s]  INFO(runtime::session): running startup kernel
[     4.083571s]  INFO(runtime::session): no startup kernel found
[     4.089256s]  INFO(runtime::session): no connection, starting idle kernel
[     4.122930s]  INFO(runtime::kern_hwreq): resetting RTIO
[     4.127229s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 62

```

jbqubit, 2018-03-26:
Correction... Above test was for different sawg channels. Repeating using using sawg0 and sawg1 as in stock demo-2tone.py the error is 

```
[     4.772822s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 22
```

jbqubit, 2018-03-27:
This looks reproducible day to day using same build. Seems to be triggered by sawg.reset().

jbqubit, 2018-04-04:
Is this expected behavior?

jbqubit, 2018-04-12:
@sbourdeauducq @jordens What do you see when running this? 

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+796.g5ca59467.dirty
Initializing SDRAM...
DQS initial delay: 110 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110100000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000110111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101100000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 110 taps
Write leveling: 109 106 119 118 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000101110001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 233+-86 217+-93 200+-89 188+-86 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003889s]  INFO(runtime): software version 4.0.dev+796.g5ca59467
[     0.010157s]  INFO(runtime): gateware version 4.0.dev+796.g5ca59467.dirty
[     0.016980s]  INFO(runtime): log level set to INFO by default
[     0.022671s]  INFO(runtime): UART log level set to INFO by default
[     0.028809s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.750526s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     1.471482s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     2.192435s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     2.913387s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     3.634339s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     4.355291s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     5.076243s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[... cut ...]
[    11.997243s]  INFO(runtime): using IP address 192.168.1.75
[    12.002798s] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry, using internal RTIO clock
[    12.014296s]  INFO(runtime::mgmt): management interface active
[    12.027480s]  INFO(runtime::session): accepting network sessions
[    12.041648s]  INFO(runtime::session): running startup kernel
[    12.073132s]  INFO(runtime::kern_hwreq): resetting RTIO
[    12.077434s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 22

```

sbourdeauducq, 2018-04-13:
This is not the expected behavior, and I did not see any of this before.

sbourdeauducq, 2018-04-13:
Maybe RTIO channel mixup? @whitequark how far did you get with the Sayma tests?

whitequark, 2018-04-13:
@sbourdeauducq I've only tested without SAWG so far. I'm going to first bring everything up using NVSynth, and then move to testing with SAWG.

sbourdeauducq, 2018-04-13:
Also @jbqubit, RTIO sequence or busy errors are almost certain to be software or gateware (not hardware) problems, so you should file this into the artiq issues (but leave this one here for now).

jbqubit, 2018-04-13:
@whitequark Please get on with this. It's been weeks now since you started trying to get DACs on Sayma to work. 

@sbourdeauducq OK. If I see something like this in the future I'll create an ARTIQ issue (but not in this case).

hartytp, 2018-05-03:
Not a hardware issue.

jbqubit, 2018-05-07:
@sbourdeauducq Is this resolved? If not please create an ARTIQ Issue. I left it in sinara at your request... 

> @sbourdeauducq said
so you should file this into the artiq issues (but leave this one here for now).

sbourdeauducq, 2018-05-10:
Sine wave demo works as expected.
Sequence error bug reproduced and minimized. https://github.com/m-labs/artiq/issues/998

sbourdeauducq, 2018-05-10:
demo_2tone also works as expected, at least after applying the workaround in https://github.com/m-labs/artiq/issues/998.
Note that you have to trigger the scope on the TTL (J15/IO1), this demo only produces short bursts ~800ns after the rising edge of the TTL.