sinara-hw/sinara #530: Rectifier: RFC

hartytp, 2018-03-16:
Proposal to build a new EEM for RF power detection.
- Working name is "Rectifier" unless anyone cares to suggest something better
- Based on ADL5511 RMS detector or similar
- Inputs are SMAs, target something that works reasonably up to 6GHz (limited by the ADL5511)
- Analog outputs (RMS) on an IDC to work with Sampler. The IDC should be the sole source of power for this design (need to check that the power consumption is okay to be supplied by Sampler).
- Make the design 8 channels to work with Sampler IDC. Will require stacked SMAs.
- Only use the RMS outputs from ADL551, not the envelope detector outputs unless someone really wants the envelop detector (they need a bit more circuitry, since they require differential detection IIRC).
- IIRC, the ADL5511 outputs are very weak and should have an OpAmp buffer. Also probably worth adding a CMC per channel.
- Power LED on the FP.

gkasprow, 2018-03-16:
@hartytp Do we need the AFE part of the sampler? And we occupy precious IDC just to get power of it.
It may make sense to just replace AFEs on the Sampler with RMS detectors to keep the software compatibility and keep the cost lower. Do we need 1MS/S ADC? Maybe one used on Novo would be sufficient. We have ARTIQ driver for it already.

hartytp, 2018-03-16:
> Maybe one used on Novo would be sufficient. We have ARTIQ driver for it already.

I have an ARTIQ driver for Sampler nearly complete. Just need to find a few hours to tidy it up and push it. I expect it a PR in very soon.

I'd prefer to keep the same ADC as Sampler for simplicity unless there is a big cost difference. In the long run, we won't produce any more Novo, but only Sampler, so it's nice to keep all designs using the same ADC.

> Do we need the AFE part of the sampler?

No, I don't think we do.

> And we occupy precious IDC just to get power of it.

My idea was that we would use the IDC on Sampler both to get power off it and to send the Analog signals (RMS measurements) to sampler for digitization.

> It may make sense to just replace AFEs on the Sampler with RMS detectors to keep the software compatibility and keep the cost lower.

Sure, that's another option as well. If we go down this route, we might want to make it a 4-channel PCB.

My thinking was that it might be easier to reuse the Sampler design as it is, and just add a really simple analog PCB to it via the IDC for the rectification. That way, there is less extra code to write, and we can benefit from cost savings with producing more Sampler -- I don't expect us to need many of the Rectifier designs, so the production cost will be higher than Sampler. But, I'd also be happy with your approach.

hartytp, 2018-03-16:
@gkasprow For some of the lab uses I would have for Rectifer, I would definitely want simultaneous sampling, and ideally the higher BW of Sampler (we can write some simple code to use all 4 lanes and run at 1.5MSPS). So, let's either just make this a separate PCB that plugs into Sampler, or make it a full EEM with the same ADC as Sampler. I don't mind which approach you take, go for the easier/cheaper one!

gkasprow, 2018-03-16:
It's easier to modify and simplify the Sampler due to power limitations. I would have to convert low noise voltage that supplies Sampler AFE to 5V, 180mA using DC/DC covnerter.
Another fact is that if you connect it to the Sampler using IDC, someone may plug BNC connectors simultaneously and break something or at least get confused with results.

gkasprow, 2018-03-16:
On the other hand, if we want to use 8 SMA connectors at 6GHz, we have to go for SMA ones. Since stacked SMAs don't exist (at least I didn't manage to find any) the easiest way is to use a PCB that is installed in parallel with the panel. There are plenty of low cost vertical SMD SMAs with THT mounting pins. If we go for 8HP panel, we would be able to fit the ADL chips and IDC connector on tiny board.
I will have a look once I finish other critical things.

hartytp, 2018-03-16:
No rush for this, as Booster is much more important.

FWIW, if we keep this as a small analog-only PCB with an optional FP then I can imagine other use-cases for it. e.g. use it for microwave power levelling loops with the Sampler ADC. For that, it might be worth adding some internal MMCX outputs and a molex DC connector, in parallel with the IDC.

Vertical SMAs sound like a good idea. 

IIRC, these power detectors aren't incredibly low-noise anyway, so we can use a pretty cheap Ti SMPS IC if necessary.

dhslichter, 2018-03-16:
Not sure how accurate you want to be, but a really simple solution would be to spin up a very simple board with an unbiased Schottky diode, input impedance matching, and output RC filtering, with an input and output connector, and just plug that onto the Sampler input connectors.  If you just need a bunch of feedback loops for power leveling, that would be just fine, as long as your lab temperature is reasonably stable.

hartytp, 2018-03-16:
I did consider that, but the ic detectors are nice as they come with a good initial calibration and are really quite stable and well matched.

dhslichter, 2018-03-16:
> I did consider that, but the ic detectors are nice as they come with a good initial calibration and are really quite stable and well matched.

I guess it depends on what you want to do -- if you just need amplitude stabilization you can forgo the calibration, but it seems that with an Urukul driving it would be very straightforward to calibrate each diode (since you have clean control of the drive amplitude, and can use a nice power meter to get an absolute number for a given Urukul amplitude word).  You just put an Urukul through a power sweep and look at what you see on your Sampler and then boom, there it is.  Save it as a lookup table for each Sampler channel.  The advantage here is:

- no power required
- incredibly simple board, 2-layer FR4 will work
- dirt cheap parts and board
- no modifications to Sampler are necessary

Basically you could have these things up and running in 3 days for prototyping.

hartytp, 2018-03-17:
The things I want input on are:
- would anyone else use this?
- number of channels?
- put adc on board or rely on a separate sampler/sayma?

hartytp, 2018-03-17:
On second thoughts let's not bother with this.

If Greg doesn't want to use it for production testing I have a thumbnail sized single channel adl5511 breakout pcb which is probably better suited to my needs than this.

closed 2018-03-17 by hartytp

gkasprow, 2018-03-17:
For production testing we use 14bit 100Mhz ADC that sample the DAC signal, calculate FFT and verify linearity.

hartytp, 2018-03-17:
Sounds good @gkasprow

dhslichter, 2018-03-19:
@hartytp thumbnail sized breakout PCB sounds good...that's what i was driving at, something that doesn't mean a redesign/respin of the whole sampler deal.  If you want to make a cheap, packaged version of the ADL5511 thumbnail board, we'd potentially go in on that with you...

hartytp, 2018-03-19:
> that's what i was driving at, something that doesn't mean a redesign/respin of the whole sampler deal. 

FWIW, I was never diving at a complete res-pin of Sampler. Rather, a simple PCB that provided an AFE for the existing Sampler design, and plugged in via the IDC that's already on Sampler.

Essentially, the difference is between a PCB the size of a single Illuminati with a single ADL5511 or a PCB the size of 8 thumbnails with 8 ADL5511s on it.

>  If you want to make a cheap, packaged version of the ADL5511 thumbnail board, we'd potentially go in on that with you...

I have a simple eagle design for this already, which we'll probably stick with. Happy to distribute if it's helpful, but it's basically just trivial.

hartytp, 2018-03-19:
;) oops.