sinara-hw/sinara #534: Zotino v1.1 testing:

hartytp, 2018-03-19:
Things that I think are work testing on Zotino v1.1:
- [ ] Step response/rise time: switch from -10V to +10V and look at output on scope
- [ ] Noise spectrum with & without constant DAC updates. Look at +-FS and 0V
- [ ] Cross-talk (both before and after a 2m SCSI cable)
- [ ] Noise with TEC running
- [ ] Stability with + without TEC connected
- [ ] Small + large signal BW

hartytp, 2018-03-19:
@gkasprow I want to get the rest of the Zotinos we've ordered shipped asap. What's the minimal amount of testing we need to do before shipping? I'd say a rise-time + quick noise measurement should be fine.

gkasprow, 2018-03-21:
I've just populated remaining boards with TEC and started testing them.

gkasprow, 2018-03-21:
@hartytp 
step response 0x0000 -> 0xFFFF

[image]

gkasprow, 2018-03-21:
step response 0xFFFF->0x0000
[image]

gkasprow, 2018-03-21:
The same channel just at the DAC output. The tiny glitch disappeared. It seems it is produced by the output amplifier during transients. It resembles situation where  the opamp is overdriven for short time during the ramp. The output moves with the OA slew rate and then the feedback loop is closed at the end. Such transition causes the glitch. But the slope is far below opamp SR...
Similar glitch occurs at the same voltage level during high to low transition.
[image]

gkasprow, 2018-03-21:
other channels have same glitch

gkasprow, 2018-03-21:
The DC level on CH2 varies between 5.33 and 5.30mV when I change CH1 0x0000 -> 0xFFFF and back.
The CH1 output varies -6.678 and 11.293V so the DC cross-talk is 30uV/ 18V

gkasprow, 2018-03-21:
I don't have any SCSI-BNC adapters so cannot test it with ADC board...I have to focus on Booster. Another chassis is assembled, mechanical bugs seem to be fixed.

hartytp, 2018-03-21:
30uV/18V=-116dB for the DC cross-talk. Data sheet specifies 20uV typ/100uV max on a channel set to mid-scale (0V) when another channel has a full scale change. So we seem to hit the dac limit here. Nice @gkasprow!

hartytp, 2018-03-21:
@gkasprow I think the glitch is an OpAmp slew rate thing because of the active filter topology we're using -- I bet it would go if we changed the active filter to a unity gain buffer (would be a good thing to try at some point as a diagnostic).

Assuming that's the case I vote that we note this in the docs but don't worry about it: one generally doesn't operate the DACs hard in the slew rate limit when one cares about noise (program in a slower ramp via SPI). We looked at OpAmps carefully during the initial design and couldn't find any with better slew rate without significantly higher cost and quiescent current draw.

So all tests so far look good!

hartytp, 2018-03-21:
Ps great 're Booster!

jordens, 2018-03-26:
### FEXT

Setup: agressor on ch1, ch0,2-7 terminated, 8-31 open, BNC-IDC, 15cm IDC26, IDC-HD68, 2m HD68-HD68 SCSI-3 shielded twisted pair, HD68-IDC, 15cm IDC26, IDC-BNC, victim on ch0.

#### High impedance load, step crosstalk, time domain
4V <20ns edge agressor, high impedance scope on victim.
3mV/4V crosstalk 7µs ring down period, the transient in the first µs is about 7 mV but not relevant for Zotino given its 70 kHz 3 dB bandwidth.
That's ~15 nVs, c.f. 10 nVs for the DAC itself.
[image]

#### Low impedance load, frequency domain
Network analyzer, agressor 50 Ohm, victim 50 Ohm
0.2 MHz: -50 dB
0.5 MHz: -80 dB
1 MHz: -70 dB
2 MHz: -60 dB
4 MHz: -50 dB

hartytp, 2018-03-26:
Nice! Thanks @jordens

hartytp, 2018-03-26:
[image]

Greg, here is what I measure the step response to be. No glitches evident.

- Looking at the DAC output on a 12-bit fast scope (MSO-S 404A). BW limit set to 20MHz (digital + analog) no averaging, DC coupled, high-z
- Using a short IDC cable, IDC->BNC adapter, 30cm BNC.

```from artiq.experiment import *

class Zotino(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("zotino0")

    @kernel
    def run(self):
        zotino = self.zotino0
        self.core.reset()
        zotino.init()

        while(1):
            zotino.set_dac_mu([0], channels=[0])
            delay(100*us)
            zotino.set_dac_mu([0xffff], channels=[0])
            delay(100*us)```

hartytp, 2018-03-26:
@gkasprow can you give me any more info about your setup to help me reproduce the glitches you saw?

gkasprow, 2018-03-26:
I observed the signal directly at the opamp output using scope probe.
I used devkit connected to the Zotino and ADI config software.

gkasprow, 2018-03-26:
The glitch was probably filtered out by the output filter.

hartytp, 2018-03-26:
Here is what I see at the DAC output, measured at the DAC side of R24.

[image]

hartytp, 2018-03-26:
So, my guess is that this glitch is just the DAC's expected behavior under a large step change, which the filter is doing an impressive job of smoothing out (although, I can't find a figure on the DAC data sheet to confirm that).

I don't see any cause for concern here. Does anyone else?

hartytp, 2018-03-26:
Falling edge data as csv.

[falling_edge.txt](https://github.com/m-labs/sinara/files/1847695/falling_edge.txt)

hartytp, 2018-03-26:
Started looking at the noise measurements, but I don't have anything with enough BW and a low enough noise floor to make that measurement. Plus, in my lab I'm swamped by ambient pickup (appears in the cabling after Zotino). So, I don't think I can make any useful measurements there. @jordens or @gkasprow thoughts?

jordens, 2018-03-26:
I'd hunt down an old HP3561A (with or without the bubble memory ;) or build something equivalent. Ambient pickup would probably also be an issue for your trap. I suspect that getting away from Zotino etc using the HD68 cable into a quiet box and out of it again using IDC-BNC might actually make a difference compared to a flying ribbon from Zotino to IDC-BNC. Or just solder a thin coax to your dynamic signal analyzer to a channel on the underside of the IDC26 on Zotino.

hartytp, 2018-03-26:
Yes, a 2ch adc would probably be better, just don't have one conveniently set up. If someone else takes the data they might find another setup more convenient.

dtcallcock, 2018-03-26:
We picked up a refurbed HP89410A for around $3k. 2ch, DC-10MHz and -148dBm/Hz sensitivity.  It's in the sweet spot where it's old enough that it's cheap but new enough that the user interface is fairly useable (4 colour screen!!!).

dhslichter, 2018-03-26:
We performed a measurement in our lab of our own eval-board version of Zotino (identical output filter stage) with the HP89410A a while back and IIRC it met the nominal noise spec.  You have to split the channel into both inputs and do the cross-correlation so it takes a while.  If you care I can set it up again and save the data for posting.

hartytp, 2018-03-27:
@dtcallcock @dhslichter Thanks!

Are you planning to buy a Zotino to play around with? Apart from the uncertainty over the noise spectrum I've given it a pretty decent test and am really happy with it.

I'd be more interested in seeing a spectrum taken with Zotino itself than with an eval board so we can check for power supply noise etc If you could do that it'd be a great contribution and I'd be really grateful. You'll need to be quite careful over screening etc to avoid pickup but you know all that stuff.

sotirova, 2018-04-11:
We measured the temperature coefficient of 2 Zotino v1.1 boards. The temperature was controlled using the peltier contacts in a feedback loop with the thermistor. We measured the actual temperature of the dac by using an external thermistor. Dac voltage set to 9V

**Results:**
0.56 ± 0.046 ppm/°C
1.6 ± 0.11 ppm/°C

[image]

jordens, 2018-04-11:
Hmm. What's your hypothesis for the mismatch between the two boards and w.r.t. the datasheet 5 ppm/K full scale typical value?

jordens, 2018-04-11:
Here are some hasty measurements of the noise. May be worth repeating some of it (esp. the 1MHz-10MHz band) with correlation on a single channel.

Measurement floor. Single channel. 1 M impedance.
[image]

CH0, 0V, unfortunately it ranged up by 8 dB to -22dBm
[image]

CH0, -9V, no difference w.r.t. 0V
[image]

CH0, -9V low frequencies
[image]

CH0, -9V very low frequencies, but only a partial sweep up to 15 Hz.
[image]

CH0-CH1 -9V correlation, unfortunately not on a single channel. 100 kHz knee and plateau.
[image]

sotirova, 2018-04-13:
I cannot think of an obvious reason that the temperature coefficient is different from the datasheet. As we can see the temperature coefficients of the two boards are also not quite the same.

hartytp, 2018-04-13:
hmm 5ppm FSR/K is specified as the maximum temp co for the DAC without any sign or other information about the statistics. One interpretation of that would be that the temp-cos are a normal distribution with 0 mean and (a few)*sigma of 5ppm. That's consistent with the RMS being 1ppm or so.

There could also be some degree of fortuitous cancellation with, e.g., the 1ppm/K temp co of the reference.

Without better statistics (testing more boards) it's hard to draw a definite conclusion from this. However, if this is really repeatable on a decent number of boards then I'd argue that we can DNP the temperature stabilization stuff -- that was only just needed to meet our spec assuming a 5ppm FS/K temp co.

jordens, 2018-04-13:
As I said the datasheet specifies 5 ppmFSR/C **typical**, not max.

hartytp, 2018-04-13:
Apologies, you're right, I misread/misremembered that! No explanation then.

Looking again at the schematic, there should be no other temp cos at that level which could result in a cancellation.

One possible (but doesn't seem that likely) explanation is that the offset and gain drift happen to cancel out at the one voltage used for this measurement.

hartytp, 2018-04-13:
> But the the fact that it's 10dB above the -30dBm range floor (and clearly not 8) is weird.

Slight calibration issue? Or slightly different noise floors for different range settings?

hartytp, 2018-04-13:
- If I understand what you've done correctly, the low frequency noise floor for the -22dBm measurement range is expected to be something like -134dBVrms/rtHz + 8dB = -126dBVrms/rtHz=500nV/rtHz.
- The plot above gives something like -128dBVrms/rtHz, so consistent with the instrument noise floor give or take a couple of dB (which could be cal error etc).
- We expect 200nV/rtHz = -133dBVrms/rtHz for Zotino. If the instrument noise were -133dBVrms/rtHz as well then we would measure -130dBVrms/rtHz, so this is shows that the noise can't be more than a couple of dB worse than expected.

hartytp, 2018-04-13:
For the higher-frequency noise (ignoring spurs):
- Noise floor for the -22dBm range is expected to be about -163dBVrms/rtHz + 8 = -155dBVrms/rtHz = 18nV/rtHz.
- You measure -153dBVrms/rtHz=22nV/rtHz. 
- We expect Zotino to hit 22nV/rtHz at around 200kHz.
- Looking at your plots, the noise corner (3dB up from the noise floor) looks to be around 250kHz, so that's roughly consistent with our expectations.
- FWIW, 22nV/rtHz is already about at the point where we can live with it since our trap-end filter should give at least 20dB by the secular frequency, getting us close to our 1nV/rtHz.
- A measurement on the lower range would be really nice if you have time though!

hartytp, 2018-04-13:
- Those spurs are potentially worrying if they're not just pickup in your wiring?
- It would be good to know: (a) do they correspond to any noise frequency generated by Zotino (e.g. the SMPS frequency) (b) are they affected by adding the screening cans to Zotino (c) are they affected by the proximity of Zotino to Kasli/any other local EMI source.
- The high-frequency noise peak is also potentially worrying if it's real. Could be something like the PSRR of the OpAmps dropping off a cliff as they run out of gain BW. Maybe another measurement artifact?

hartytp, 2018-04-13:
@gkasprow do you have the equipment required to take this kind of data with a slightly lower noise floor and in a really low-noise EMI environment?

gkasprow, 2018-04-13:
@hartytp I have access to the Faraday cage at the University. I can use Sampler to take some measurements. I can also use spectrum analyser but it works down to 9kHz.

hartytp, 2018-04-13:
@gkasprow Great!

I don't think Sampler is much use as the analog BW is only 250kHz, but I want to check the noise up to 10MHz if possible.

We don't need noise measurements below 9kHz IMHO.

gkasprow, 2018-04-13:
I have also access to sophisticated noise analyser, but it is heavily loaded and have to subscribe for the time slot :)

gkasprow, 2018-04-13:
So what the setup should look line?
The best one would be without any laptop in the room.
So we could use Kasli with programmed FLASH and Zotino.
Then IDC-> BNC adapter. Something else?

hartytp, 2018-04-13:
@gkasprow That sounds good to me. Use Kasli and a decent linear PSU. Measure the noise on a few channels with the outputs at +-FS and 0V. Use a IDC to BNC adapter for the measurement. Ideally take a noise spectrum from 10kHz to 10MHz.

closed 2018-04-18 by hartytp