sinara-hw/sinara #535: Sayma_AMC heatsink detached during shipping

jbqubit, 2018-03-27:
Heat sink attachment failed during shipping from WUT to UMD. 

[image]

[image]

hartytp, 2018-03-27:
Ouch! That must have taken quite a bang!

gkasprow, 2018-03-27:
Oh, it menas the board survived a lot.
After you re-attach, make sure the brown plastic things are plugged deeply. They will secure the latches.
It's quite possible they were not secured.

hartytp, 2018-03-28:
Joe, what's the action item here? Can we close this issue, or does something need to be done?

jbqubit, 2018-03-29:
@gkasprow Please close after brining to attention of technosystem or whoever will be assembling Sayma 2. The hooks on the brass-colored pins can be manually flared after insertion through the PCB to ensure good grip.

gkasprow, 2018-03-29:
This is what I meant by the brown plastic things that need to be plugged :)

jbqubit, 2018-03-29:
I've never seen brown plastic things in connection with the AMC heat sink. On the back side there's just an air gap. The pins are somewhat springy.  

[image]

jbqubit, 2018-03-30:
@gkasprow Please close if you've got the information you need from me.

closed 2018-04-02 by gkasprow