sinara-hw/sinara #536: Sayma: RF output

jbqubit, 2018-03-28:
@sbourdeauducq Using external 1.2 GHz clock, 180-deg splitter to drive DAC_CLK P and N. Loading the following as startup kernel. Scope shows ~50 mVpp square waves at 150 MHz. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]
        self.setattr_device("led0")

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma_out.output()
        for sawg in self.sawgs:
            delay(10*us)
            sawg.reset()

        delay(10*ms)

        for sawg in self.sawgs:
            delay(10*us)
            sawg.frequency0.set(0.)
            sawg.amplitude1.set(1.)
            sawg.frequency1.set(50*MHz)

        while True:
            delay(10*ms)
            self.ttl_sma_out.pulse(100*ns)

```

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+796.g5ca59467.dirty
Initializing SDRAM...
DQS initial delay: 93 taps
Write leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100110101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101110100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 93 taps
Write leveling: 85 90 111 110 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000001101111011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000010001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 144+-71 134+-74 112+-74 104+-71 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003889s]  INFO(runtime): software version 4.0.dev+796.g5ca59467
[     0.010157s]  INFO(runtime): gateware version 4.0.dev+796.g5ca59467.dirty
[     0.016971s]  INFO(runtime): log level set to INFO by default
[     0.022669s]  INFO(runtime): UART log level set to INFO by default
[     0.028809s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.502681s]  INFO(board_artiq::serwb): done.
[     0.505830s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+796.g5ca59467.dirty
[     0.513800s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1.513007s]  INFO(runtime): continuing boot
[     1.516027s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     1.522295s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 configuration...
[     1.540749s]  INFO(board_artiq::ad9154): AD9154-0 found
[     1.544674s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     1.627616s]  INFO(board_artiq::ad9154): AD9154-0 PRBS test
[     2.642609s]  INFO(board_artiq::ad9154): AD9154-0 found
[     2.646526s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     2.739782s]  INFO(board_artiq::ad9154): AD9154-1 found
[     2.743704s]  INFO(board_artiq::ad9154): AD9154-1 configuration...
[     2.826664s]  INFO(board_artiq::ad9154): AD9154-1 PRBS test
[     3.841667s]  INFO(board_artiq::ad9154): AD9154-1 found
[     3.845583s]  INFO(board_artiq::ad9154): AD9154-1 configuration...
[     3.944431s]  INFO(runtime): using MAC address 00-0a-35-03-1e-75
[     3.949304s]  INFO(runtime): using IP address 192.168.1.75
[     3.954853s] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry, using internal RTIO clock
[     3.966355s]  INFO(runtime::mgmt): management interface active
[     3.979535s]  INFO(runtime::session): accepting network sessions
[     3.993699s]  INFO(runtime::session): running startup kernel
[     4.024973s]  INFO(runtime::kern_hwreq): resetting RTIO
```

sbourdeauducq, 2018-03-29:
Scope where? Do not use any Allaki because, of course, they don't work (https://github.com/m-labs/artiq/issues/919).

sbourdeauducq, 2018-03-29:
And is the frequency on the scope changing when you change it in the kernel?

jbqubit, 2018-03-29:
I've looked both with and without Allaki. For example at J16 on motherboard. 

> And is the frequency on the scope changing when you change it in the kernel?

Output is occasionally periodic. More often aperiodic. Never sinusoidal. Doesn't seem to change in a way obviously related to frequency I command it to output.

sbourdeauducq, 2018-03-29:
Build it without SAWG and get the triangle waves to work first.

jbqubit, 2018-03-29:
Please clarify. If I build --without-sawg what do I do to see triangle wave?

sbourdeauducq, 2018-03-29:
Just boot the runtime without errors, no kernel needed.

jbqubit, 2018-03-30:
Counting channels from bottom (0) to top (7) of RTM. With Allaki installed. Running ```4.0.dev+796.g5ca59467``` with no SAWG, no kernel. Sawtooth is observed on all but one channel. Significant amplitude jitter observed on one channel. 

[image]
(above) 0,1,2,3

[image]
(above) 4,5,6,7

[image]
(above) 0 jitter

sbourdeauducq, 2018-03-30:
Can you measure the signals at the DACs (on the SMP connectors directly) on the channel that is not working?
Ditto the amplitude jitter, where is it coming from?

sbourdeauducq, 2018-03-30:
And I suspect your kernel may be incorrect; try the examples in the repository unmodified.

jbqubit, 2018-03-30:
>Can you measure the signals at the DACs (on the SMP connectors directly) 
>on the channel that is not working?

Sawtooth is fine on 1 coming out of SMP. Must be a problem with that Allaki. 

> Ditto the amplitude jitter, where is it coming from?

Looks like its coming from Allaki too. 

[image]
(above) for 0: trace1 and trace2 are SMP differential outputs; math is trace1 - trace2; eye and histogram are for falling edge (dashed orange line waveform view) of math

A new feature is now visible in 0. Glitch in second step.

jbqubit, 2018-03-30:
Upon reinstalling Allaki identical bad behavior in 0 and 1 reappear.

jbqubit, 2018-03-30:
>And I suspect your kernel may be incorrect; try the examples in the repository unmodified.

Using ```4.0.dev+796.g5ca59467``` from 3/26 with SAWG. With Allaki installed for all AFE. LED flashing OK. Running [sines.py](https://github.com/m-labs/artiq/blob/master/artiq/examples/sayma_standalone/repository/sines.py) as startup kernel. 

[image]
(above) 0123 with persistence; missing 1 due to Allaki error

[image]
(above) 4567 with persistence

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+796.g5ca59467.dirty
Initializing SDRAM...
DQS initial delay: 93 taps
Write leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111011010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 93 taps
Write leveling: 85 90 111 106 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000001111011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000100011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100101010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111100011010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 143+-71 134+-73 111+-74 102+-71 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003889s]  INFO(runtime): software version 4.0.dev+796.g5ca59467
[     0.010157s]  INFO(runtime): gateware version 4.0.dev+796.g5ca59467.dirty
[     0.016979s]  INFO(runtime): log level set to INFO by default
[     0.022671s]  INFO(runtime): UART log level set to INFO by default
[     0.028809s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.750550s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     0.908850s]  INFO(board_artiq::serwb): done.
[     0.911992s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+796.g5ca59467.dirty
[... cut ...]
[     4.754317s]  INFO(runtime): using MAC address 00-0a-35-03-1e-75
[     4.759190s]  INFO(runtime): using IP address 192.168.1.75
[     4.764746s] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry, using internal RTIO clock
[     4.776241s]  INFO(runtime::mgmt): management interface active
[     4.789421s]  INFO(runtime::session): accepting network sessions
[     4.803585s]  INFO(runtime::session): running startup kernel
[     4.832309s]  INFO(runtime::kern_hwreq): resetting RTIO
```

jbqubit, 2018-03-30:
@whitequark Can you see RF output on your board?

jbqubit, 2018-04-02:
@sbourdeauducq @jordens Do you see clean SAWG output on all Sayma channels? What I see is pretty messy. From sawtooth tests I guess DACs are nominally working fine.

hartytp, 2018-04-02:
@jbqubit what did you see on the sawtooth? Can you post scope screenshots? I saw a staircase pattern on the output, which I wasn't expecting.

jbqubit, 2018-04-02:
@hartytp See earlier in [this](https://github.com/sinara-hw/sinara/issues/536#issuecomment-377556508) Issue for sawtooth. The class that implements the ramp ```AD9154NoSAWG```. Its not clear to me how many steps to expect.

jbqubit, 2018-04-02:
@gkasprow Do you get RF output on your Sayma board running ARTIQ at WUT?

gkasprow, 2018-04-02:
@jbqubit We didn't run ARTIQ to produce RF output. The only RF we got on all boards was with Florent code.

jbqubit, 2018-04-12:
@jordens @whitequark @sbourdeauducq What do you see when running sines.py as startup kernel?

sbourdeauducq, 2018-04-13:
Last time I tested it, it would produce the expected output (barring various 1.8V, serwb, SDRAM, and JESD initialization problems; some having been addressed since).

jbqubit, 2018-04-17:
@sbourdeauducq What is expected voltage step size for sawtooth? Please test SAWG operation again. It looks like @whitequark is also having trouble. https://github.com/m-labs/artiq/issues/919#issuecomment-382104615

jbqubit, 2018-04-17:
@gkasprow Have you tried to get RF output on your Sayma using SAWG?

sbourdeauducq, 2018-04-18:
@gkasprow As you have noticed with Ethernet, HMC830 and the 1.8V bug, Sayma is very fragile and with a lot of board-to-board variation. It would definitely help to have more competent testers look into the massive amount of difficult bugs.

gkasprow, 2018-04-18:
Did you apply the 1.8V  fix?

gkasprow, 2018-04-18:
Sure, I'd love to help but need to get the board to my place.
1V8 issue was identified on the boards I have and was fixed.
It's complex board and there will always be issues, especially when you start production series, some bugs not present in single proto emerge.
The same with Ethernet - we identified the bug and fixed so your board works. Then Joe sent me a board that behaves differently and I'm struggling to understand it. The Ethernet works, but the LINK is off. When I plug another transceiver, the LINK is on, but Ethernet does not work. I do my best to understand this behaviour and I'm pretty sure that once I dig deep enough, will find a solution.
Good thing is that we have 10 sets of the board so various issues were identified and we can patch them and then implement on rev2. I'm convinced that next revision won't have many issues due to the fact that we identified them on rev1. If we produced only 2 pieces of rev1, most of the issues won't be identified and we would have to fight with them anyway.
I understand your point that you'd refer to focus on SW side of the project but the development we do is not usual. We do it for nearly 20 modules in parallel so response time of my team is not immediate.