sinara-hw/sinara #538: Sayma fix input/output grounding

hartytp, 2018-03-28:
https://github.com/sinara-hw/sinara/issues/499#issuecomment-377071799

gkasprow, 2018-08-10:
added 100k || 10pF to GND

marmeladapk, 2018-12-13:
@hartytp Shouldn't it be 0R || 10 pF?

hartytp, 2018-12-13:
@marmeladapk Yes.

In the original version we used 100k, to help break ground loops. However, we had some bad experiences with Sayma that made us to decide to replace the 100k with 0R.

The thinking is that the default configuration should be to have all grounds tied together. So long as we keep the pads, users can replace 0R with a higher resistance to remove ground currents, however they do that at their own risk!

marmeladapk, 2018-12-13:
@gkasprow So default configuration is grounded with a note on schematics, that you can unground it.

closed 2019-03-11 by gkasprow