sinara-hw/sinara #539: ARTIQ on Sayma at WUT

jbqubit, 2018-04-04:
@gkasprow @marmeladapk What's the status of running ARTIQ on Sayma at WUT? Can you get SAWG RF output? Can you run kernels interactively with Ethernet?

marmeladapk, 2018-04-06:
Right now we have problems with running ARTIQ on our board. Seems like there's a problem with flash, FPGA can't start from it. We'll investigate next week and I'll try to get it running. Also if we get your board we'll try to start it.

jbqubit, 2018-04-09:
Have you tried skipping the use of FLASH? Here’s what I’m doing when testing.  

https://github.com/jbqubit/sinara-testing/blob/master/sayma/tools/load_amcrtm.bash

https://github.com/jbqubit/sinara-testing/blob/master/sayma/tools/load_kernels.bash

gkasprow, 2018-04-11:
@jbqubit we are trying to make the board we got from you running.
 And the ARTIQ works, but we have issues with PHY that does not establish a link. It gets initialised from time to time but most of the time it does not respond.

gkasprow, 2018-04-11:
I found that the PHY responds but the MMC writes only first  register.

jbqubit, 2018-04-12:
There are dozens of ARTIQ "PHYs" on this board. What exactly do you see? @gkasprow I don't understand your last two comments. What does it mean for the MMC to write only the first register?

sbourdeauducq, 2018-04-13:
I guess he's talking about the Ethernet PHY. @gkasprow you can use the startup/idle kernels to work around the issue.

gkasprow, 2018-04-13:
@jbqubit it seems the MMC configures the Ethernet PHY correctly, but shows 0xFFF while reading the registers. However sometimes it shows correct config content. The funny thing is that on another board it works correctly. yet another strange thing is that I observe correct MDIO patterns on the MMC pin.
So I observe on the scope  i.e. that MMC reads 0x880, but it prints 0xFFFF. Sometimes it prints correct number. So it is neither HW issue nor something wrong with the code, because it works on other 2 boards...

jbqubit, 2018-04-18:
Is that

jbqubit, 2018-05-07:
@marmeladapk Can you get SAWG RF output?

marmeladapk, 2018-06-06:
@jbqubit Yes, today we checked that we have sines on output (using sines.py experiment from examples).

closed 2018-06-06 by marmeladapk

hartytp, 2018-06-06:
nice! Did you see https://github.com/m-labs/artiq/issues/1044

marmeladapk, 2018-06-06:
@hartytp Yes, and even better - over 300 MHz amplitude was oscillating from 0 to set value per ~2 seconds.

jordens, 2018-06-06:
@marmeladapk which versions (jesd204b and artiq) are those? and could you post scope screenshots including the corresponding experiment code?

hartytp, 2018-06-06:
@marmeladapk any init issues (JESD errors, serwb crashes, freezes)? Or, is that all good with the new init?

marmeladapk, 2018-06-06:
@jordens On it.

@hartytp No, only DACs sometimes takie 2-3 tries to properly initialise, but I haven't looked into that.

```
Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003882s]  INFO(runtime): software version 4.0.dev+1116.gc28fe471
[     0.010233s]  INFO(runtime): gateware version 4.0.dev+1110.g07d4145a
[     0.016601s]  INFO(runtime): log level set to INFO by default
[     0.022308s]  INFO(runtime): UART log level set to INFO by default
[     0.028456s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.062135s]  INFO(board_artiq::serwb): done.
[     0.065166s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     1.547208s]  INFO(board_artiq::serwb): 0 errors
[     1.550511s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     3.032560s]  INFO(board_artiq::serwb): 0 errors
[     3.035859s]  INFO(board_artiq::serwb): Wishbone test...
[     4.967727s]  INFO(board_artiq::serwb): 0 errors
[     4.971034s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.976583s]  INFO(board_artiq::serwb):   bitslip: 27
[     4.981619s]  INFO(board_artiq::serwb):   ready: 1
[     4.986395s]  INFO(board_artiq::serwb):   error: 0
[     4.991170s]  INFO(board_artiq::serwb): RTM serwb settings:
[     4.996736s]  INFO(board_artiq::serwb):   bitslip: 16
[     5.001772s]  INFO(board_artiq::serwb):   ready: 1
[     5.006547s]  INFO(board_artiq::serwb):   error: 0
[     5.011557s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1116.gc28fe471
[     5.018963s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     6.018006s]  INFO(runtime): continuing boot
[     6.021065s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     6.027133s]  INFO(board_artiq::hmc830_7043::hmc830): loading configuration...
[     6.034641s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[     6.040230s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for lock...
[     6.047035s]  INFO(board_artiq::hmc830_7043::hmc830):   ...locked
[     6.053080s]  INFO(board_artiq::hmc830_7043::hmc7043): enabling hmc7043
[     6.059955s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     6.066033s]  INFO(board_artiq::hmc830_7043::hmc7043): loading configuration...
[     6.075314s]  INFO(board_artiq::hmc830_7043::hmc7043):   ...done
[     6.080368s]  INFO(board_artiq::ad9154): AD9154-0 found
[     6.095507s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     6.108435s]  INFO(board_artiq::ad9154):   ...done
[     6.182324s]  INFO(board_artiq::ad9154): AD9154-0 running PRBS test...
[     7.188701s]  INFO(board_artiq::ad9154):   ...passed
[     7.192361s]  INFO(board_artiq::ad9154): AD9154-0 running STPL test...
[     7.199150s]  INFO(board_artiq::ad9154):   c0 errors: 0
[     7.204355s]  INFO(board_artiq::ad9154):   c1 errors: 0
[     7.209565s]  INFO(board_artiq::ad9154):   c2 errors: 0
[     7.214774s]  INFO(board_artiq::ad9154):   c3 errors: 0
[     7.219698s]  INFO(board_artiq::ad9154):   ...passed
[     7.224647s]  INFO(board_artiq::ad9154): AD9154-0 SYSREF scan:
[     7.504737s]  INFO(board_artiq::ad9154):   phase: 26, sync error: 481
[     8.159976s]  INFO(board_artiq::ad9154):   phase: 90, sync error: 482
[     8.165115s]  INFO(board_artiq::ad9154):   phase min: Some(26), phase max: Some(89)
[     8.172757s]  INFO(board_artiq::ad9154): AD9154-0 setting SYSREF phase to 88
[     8.190141s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     8.203033s]  INFO(board_artiq::ad9154):   ...done
[     8.277242s]  INFO(board_artiq::ad9154): AD9154-1 found                           
[     8.291440s]  INFO(board_artiq::ad9154): AD9154-1 initializing...                 
[     8.304292s]  INFO(board_artiq::ad9154):   ...done                                
[     8.381706s]  WARN(board_artiq::ad9154): AD9154-1 config attempt #0 failed (bad SYNC), retrying
[     8.399409s]  INFO(board_artiq::ad9154): AD9154-1 initializing...                 
[     8.412264s]  INFO(board_artiq::ad9154):   ...done
[     8.486140s]  INFO(board_artiq::ad9154): AD9154-1 running PRBS test...
[     9.492508s]  INFO(board_artiq::ad9154):   ...passed
[     9.496166s]  INFO(board_artiq::ad9154): AD9154-1 running STPL test...
[     9.502951s]  INFO(board_artiq::ad9154):   c0 errors: 0
[     9.508161s]  INFO(board_artiq::ad9154):   c1 errors: 0
[     9.513371s]  INFO(board_artiq::ad9154):   c2 errors: 0
[... 3 lines cut ...]
[     9.788214s]  INFO(board_artiq::ad9154):   phase: 24, sync error: 481
[    10.443470s]  INFO(board_artiq::ad9154):   phase: 88, sync error: 482
[    10.448607s]  INFO(board_artiq::ad9154):   phase min: Some(24), phase max: Some(87)
[    10.456248s]  INFO(board_artiq::ad9154): AD9154-1 setting SYSREF phase to 88
[    10.473635s]  INFO(board_artiq::ad9154): AD9154-1 initializing...
[    10.486490s]  INFO(board_artiq::ad9154):   ...done
[    10.560358s]  INFO(board_artiq::hmc542): card 0 channel 0 set to 4 dB
[    10.567602s]  INFO(board_artiq::hmc542): card 0 channel 1 set to 4 dB
[    10.574838s]  INFO(board_artiq::hmc542): card 1 channel 0 set to 4 dB
[    10.582074s]  INFO(board_artiq::hmc542): card 1 channel 1 set to 4 dB
[    10.589309s]  INFO(board_artiq::hmc542): card 2 channel 0 set to 4 dB
[    10.596544s]  INFO(board_artiq::hmc542): card 2 channel 1 set to 4 dB
[    10.603780s]  INFO(board_artiq::hmc542): card 3 channel 0 set to 4 dB
[    10.611015s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
[... cut ...]
[    10.630310s]  INFO(runtime::mgmt): management interface active
[    10.643619s]  INFO(runtime::session): accepting network sessions
[    10.657822s]  INFO(runtime::session): running startup kernel
[    10.662287s]  INFO(runtime::session): no startup kernel found
[    10.668028s]  INFO(runtime::session): no connection, starting idle kernel
[    10.674893s]  INFO(runtime::session): no idle kernel found
[    20.221818s]  INFO(runtime::session): new connection from 192.168.95.122:48926
[    20.259751s]  INFO(runtime::kern_hwreq): resetting RTIO
```