sinara-hw/sinara #540: PCA9548 failed to ack write address

hartytp, 2018-04-11:
One of my Kasli boards is now showing up this error on boot "panic at src/libcore/result.rs:906: cannot initialize Si5324: "PCA9548 failed to ack write address"

gkasprow, 2018-04-11:
Is it rev 1 board?
That revision does not have ESD protection for I2C so the mux can be easily damaged by discharge.
I2C has open drain ports without protection diodes.

hartytp, 2018-04-11:
Best guess is that this is related to the grounding issues; if Kasli floats up to a high voltage when powered but not grounded (e.g. no USB) then it could be damaged by handling quite easily...

marmeladapk, 2018-04-12:
@hartytp Could you look with scope on levels of I2C signals? IC13 could be damaged.

If there's a signal on USB I2C SDA and SCL test points (below JTAG connector) then IC13 works. 

You can test it with:

```
class dio_test(EnvExperiment):
    def build(self):
          self.setattr_device("core")
          self.setattr_device("i2c_switch1")

          self.i2c_switch1.set(7)  # 0 - 7
          # + some loop with catching exceptions.
```

hartytp, 2018-04-12:
> @hartytp Could you look with scope on levels of I2C signals? IC13 could be damaged.

Yes, I plan to do that.

> You can test it with:

Don't think I can. Without this switch ARTIQ can't program the Si5324, so it won't boot.

hartytp, 2018-04-12:
But, I will have a look with @jordens's Kasli I2C tool.

marmeladapk, 2018-04-12:
Ok, so then if IC13 works we expect signal on I2C SW SDA SCL test pads.

closed 2018-05-02 by hartytp