sinara-hw/sinara #541: Saya fails mem test with RTM connected

hartytp, 2018-04-13:
Moving from m-labs/artiq for now as this may be a hw issue.

@gkasprow:
- I remove the jumper on the programming connector to re-enable the RTM Exar chip, but left the 2V and 4V rails disconnected.
- Mem test still looks good, so the issue seems to be with something powered from one of those rails
- IIRC the current consumption from the 12V rail was over 2A before, it's now 1.2A. That seems like a high current draw for some DACs and ADCs that aren't enabled, doesn't it?

gkasprow, 2018-04-13:
@hartyp you can quickly check if this is problem with return current path.
Just make sure that 2V and 4V rails are disabled and connect roughly 5 Ohm load to the 12V to simulate missing 2V/4V load.
You can use input capacitor of the buck converter that generates 2/4V rails.
[image]

jbqubit, 2018-04-13:
On both my Sayma_AMC (s/n 7 and 8) with RTMs attached and powered memtest has failed only once out of about 60 restarts. I suspect this is a problem just with your board @hartytp.

hartytp, 2018-04-13:
> On both my Sayma_AMC (s/n 7 and 8) with RTMs attached and powered memtest has failed only once out of about 60 restarts. I suspect this is a problem just with your board @hartytp.

How so? Do you consider memory test failing 2% of the time normal behaviour? Would you accept that from your desktop PC? If this is working correctly you should not ever see errors. If it fails on your board at least sometimes then there is clearly a real issue here -- it's just worse on my board for some reason.

hartytp, 2018-04-13:
It needs to be understood and fixed. Obvious point, but add enough bugs together that only happen a few percent of the time and you end up with a system that rarely works at all.

hartytp, 2018-04-13:
@gkasprow the test you suggest (which I will do on Monday) will show us if there is a grounding (return current) issue with constant currents. It won't show us if there is some transient issue (e.g. if the DAC regulator starts up while the AMC is doing mem test and causes SI errors).

gkasprow, 2018-04-13:
Yes, but you have issue with DC load since the RTM FPGA is not loaded, don't you?

hartytp, 2018-04-14:
My worry was more that the MAC FPGA could be starting too early while the regulators on the RTM are still starting up.

gkasprow, 2018-04-14:
Maybe one can delay SDRAM init procedure?
Anyway booting the FPGA takes some time.
@wizath does the MMC control the FPGA boot process? I mean that it should let the FPGA configure after all the power rails are on.

hartytp, 2018-04-15:
@gkasprow ok next week I will add the 5R resistor to the RTM 12V. I will then try repopulating the soft start caps I removed and will add a 1s delay to the artiq firmware boot up.

If all else fails I will remove the inductors supplying power to the ADCs and see if that helps.

hartytp, 2018-04-15:
One request though @gkasprow : if I remember correctly, you checked the power integrity on Sayma amc carefully with artiq running but no RTM. Could you repeat this with rtm installed?

gkasprow, 2018-04-15:
I did PI simulations using Hyperlynx with fully loaded RTM. The return current practically does not flow below the FPGA.

hartytp, 2018-04-15:
@gkasprow ACK. So far, we've made a lot of different SI/PI measurements on this board, and all have shown that your layout is excellent. So, I don't expect this to be an issue.

But...

- There is some kind of issue with the SDRAM on ARTIQ when the RTM is connected, but no RTM gateware is loaded.
- The issue goes away when the DACs and ADCs are powered down, even if the RTM FPGA is powered
- It's hard to see how that could be caused by ARTIQ gateware/firmware (all transceivers are held in reset state, etc)
- So, the question is: what is the coupling between the AMC and RTM that causes this issue? The things I can think of are (let me know if you can think of any other potential AMC<->RTM couplings): a SI/PI issue; something to do with the MMC; or something to do with one of the signal lines connecting the AMC FPGA to the RTM.
- Double checking that there is no PI/ground noise issue would help us narrow this down.
- Once I've done the tests above, I'll also remove all coupling caps for signals connecting the AMC FPGA to the DACs/ADCs and repower the DACs + ADCs.

hartytp, 2018-04-16:
@gkasprow I soldered a 5R0 resistor across the 12V supply. Total 12V current draw is now 3.6A.

Mem tests still looks good, so it does not seem to be an issue with DC current draw from the DACs/ADCs. https://hastebin.com/ayakevulum.sql

hartytp, 2018-04-16:
Posting my AMC + RTM to @gkasprow for further investigation.

jordens, 2018-04-16:
Did that test with the signal caps removed happen?
I would also try to remove/disable code (jesd, serwb, etc) from the AMC and FPGA bitstreams and see whether that either triggers the behavior or prevents it.

hartytp, 2018-04-16:
@jordens nope, not yet. I think @gkasprow will try that once he receives the board (easier than helping me debug it remotely).

gkasprow, 2018-04-25:
@hartytp on your RTM and my AMC the SDRAM works:

``

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+822.g1ef673c2
Initializing SDRAM...
DQS initial delay: 90 taps
Write leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110010011000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000001100100010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111111101010100001010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101001010100000010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100101111
DQS initial delay: 90 taps
Write leveling: 68 78 86 92 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000001001100111110111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011111101010000011101011100100000001000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000001001101111011111011111011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111111011110110100101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000001010001010000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111110101100000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000100000000011110110111110111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101111111111111110011001001101000010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 125+-34 120+-42 104+-38 100+-38 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003888s]  INFO(runtime): software version 4.0.dev+822.g1ef673c2
[     0.010155s]  INFO(runtime): gateware version 4.0.dev+822.g1ef673c2
[     0.016425s]  INFO(runtime): log level set to INFO by default
[     0.022135s]  INFO(runtime): UART log level set to INFO by default
[     0.028287s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     2.721263s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     5.415939s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[     8.115100s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
[    10.814578s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
``

hartytp, 2018-04-25:
Two comments @gkasprow:
1. How reliably did it work? I found it would work sometimes, but not others.
2. Can you post the eye scan for just the AMC with the RTM attached, please? In the post above, there are some serious issues with the eye scan, such as dead points in the middle of the working region. This suggests that the SDRAM will not be reliable.

gkasprow, 2018-04-25:
next time it failed:

``

Bootloader CRC passed
Gateware ident 4.0.dev+822.g1ef673c2
Initializing SDRAM...
DQS initial delay: 88 taps
Write leveling scan:
Module 3:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111011101100000101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000101011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111010111010010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
0000000000000000000000000000000000000000000000000000000000000000000000000000001000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101110011011100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
0000000000000000000000000000000000000000000000000000000000000000000010011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101110100101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100001000111110001
DQS initial delay: 88 taps
Write leveling: 66 77 93 90 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000001000100010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011111011101101011001000000110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000100100000111110100110010111111011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101101010000011010110110000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000110000000101111111111100011011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111111111010111010100101110000001001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000010000100110011011111100111011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111111111111111011110010101001110000100100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 119+-27 105+-35 98+-31 85+-29 done
SDRAM initialized
Memory test failed (3/4458496 words incorrect)
Halting.

``

gkasprow, 2018-04-25:
OK, i got it. The eye scan boundaries are much worse with RTM attached.

gkasprow, 2018-04-25:
I turned off 2V rail - no effect.
Then I turned off 4V rail and let it start during memory test.
Half of the eye scans look good, another half look much worse.

gkasprow, 2018-04-25:
I got it.
When I power down the HMC7043, the effect disappears!

hartytp, 2018-04-25:
???? Any idea why?

gkasprow, 2018-04-25:
I will connect 5R resistor instead

gkasprow, 2018-04-25:
with 4.7 || 5,6R resistor connected at the LDO input, the eye scan looks good

hartytp, 2018-04-25:
Okay, how odd! The HMC7043 doesn't connect to the AMC apart from via the power supply, right?

gkasprow, 2018-04-25:
OK. I got it. Uninitialized HMC transfers broadband noise to the FPGA clock inputs.

hartytp, 2018-04-25:
Aaaah...I see. That's the clock on the AMC, right? So, that's causing problems even though all logic that's clocked from it should be held in the reset state during boot, right?

gkasprow, 2018-04-25:
Exactly. The clock is fed to both GTH and logic inputs. And it produces continuous spectrum up to 2GHz

gkasprow, 2018-04-25:
I hold it in reset state and it does not generate mess.

gkasprow, 2018-04-25:
reset does not help. I kept in reset state during first part of memory test, then I released the reset and the problem arrived

gkasprow, 2018-04-25:
So it seems that we have 2 solutions:
 - init the chip before SDRAM 
- add negator and pull up to hold it in reset state until RTM FPGA gets loaded. At the moment reset is connected to the GND.

gkasprow, 2018-04-25:
We do not observe this issue on another AMC+RTM set. It seems the HMC does not produce mess there or it has lower frequency.

hartytp, 2018-04-25:
Remind me where the HMC7043 clock connects to the AMC? Is it the DAC_REFCLK signal https://github.com/m-labs/migen/blob/81d0be313785c8d439dcacb2adbed403397c8e41/migen/build/platforms/sinara/sayma_amc.py#L202-L208?

hartytp, 2018-04-25:
If so, then @gkasprow can you want to try disabling the clock input buffer here https://github.com/m-labs/artiq/blob/fe689ab4f2c24a06d45e02ed5604c7bdf19e2f0b/artiq/gateware/targets/sayma_amc.py#L62 by setting the active-low clock enable pin to 1?

If that works then it seems we can fix this issue with a simple gateware fix that only enables the clock buffer input after the RTM has been successfully configured.

gkasprow, 2018-04-25:
@hartytp there are several clocks:
FPGA_ADC_SYSREF
AMC_MASTER_AUX_CLK
GTP_CLK2_IN
GTP_CLK1_IN
FPGA_DAC_SYSREF_P
So I'm not sure if disabling one really solves the issue.
It would be better to init this HMC chip first.

enjoy-digital, 2018-04-25:
@gkasprow: interesting! I'm wondering if this is not related to the issues we have with serwb (not the same behaviour between boards). The thing is that with the current architecture, we need serwb to be operational to be able to configure the HMC7043 over it... So if this is affecting serwb, the best thing to do would probably to have a hardware module inside the HMC7043 that at least make sure it's not generating noise...

marmeladapk, 2018-04-25:
@hartytp My idea was to configure HMC before memtest. This wouldn't even require gateware change.

hartytp, 2018-04-25:
HMC7043 Signal | AMC FPGA pin |
---|---|
FPGA_ADC_SYSREF | C11 B11|
AMC_MASTER_AUX_CLK  | AF10 AG10 |
GTP_CLK2_IN | P5 P6 | 
GTP_CLK1_IN | V5 V6|
FPGA_DAC_SYSREF_P | A10 B10|
FPGA_ADC_SYSREF_P | C11 B11|

hartytp, 2018-04-25:
So, all we have to worry about for the time being is dac_refclk and dac_sysref.

dac_refclk can be held in an disable state until the rtm is configured using the fix I suggested above.

hartytp, 2018-04-25:
hmm...this could potentially explain **a lot**. As we've seen before, FPGAs don't like having garbage sprayed into their inputs!

hartytp, 2018-04-25:
Anyway, great catch @gkasprow and @marmeladapk!

gkasprow, 2018-04-25:
> @hartytp <https://github.com/hartytp>: yes we should test that. If that's
> work that's an easy workaround until we have a better solution.

Anyway I will add reset circuit in the final revision that will keep HMC in
reset state until RTM FPGA gets loaded.

enjoy-digital, 2018-04-26:
What's the plan for the actual boards for this point? @gkasprow / @marmeladapk  are you going to try @hartytp's idea (do a test with clock input buffer disabled?). I'd really like this to be in a clear state before doing any others sdram/serwb investigation. 

@gkasprow: as i said, with the current architecture, keeping it in reset state until RTM FPGA gets loaded is not enough. If you do this test, it will allow us to know if we have to change the architecture (implement HMC7043 initialization directly in the RTM, or at least a partial config to disable the clocks) or we can simply disable the clock buffers until HMC7043 is initialized. This second option is easier to implement.

hartytp, 2018-04-26:
> What's the plan for the actual boards for this point? @gkasprow / @marmeladapk are you going to try @hartytp's idea (do a test with clock input buffer disabled?). I'd really like this to be in a clear state before doing any others sdram/serwb investigation.

I think that's a very good idea!

gkasprow, 2018-04-26:
We will try it, but:
- we have a set of 3U boards to test, which is of highest priority
- we have to test HMC830 with ARTIQ which has higher priority
- whole next week the university is closed due to holidays.
- I want to dig in the Erhernet PHY and understand its behaviour on all boards I have. I have a few points to check.

hartytp, 2018-04-26:
@gkasprow ACK.

Would it help if I made the changes, built the gateware and sent you binaries?

gkasprow, 2018-04-26:
Yes.

hartytp, 2018-04-26:
Okay, I'll do that now.

hartytp, 2018-04-26:
I'll include the updated HMC830 regs as well.

gkasprow, 2018-04-26:
This will help us a lot.

hartytp, 2018-04-26:
Building this atm https://github.com/hartytp/artiq/tree/hmc830_test

@gkasprow are you happy that the HMC830 register writes are the same as the ones in your FORTH code? https://github.com/hartytp/artiq/blob/a1aba3c8448c471623da5c88408dc7cb43335c42/artiq/firmware/libboard_artiq/hmc830_7043.rs#L33-L53

@enjoy-digital are you happy with the way I disabled the HMC7043 clock inputs?

enjoy-digital, 2018-04-26:
@hartytp: yes that seems fine.

gkasprow, 2018-04-26:
@hartytp now it works, the eye diagrams look lovely

hartytp, 2018-04-26:
Great! Thanks for tracking down this disgusting issue so quickly ;)

If you have a chance, could you confirm whether eye diagrams now look good on all Sayma with attached RTM, please? Also, could you post the UART output, please?

@gkasprow Final request (feel free to say "no" if you don't have time) but could you try using OpenOCD to load the RTM (I can tell you how to do this) and give me the UART output. It would be great to see if serwb now works reliably (but maybe that's hoping for too much).

@enjoy-digital how do you want to integrate the clock buffer enable into ARTIQ?

gkasprow, 2018-04-26:
And the HMC locks with ARTIQ and my values!

gkasprow, 2018-04-26:
__  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+861.gc8206d62
Initializing SDRAM...
DQS initial delay: 97 taps
Write leveling scan:
Module 3:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
0000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 97 taps
Write leveling: 77 87 103 102 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000110101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 171+-76 159+-82 148+-79 140+-78 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003890s]  INFO(runtime): software version 4.0.dev+861.gc8206d62
[     0.010158s]  INFO(runtime): gateware version 4.0.dev+861.gc8206d62
[     0.016424s]  INFO(runtime): log level set to INFO by default
[     0.022137s]  INFO(runtime): UART log level set to INFO by default
[     0.028291s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     2.791764s]  INFO(board_artiq::serwb): done.
[     2.794803s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     3.800450s]  INFO(board_artiq::serwb): 0 errors
[     3.803764s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     4.809411s]  INFO(board_artiq::serwb): 0 errors
[     4.812723s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.818278s]  INFO(board_artiq::serwb):   delay_min_found: 1
[     4.823923s]  INFO(board_artiq::serwb):   delay_min: 34
[     4.829132s]  INFO(board_artiq::serwb):   delay_max_found: 1
[     4.834777s]  INFO(board_artiq::serwb):   delay_max: 194
[     4.840073s]  INFO(board_artiq::serwb):   delay: 114
[     4.845022s]  INFO(board_artiq::serwb):   bitslip: 36
[... cut ...]
[     5.942355s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 configuration...
[     5.960467s]  INFO(board_artiq::ad9154): AD9154-0 found
[     5.964393s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     5.976108s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #0 failed (AD9154 no sync lock), retrying
[     5.994730s]  INFO(board_artiq::ad9154): AD9154-0 found
[     5.998649s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     6.010343s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #1 failed (AD9154 no sync lock), retrying

hartytp, 2018-04-26:
@gkasprow whooooo!!!!!!!!!!!!!!!

I'll dig out my nominally optimized HMC830 register writes and send them to you so we can see if they work as well.

gkasprow, 2018-04-26:
Look [here](https://www.dropbox.com/s/relisxduo3z4iyb/2018-04-26%2016.14.09.mp4?dl=0)

hartytp, 2018-04-26:
> @hartytp: we should have a register to control the input buffers and enable it by the firmware at ad9154 initialisation.

@enjoy-digital are you happy to do that? I can probably do that myself, but you know the code better than me that you'll probably be much faster...

[7 more replies]