sinara-hw/sinara #542: MMC serial terminal on Linux

jbqubit, 2018-04-13:
The fourth serial interface exposed by Sayma over USB is MMC Console. How to use it is discussed [here](https://github.com/sinara-hw/sinara/issues/358#issuecomment-380463815) and [here](https://github.com/sinara-hw/sinara/issues/521#issuecomment-371195293). The MMC terminal works in FlashMagic and Terminalbpp, both on Windows. But I can't get this to work under linux. 

Here's what I'm doing.  
- 115200 8N1
- no hardware handshaking = 
- minicom on Linux configured as follows (`minicom -D /dev/ttyUSB4`)

```
$ cat ~/.minirc.dfl
# Machine-generated file - use setup menu in minicom to change parameters.
pu port             /dev/ttyUSB4
pu rtscts           No
pu linewrap         Yes
pu addcarreturn     Yes

```

What I observe: nothing in minicom.

gkasprow, 2018-04-13:
Are U sure that the MMC did not enter the bootloader mode ?

hartytp, 2018-04-13:
@jbqubit As @gkasprow says, this doesn't work under linux for some reason as the MMC enters bootloader mode. I removed a jumper wire to get it to work with Linux. Which is a PITA, as I now have to solder it to refresh and desolder it connect to the terminal.

hartytp, 2018-04-14:
Is that necessary? Can't the firmware be made more robust?

Cant we do something like use a magic string sent over the uart to restart into boot loader so the uart works by default? It would also be great if firmware was done using a simple python program rather than flash magic.

gkasprow, 2018-04-14:
This bootloader is carved in stone and cannot be changed.
We can however write own bootloader or use one developed for Open MMC and disconnect lines responsible for programming.
Another idea is to add some logic that would trigger the update sequence only with certain IO state on FTDI chip

hartytp, 2018-04-14:
@gkasprow I don't mind too much how we do this, but I think it is important to have flashing robust and automated. I.e.not relying on users to toggle dip switches. Id also like a simple command line script I can run  to do this like the artiq_flash script, which works on both windows and Linux.

gkasprow, 2018-04-14:
This can be done with Open MMC bootloader developed by LNLS. We can add the switch as temporary / emergency solution and implement automated approach.

jbqubit, 2018-04-16:
@gkasprow Have you tried the Open MMC bootloader developed by LNLS?

sbourdeauducq, 2018-04-24:
> Is that necessary?

This has to do with the status of the serial port control lines. It is possible to make it work under Linux without touching the hardware: in another issue, I posted a hacky C program that resets the control lines and makes the Sayma not enter the bootloader. It is possible to integrate it into a serial port terminal program, which would make using the MMC console relatively seamless.

> It would also be great if firmware was done using a simple python program rather than flash magic.

Sure.

You can also use OpenOCD and a JTAG cable, though they are not very reliable IME.

sbourdeauducq, 2018-04-24:
```
#include <sys/ioctl.h> //ioctl() call defenitions
       #include <sys/types.h>
       #include <sys/stat.h>
       #include <fcntl.h>
main()
{
           int fd;
              fd = open("/dev/ttyUSB_sayma-3_2",O_RDWR | O_NOCTTY );//Open Serial Port
                
                 int RTS_flag;
                    RTS_flag = TIOCM_RTS;
                       /*ioctl(fd,TIOCMBIS,&RTS_flag);//Set RTS pin
                          getchar();*/
                             ioctl(fd,TIOCMBIC,&RTS_flag);//Clear RTS pin
                                close(fd);
}
```