sinara-hw/sinara #546: RTM with SFP & QSFP - request for comments

gkasprow, 2018-05-14:
I'm designing a [RTM](https://github.com/gkasprow/RTM_4SFP_3QSFP/wiki) module for CERN that is compatible with both Sayma AMC and AFCZ (ZynQ US+).
The schematics and initial placement are done.
If there are any wishes, I can take them into account.
Schematics are here
[RTM_4SFP_3QSFP.PDF](https://github.com/sinara-hw/sinara/files/2002283/RTM_4SFP_3QSFP.PDF)

jbqubit, 2018-05-17:
Do you have a link for AFCZ?

Does it make sense to add the tiny uP to this board that automatically handles the range of SFP connectors that you've battled with for Sayma_AMC?

gkasprow, 2018-05-17:
Here is AFCZ [repo](https://github.com/gkasprow/AFCZ/wiki). The Xpedition project is too big for github so I could upload only viewer and pdf files .
It is going to be controlled by Sayma or AFCZ so this can be handled by MMC easily.
AFCZ is in production. It is compatible with Sayma RTM.

hartytp, 2018-05-24:
Nice project @gkasprow.

Do keep us posted with how you get on with it.

Closing this issue, as it seems that there are no ARTIQ-specific requirements for this board.

closed 2018-05-24 by hartytp