sinara-hw/sinara #547: EEM front panels washer cutout.

jordens, 2018-05-15:
The cutouts that receive the rectangular snap-in grey plastic "captive" washers for the panel mounting screws were too narrow on the older boards. The washer was squeezing the screw and not allowing it to move freely.
More recent ones seem to be fixed. Has that fix been applied to all boards?

gkasprow, 2018-05-22:
Did you observe it only with first revision or later as well?

jordens, 2018-05-22:
The cutout was too small for all board I received until a month ago. In the last batch there were still some panels with bad cutouts but those may have been from older runs. I just wanted to make sure all panels are updated.
As a sidenote: receiving 8HP panels with 4HP handles (like for the last Kasli batch) is disappointing.

jordens, 2018-06-06:
This issue was moved to sinara-hw/meta#3

closed 2018-06-06 by jordens