sinara-hw/sinara #548: EEM power supply

gkasprow, 2018-05-22:
Today we finally collected several modules and placed them to single 19" chassis.
We need to make some quick Artiq demo and here are some thoughts:
- the chassis is missing rear panel. Ventilation supplied by Schroff does not work at all without rear panel.
@jbqubit tested independent cooling tray with open-frame 19" 3U rack. But for table-top applications, fully enclosed rack is better and we want to use 12V cooling fans embedded into the chassis .
- the supply delivered from external SMPS works, but there are grounding issues and for various reasons I'd like to have it all enclosed in one box.
So I see two solutions:
- attach the SMPS module together with IEC inlet to the rear panel. This is enclosure specific.
- make simple 3U module with SMPS, several DC jacks and trivial temperature controller (can be based on NE555 or tiny Arduino) that   stabilizes the temperature inside. This takes precious space but in case of 19" racks is not that critical.
Here is some quick sketch:
[image]
We already discussed this and decided to go for stand-alone SMPS. Then grounding issue emerged. 
I'd like to know your opinion on that.

hartytp, 2018-05-22:
> the chassis is missing rear panel. Ventilation supplied by Schroff does not work at all without rear panel. @jbqubit tested independent cooling tray with open-frame 19" 3U rack. But for table-top [...]

For this kind of application, I'd use the RatioPack Pro Air (or whatever it's called). It's a bit expensive and the ventilation isn't totally uniform across the slots, but it seems to work fine for EEMs. FWIW, I've been using the 24V fan options and running the fans at 12V. That seems to work fine.

hartytp, 2018-05-22:
> make simple 3U module with SMPS, several DC jacks and trivial temperature controller (can be based on NE555 or tiny Arduino) that stabilizes the temperature inside. This takes precious space but in case of 19" racks is not that critical.

- I'm not sure how useful temperature control of a single point is, since the loads will be very non-uniform across the rack. You'd also need to do all the mechanics for the fans and fan connections, which sounds like a lot of work. Generally, I feel that this is straying into "reinventing uTCA" territory and worry that this could turn into a huge time sink. I don't think I'd bother using it.
- A decent, relatively low cost 12V SMPS with proper grounding sounds like a good thing -- I'd probably use a few. Although, I'd be equally happy using a wall-wart if we make sure that Kasli's input power ground connects to the FP, and that the FP connects to the 19'' chassis. I'd then use a chassis with a grounding stud (most of them have that) and connect that to a suitable earth connection. Note that the current situation with Kasli is not really any different to using a KC705 or any other eval board that's powered by a wall wart.
- If you do make this PSU unit, how about adding some internal ATX-type connections that could be used to connect to an internal ATX connector on Kasli to supply 12V (this was discussed on another issue). Also, please make sure that all mains voltage is fully enclosed for safety reasons.

gkasprow, 2018-05-22:
@hartytp we have RatioPack Pro Air in the lab.

gkasprow, 2018-05-22:
I think about simple temperature control only to limit noise in my lab and make sure I won't fry the EEMs when operated with reduce air flow.

hartytp, 2018-05-22:
> I think about simple temperature control only to limit noise in my lab 

Currently, I'm not so worried about this. I've tried to design things so that few degree temperature variations won't be an issue. For me, the only place where I really care about temperature is Zotino, which is why we added the local temperature regulation. 

I'm not worried about Urukul, since we use a feedback loop to control the amplitude (laser drifts are worse). I'm not worried about Samper, since the temp co is low by design.

So, I'm not overly worried about this (and, in any case, it can easily be retrofitted to a ratioPack pro Air unit with a simple arduino-controlled SMPS to supply the fans). Others may feel differently.

> and make sure I won't fry the EEMs when operated with reduce air flow.

For that, would it be easier to give your SMPS a header that can connect to a thermal switch? Then you can glue a thermal switch into your rack to interlock against over temperature problems. Seems simpler than active control, and can be replaced with a jumper if users don't want to bother with it.

gkasprow, 2018-05-22:
@hartytp good idea with a thermal switch. Single resistor shorted by the thermal switch would do the job.

hartytp, 2018-05-22:
I still think it's going to be hard to protect eems with a thermal switch at the top of the rack since the heat sources are so localised.

Probably better to add thermal cut outs to all high power eems (iirc we did this on Urukul already)

jordens, 2018-05-22:
Even on Urukul the thermometer is quite far away from the hottest place, the gradients from hot spots to sensor are large (20K), and the threshold is high given the gradients (80C).

dhslichter, 2018-05-22:
@gkasprow what is the current plan for the grounding issue that we settled on?

jordens, 2018-05-23:
Hmm. If the impetus for designing this is the grounding issue, can't we just solve that by giving people who are unsure about it a [class I](https://media.digikey.com/pdf/Data%20Sheets/Mean%20Well%20PDF's/GSM90A_Series_Ds.pdf) supply and supply those who want to bear the risks and joys of isolated SMPSs with the corresponding [class II](http://www.meanwell.com/Upload/PDF/GSM90B/GSM90B-SPEC.PDF)?

I'd decouple the cooling and thermal design from the choice of power supply.

dhslichter, 2018-05-23:
@jordens IIRC there was some desire to have a power supply internal to the crate, so you just have an IEC connector and aren't mucking around with external bricks -- potentially more "idiot proof".  https://github.com/sinara-hw/sinara/issues/499#issuecomment-380611528 https://github.com/sinara-hw/sinara/issues/315#issuecomment-373941841.  Around and around the mulberry bush we go!  

My two cents: either we use external supplies as proposed, or we bolt one to the inside wall of the chassis with an IEC C14 on the back.  I don't like the idea of using Eurocard power supplies which take wall voltage in on a front panel as much.

> I'd decouple the cooling and thermal design from the choice of power supply.
I think this seems sensible, but if the power supply is going to be inside the crate one probably has to consider its thermal impacts and cooling needs.

gkasprow, 2018-05-23:
@dhslichter the idea with SMPS attached to the rear panel seem to be the simplest one and we will go for it. It is also low cost, no panel needed and no wasted precious place. The SMPS I use have 90..93% efficiency so they don't need special cooling. Connection between IEC grounding pin and chassis solves the grounding issue and EMI filter embedded into the IEC socket cuts the interferences that would come from mains.

dhslichter, 2018-05-23:
@gkasprow sounds good to me -- simple, cheap, low engineering, lots of SMPS to choose from, reliable grounding.  People can always make their own version of power supplies (external brick with barrel, or front panel, or whatever) if they are so inclined.

closed 2018-07-11 by gkasprow