sinara-hw/sinara #549: [RFC] use EMC shielded front panels for EEMs

jordens, 2018-05-24:
They are a about 6-8 € each (including gaskets) and would help when putting EEMs into shielded crates (~130 €). Don't know about secondary effects on the assembly costs though. Needs double checking that the lip doesn't collide with the RJ45/SMA connectors.

jordens, 2018-05-28:
Pretty sure that the left lip (underside) will touch/collide with the SMAs on Urukul.

gkasprow, 2018-05-28:
They should fit - of course the THT pins will need to be trimmed.
[image]

jordens, 2018-05-28:
That's one solution.
Also that washer under the screw doesn't exist on the boards I have. Might not be needed but it affects clearance.

jordens, 2018-06-06:
This issue was moved to sinara-hw/meta#2

closed 2018-06-06 by jordens