sinara-hw/sinara #550: Sayma_AMC Si5324 CLKINn VICM missing

jbqubit, 2018-05-29:
Si5324 data sheet Table 2 calls for CLKINn to have Input Common Mode Voltage (VICM) of 1.1 to 1.95 V DC and input impedance of 50 Ohm. Sayma AMC v1 neglects both these requirements for both CKIN1 and CKIN2. In Si5324 example (Figure 3) a voltage divider sets VICM to 1.3 V offset. Why don't we do this? It may have an adverse effect on both external clocking and clock recovery. 

This came to my attention while looking at how to supply an external clock to Sayma via J24/J25.

closed 2018-05-30 by hartytp

jbqubit, 2018-05-30:
The data sheet is explicit about the requirement for external biasing. In both Table 2, the associated figure and I’m Figure 3.

klickverbot, 2018-05-30:
The resistors in figure 3 (datasheet rev 1.1) provide DC termination for the clock source, not the chip – or is that not what you are referring to?

hartytp, 2018-05-30:
Joe, I think table 2 is a bit misleading. It tells you what voltages must be there, but doesn't tell you anything about the internal biasing of the chip. My reading is that it is only relevant for DC-coupled signals, like the DC coupled LVPECL in the figure you reference.

As I said, look at what the eval board does for this -- it doesn't use any external biasing components. We've spent a lot of time playing with this chip with AC coupled signals with no problems.

jbqubit, 2018-05-30:
Found a the spec sheet for the [whole family](https://www.silabs.com/documents/public/reference-manuals/si53xx-reference-manual.pdf) of Si chips including Si5324. 

> Input Clock Buffers
Any-Frequency Precision Clock devices provide differential inputs for the CKINn clock inputs. These inputs are internally biased to a common mode voltage and can be driven by either a single-ended or differential source.

This language is unambiguous about internal biasing. :)