sinara-hw/sinara #551: Sayma sawtooth glitches

jbqubit, 2018-06-01:
As requested https://github.com/m-labs/artiq/issues/1022#issuecomment-393850422 differential measurement of sawtooth. 20180530 HMC830 NoSAWG UpdatedJESD 38b51282226f9feb

CH1 and CH2 are externally terminated 50 Ohm. Observe surprising glitches near end of range. Glitches are not coincident in time on differential channels so seems more like a hardware problem than a gateware problem. 

[image]
[image]

jbqubit, 2018-06-01:
This is not a differential probe. Using two channels on scope.

jordens, 2018-06-01:
The fact that there is about 4ns shift between them is.

hartytp, 2018-06-06:
fixed by latest JESD204b package. Thanks @enjoy-digital, @jordens and @larsc

closed 2018-06-06 by hartytp

jbqubit, 2018-06-06:
I didn't this fix work. See. https://github.com/sinara-hw/sinara/issues/557#issuecomment-394902405

jbqubit, 2018-06-06:
I still see a glitch as posted in #557.

hartytp, 2018-06-06:
If we're talking about the same thing, I was interpreting that as a reflection/SI issue after the sharp rising edge, not a glitch.

jbqubit, 2018-06-06:
To my eye glitches in this Issue look about the same as reported last night in #557. Glitches are not coincident in time on differential channels neither in this post nor in #557. I don't think this is a problem.

hartytp, 2018-06-06:
@jbqubit before the recent JESD204B fix, I saw large glitches on the sawtooth generator like https://pasteboard.co/HnSYE20.jpg See https://irclog.whitequark.org/m-labs/2018-06-01 

This wasn't present on all channels and (IIRC) changed from load to load. That's now gone.

jbqubit, 2018-06-06:
Ok. Let’s keep this closed.