sinara-hw/sinara #552: Sayma BaseMod no output

jbqubit, 2018-06-01:
Using build 20180530 HMC830 SAWG updatedJESD204B 38b51282226f9. I see no RF output from BaseMod. Things I've checked. 
- swapping BaseMod boards
- different Sayma DAC channels
- confirmed that differential sinusoidal signals are appearing on N and P outputs
- powering down everything and reloading firmware and .bit
- confirmed gateware and firmware versions match

It's not easy to get scope probe on BaseMod output as its on the down-facing side. @sbourdeauducq, please check that you see RF output from BaseMod with a current build using SAWG. 

```
class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)

        for sawg in self.sawgs:
            sawg.reset()
            delay(100*ns)
            sawg.amplitude1.set(.5)
            sawg.frequency0.set(49*MHz)
            delay(100*ns)

        while True:
            delay(0.5*ms)
            self.ttl_sma_out.pulse(0.6*ms)
```

sbourdeauducq, 2018-06-01:
> @sbourdeauducq, please check that you see RF output from BaseMod with a current build using SAWG.

You tested that yourself yesterday. The ``sawg.reset()`` changes I made did not break it - I tested it on your board. Nothing else has changed AFAICT.

jbqubit, 2018-06-01:
So far I've not seen output from BaseMod. All my tests have been on SMP N/P differential from Sayma_RTM motherboard. @sbourdeauducq You confirm that you saw output with BaseMod prior to shipping? Trying to track down what might be different.

sbourdeauducq, 2018-06-02:
yes, and even after.

sbourdeauducq, 2018-06-02:
https://github.com/m-labs/artiq/commit/d4f074b1e162c80f3440fbdd6ea2b264bf72efb2
no problem since

sbourdeauducq, 2018-06-03:
Tested this again on master (d686d3309), still works.

jbqubit, 2018-06-04:
Thanks for testing again @sbourdeauducq. I can now see RF output from a BaseMod (s/n 20171115a) on all 8 channels on Sayma at UMD. I received a package with additional BaseMod from Warsaw last week and find that none of them generate RF output. Only the board which I've had for many months now at UMD (s/n 20171115a) generates RF output. Here's specifically what I tested. 

- BaseMod s/n 20171115a in all four AFE slots. RF output confirmed from all 8 channels.
- for BaseMod without s/n (received from WUT last week), none produce RF output in AFE Slot 3. 
- for BaseMod without s/n, I see no RF output for spot tests in other AFE Slots 

@gkasprow What might be different about this new batch of AFE?

closed 2018-06-04 by jbqubit