sinara-hw/sinara #553: Sayma at UMD lost Ethernet

jbqubit, 2018-06-01:
After applying TRST white wire patch https://github.com/m-labs/artiq/issues/1021 to Sayma_AMC shipped from HK to Duke to UMD (s/n TS190717-6) I was able to get reliable Ethernet operation. < 1% dropped packets. Good.

This afternoon we started to occasionally find that the board no longer responded to ping. On several of these instances unplugging power, Ethernet and SATA from the FMC-SATA dongle (made by WUT), then connecting them again fixed Ethernet. However, now we no longer have functioning Ethernet at all. Not sure if the Ethernet-SATA dongle is broken or something else. There appears to be no damage. LEDs are illuminated as earlier today. 

Perhaps related we did see twice this afternoon the current consumption of Sayma increase to 3.8 A (12 V) from it's usual value of 2.8 A. Cycling the power on the board or reprogramming .bit via JTAG restored 2.8 A state. Not sure if this is related to Ethernet issue.

@trxw will continue testing using default kernel.

gkasprow, 2018-06-02:
Is the LINK LED on ?

jbqubit, 2018-06-02:
Yes, it’s illuminated on the switch.

gkasprow, 2018-06-02:
OK, so disconnect the SATA cable - it should go off. It also means that the SATA, SFP are fine and there is some other reason why it does not work.

jbqubit, 2018-06-04:
>OK, so disconnect the SATA cable - it should go off.

When I disconnect the SATA cable the LED LD20 goes off. https://github.com/sinara-hw/sinara/wiki/Sayma#front-panel-leds

I tried another cycle of unplugging everything from the SATA-SFP adapter. Then reassembling again. Ping now works! I've tried all permutations of order of connection. All seem to work. I've tried reloading the build from last week when I first discovered the Ethernet problem. It works fine. So it doesn't seem that whatever was going on was related to software/gateware. 

Will post again to this Issue if I see this happen again. Closing for now. 

@gkasprow What's status of Ethernet in uTCA crate?

closed 2018-06-04 by jbqubit

marmeladapk, 2018-06-04:
@jbqubit https://github.com/sinara-hw/sinara/issues/483#issuecomment-389901190

jbqubit, 2018-06-04:
Watching this SATA-SFP +SFP-to-Ethernet-RJ45 connector I see a pattern. This has reproduced about 10 times. 
- power off Sayma
- power on Sayma
- link LED on Sayma illuminates (LD20), no Link LED on Ethernet switch, no ping of Sayma
- remove SATA cable and power microUSB on SATA-SFP
- connect microUSB power on SATA-SFP
- link light on Ethernet switch illuminates
- connect SATA cable
- can pin Sayma

gkasprow, 2018-06-05:
it looks like SFP module failed with Ethernet switch negotiation.
It negotiates th link on its both sides.
Which SFP module do you use?
I can implement SFP module reboot once the link status changes.

jbqubit, 2018-06-06:
The module is a SFP-RJ-45 I got from M-Labs in HK last week. 

```
1Gb SFP RJ-45 Module
453156-001
Spares PN: 453578-001
HP
```