sinara-hw/sinara #555: Sayma: new batch of BaseMod v0.1 don't work

jbqubit, 2018-06-04:
Last week I received a package with additional BaseMod from Warsaw. None of them transmit RF output. The single older BaseMod AFE that I've had for months now works fine. See #552. 

[image]

jbqubit, 2018-06-04:
[image]

gkasprow, 2018-06-05:
they probably came from other production batch

gkasprow, 2018-06-05:
@jbqubit 
I looked at remaining boards from same batch and it seems I found what is going on. Soldering of the 542 attenuator look suspicious. So please attach short wire at the input of the attenuator chip and look at it with scope. If you see a signal then use sharp soldering tip and fix joints on the sides of the attenuator. You don't need tin, just soldering flux. I'm pretty sure I tested all boards once they arrived, but it was long time ago. It could also happen that they were working and soldering joints degraded after some time.

jbqubit, 2018-06-05:
Signal is definitely going into attenuator. I’ll try reflowing the solder.

gkasprow, 2018-06-06:
@jbqubit 
It is not the attenuator.
By chance there were assembled 0R resistors that should be DNP.
They are marked DNP as a parameter called Moutned which was not exported to Pick and Place file.
There should be variant created.
In the first batch I removed these resistors manually.

gkasprow, 2018-06-06:
So when you compare the board that works with board that does not, you will notice 4 SMD resistors close to the SMA connectors.
Get rid of R3A, R3B, R4A, R4B
[image]

jbqubit, 2018-06-06:
I had tried reflow and monitoring inputs to attenuator. All looked OK so I was stumped. This makes sense now. I'll remove DNP and report what I see.

gkasprow, 2018-06-06:
It works in case of boards from same batch.