sinara-hw/sinara #556: HMC7043 output FETs

hartytp, 2018-06-05:
To try to pin this down further, looked on a scope during the init.

With the new init sequence, I no longer see any broad band noise but, as expected, the HMC7043 turns on briefly at the wrong frequency before shutting down.

Next, I altered the init sequence so that the HMC7043 input starts off being the clock mezzanine SMPs until after it has been muted. Here, the HMC7043 output oscillated with broadband noise. So, it seems that it is important to keep the HMC7043 in reset until it has a valid clock at its input.

I'm not sure if having the wrong frequency output (but not just random noise) for a brief period between RESET being deasserted and mute being programmed via SPI is an issue. If it is then we will need to add FET switches to shutdown the HMC7043 outputs.

@gkasprow what do you think? My feeling is that we should add FETs to the HMC7043 outputs going to the FPGA.

gkasprow, 2018-06-05:
What about keeping it in reset state until the input clock arrives?

hartytp, 2018-06-05:
I do that now.

However, when I release RESET, it starts up with the outputs enabled and the dividers at the wrong value.

gkasprow, 2018-06-05:
How much time do you need it to get configured? Can you afford temporary noise between releasing the reset and configuration? This won't break SDRAM because this is already resolved by input clock buffer. 
We can add FET that biases the PECL outputs in next revision.

hartytp, 2018-06-05:
> How much time do you need it to get configured?

Some us.

> Can you afford temporary noise between releasing the reset and configuration?

Well, now it's not noise, as much as RF at the wrong frequency. Not sure if that matters though. Given how much pain this has caused us, I'm inclined to be cautious about this!

> We can add FET that biases the PECL outputs in next revision.

That's my thinking. Probably only needed for the clock lines that go the the FPGA.

gkasprow, 2018-06-05:
OK, added to the todo list

hartytp, 2018-06-05:
@gkasprow note that I think we need independent lines for the HMC7043 and Si5324, since we may want to init the Si5324 while the HMC7043 is held in the reset state (e.g. if the Si5324 is used to provide the HMC830 reference clock). So, let's just hook them up to separate FPGA pins.

If we're really so short of FPGA pins that we can't do this then we should either add an I2C extender or consider using a bigger FPGA.

gkasprow, 2018-06-05:
we have a couple of pins routed to the IO header so we can use them.

hartytp, 2018-06-05:
- [ ] Good. So, to be sure, I think we need separate pins for: Si5324 RST; HMC7043 RST; HMC7043 output enable FETs.

- [ ] We also need to make sure that all these lines (as well as the HMC830 CS) have the right pull ups/pull downs to put things in the correct default state on power up.

hartytp, 2018-06-05:
> we have a couple of pins routed to the IO header so we can use them.

Anyway, if we do run out of IO, we can move up to a 50T in the same CSG325 package we're currently using (not that we should need an FPGA anywhere near that big).

BWT, @sbourdeauducq are you happy with the current RTM FPGA, or would you like to consider a bigger one to help fitting the DRTIO slave firmware on?

gkasprow, 2018-06-14:
Is it necessary to add FET that disables the outputs in v1.0 ? If so I can prepare quick manual howto. This involves cutting traces in a few places, adding 2 wires, 3..4 100nF capacitors and MOSFET.

jbqubit, 2018-06-15:
Given unknown origin of kernel crashes it might be worth trying this modification on one board at WUT to see if it improves the situation.

sbourdeauducq, 2018-06-15:
Another thing that would be useful to have (cc @jordens) is a kernel or some other method that crashes the board without involving anything on the RTM. Then we can simply unplug the RTM to help isolate the problem. At the moment I don't actually see any crashes other than with https://github.com/m-labs/artiq/issues/1065#issuecomment-396824032. There are a lots of things involved in this kernel and removing any of them inexplicably makes the crashes disappear.

gkasprow, 2018-11-27:
we won't be using HMC7043 any more, so closing.

marmeladapk, 2019-02-18:
Turns out, we will so reopening. :)

Ping @gkasprow

gkasprow, 2019-02-18:
True, now I have to revert schematics and layout to some state between v1 and v2...

gkasprow, 2019-02-25:
added FET to the lines routed to the FPGAs
[image]

closed 2019-02-25 by gkasprow