sinara-hw/sinara #557: Sayma: check HMC7043 + ramp gen

hartytp, 2018-06-05:
@marmeladapk, @jbqubit I've performed these checks (or similar ones), but it would be great to get results from a different Sayma. After applying the HMC7043 reset fix, please could you:

1. Verify that there are no longer any glitches on the sawtooth produced when Sayma ARTIQ is compiled without the SAWG.
2. Verify that the phase difference between HMC7043 outputs is constant between power cycles/FPGA loads. Use a fast scope and look at the phase difference between: both FPGA clocks (150MHz); the FPGA clock and the SYSREF (9MHz ish); the two SYSREFs; the two DAC clocks; the SYSREF and one DAC clock. In practice, you probably don't need to check all of these, but check a few

If that all looks good then we can verify that JESD is working and that the HMC7043 is operating correctly for SC1 mode.

hartytp, 2018-06-05:
NB accuracy should be <<1ns so that we can check for phase determinism at the level of a single cycle of the 1.2GHz clock.

jbqubit, 2018-06-06:
4.0.dev+1117.g38dac160 (today's master) build without SAWG. 

[image]
(above) Differential outputs J26/J27. AC-coupled with external 50-Ohm termination. 

[image]
(above) Differential outputs J26/J27. AC-coupled with external 50-Ohm termination.

jbqubit, 2018-06-06:
4.0.dev+1117.g38dac160 (today's master) build without SAWG. While waiting for SWAG build to finish. Clock comparison that's possible using RF connectors. Not sure what to expect from J60. 

[image]
(above) compare DIAG_N (J58) from ADCLK948 with GTP_CLK1_IN (J60)

@hartytp  Which FPGA clocks do you mean? 
- By SYSREF I assume you mean FPGA_DAC_SYSREF
- By DAC clocks I assume you mean DAC1_CLK and DAC2_CLK using scope probe at edge of HMC7043

hartytp, 2018-06-06:
By FPGA cloxks i mean the 150MHz outputs. Labelled something to do with gtp iirc

jbqubit, 2018-06-06:
> Labelled something to do with gtp iirc

That's not sufficiently specific.

hartytp, 2018-06-06:
In what way is that possible not sufficiently specific? 

The HMC7043 has 14 outputs, precisely two of which are labelled something to do with GTP: "GTP_CLK1_IN_{P,N}" and "GTP_CLK2_IN_{P,N}".

You can also look here https://github.com/m-labs/artiq/blob/cae92f9b44d257cfcbb81e38b62408d09972e950/artiq/firmware/libboard_artiq/hmc830_7043.rs#L151-L167

For a listing of what the output frequencies are. To check the "FPGA clocks" look at any pair of enabled outputs that have the divider set to `FPGA_CLK_DIV`, e.g. the GTP clocks, the FPGA aux clocks, etc.

jbqubit, 2018-06-06:
Wasn't clear if you're interested only in phase stability between outputs on HMC7043 or between HMC7043 and other FPGA-related clocks. What you're after is now clear.

hartytp, 2018-06-06:
Okay. Yes, the goal is to 

> Verify that the phase difference between HMC7043 outputs is constant between power cycles/FPGA loads.

jbqubit, 2018-06-06:
Baseline is system that produces RF on all SAWG channels, boots without error and has not generated any panics in several days of regular use. 4.0.dev+1117.g38dac160

Pair of 1 GHz scope probes. Measure skew between rising edges over 10 seconds. Plot as histogram. I can't also capture voltage-vs-time trace while holding scope probes but it appears on screen during measurement. 
- DAC1_CLK (pin 1)
- DAC2_CLK (pin 12) 
- Cycle power on board three times. This generates three histograms.

[image]
[image]
[image]

hartytp, 2018-06-07:
Thanks Joe.

So (with somewhat poor statistics) that verifies that the two DAC clock outputs from the 7043 have deterministic time delays w.r.t. each other to much better than the 800ps period of the 1.2GHz clock.

It would be good to look at the relationship between some other outputs as well, e.g. SYSREF to DAC clock. But, provisionally, that confirms my observation that the HMC7043 is ready for operation of Sayma with deterministic phases between the two DACs on the same Sayma.

hartytp, 2018-06-07:
@jbqubit @marmeladapk are you planning to take anymore data on this? Otherwise, let's close this issue.

marmeladapk, 2018-06-07:
@hartytp No, we sent our boards to @jbqubit.

jbqubit, 2018-06-07:
I don't yet have active differential probes for my scope. So there's most definitely distortion due to my probes. 

# GTP_CLK2_IN (J60) vs FPGA_DAC_SYSREF
The phase skew /looks/ stable for the several minutes I can hold my probes steady.

[image]

# GTP_CLK1_IN (J60) vs GTP_CLK2_IN
[image]
Not sure why there's so much distortion on GTP_CLK2_IN (green). This signal looks much worse than the other clocks coming out of HMC. I don't really trust the histograms here because of the snaggly waveform shape of GTP_CLK2_IN. But by eye the phase skew /looks/ stable for the several minutes I can hold my probes steady.

marmeladapk, 2018-09-18:
Paging @gkasprow
We have two RTMs, that sometimes don't calibrate DAC sysref correctly. Usually margins on DAC1 are -1 +127 and DAC_SYSREF signals are not in phase (>1 ns of difference). SPI communication seems fine, only there are two moments, where there is a large delay between first 16 bits and last 8.

This is the last issue preventing us from sending fully working Saymas back to UMD.

@sbourdeauducq did you see something like this on your Saymas? If so, how did you deal with it?

sbourdeauducq, 2018-09-18:
> did you see something like this on your Saymas?

No.

Did you disable RTIO/SYSREF align?

marmeladapk, 2018-09-18:
@sbourdeauducq Yes.

hartytp, 2018-09-20:
@marmeladapk I'm pretty sure that the RTM and AMC you returned me were not the ones I sent you. There were a few pieces of undocumented rework I'd done on my boards, such as adding a solder blob on a jumper to fix the JTAG issues, so it's quite inconvenient to get a different board back. Please try to keep track of which board is which...

hartytp, 2018-09-20:
I also now can't tell whether any new issues are due to me using a different board or for some other reason.

closed 2018-10-02 by hartytp