sinara-hw/sinara #558: Sayma: AD9154 2X interpolation works

jbqubit, 2018-06-10:
Using 2X interpolation on AD9154 and 2400 MHz DAC clock from HMC7043. Firmware changes

1) activate 2X interpolation 
ad9154.rs 
    INTERP_MODE, 0x01  for 2X 

2) double DAC clock 
hmc830_7043.rs 
    const DAC_CLK_DIV: u32 = 1; 

As far as I can tell this just works. 

sawg.frequency0.set(210*MHz) 
sawg.frequency1.set(50*MHz) 

#  1X interpolation 
[image]

#  2X interpolation 
[image]

jbqubit, 2018-06-10:
This is a good news Issue. Thanks for the tip @hartytp!

closed 2018-06-10 by jbqubit

jbqubit, 2018-06-11:
```
sawg.frequency0.set(210*MHz)
sawg.frequency1.set(-50*MHz)
sawg.frequency2.set(50*MHz)
```

[image]

hartytp, 2018-06-11:
Nice! Glad some things on Sayma just work.