sinara-hw/sinara #560: Sayma: HMC830 output spectrum

hartytp, 2018-06-11:
- [ ] To do: check HMC830 noise spectrum

With current ARTIQ master (4.0.dev+1133.g0b086225), looking at J89 + J59 through an AC-coupled TCM-43X+, I see:

Wide span (10kHz RBW). All looks good apart from the non-harmonic spur at 1.89GHz something like -60dBc.

Narrower span, including fundamental and first harmonic.

1MHz span around fundamental, 100Hz RBW. See some spurs at around 100kHz. I think these are SMPS spurs. @gkasprow can you confirm that? May need better power supply filtering or a layout tweak to remove them.

hartytp, 2018-06-11:
The 1.9GHz spur is still there while the board is being flashed and no clock is applied.

hartytp, 2018-06-11:
@gkasprow this could be some clock chip free oscillating without an input applied. I don't have fast enough scope probes to easily find that. Would you mind taking a look?

Thanks!

hartytp, 2018-06-11:
Oh, I think I know what this is... those clock chips are so damn fast that the put out a frequency comb up to pretty much optical frequencies.

I'm pretty sure that this is the 19th harmonic of the 100MHz reference input that's not perfectly removed by the loop filter! Wow...

Fixes for this are:
1. Add a RF filter on the PLL input to slightly reduce the rise time of the signal, while keeping it above 1V/ns
2. Add a higher-order loop filter/some RFI filtering on the loop filter.

jordens, 2018-06-11:
I'd expect to see other harmonics as well then.

hartytp, 2018-06-11:
True.

Well, anyway, this is probably something which @gkasprow or @marmeladapk are the best people to investigate...

hartytp, 2018-06-11:
Actually, on second thoughts, it can't be that, as it's still there with the clock disconnected. So, probably is just an oscillating clock buffer. Anyway, should be easy for someone with proper GHz probes to track down.

hartytp, 2018-06-12:
Here is a NSD measurement of the HMC830 output.

[image]

Simulation: 

https://cloud.githubusercontent.com/assets/21218399/20439893/a9a4e080-adb5-11e6-835f-7a54faac60f3.png

Notes:
- RBW is 10Hz, so subtracting 10dB from power measurement to convert to NSD (not bothering about various correction factors due to windows etc)
- Freq is 1.2GHz, so subtracting `20*np.log10(1200/100)=21.6dB` to normalize to 10MHz
- Simulation issue is https://github.com/sinara-hw/sinara/issues/75

hartytp, 2018-06-12:
So, this actually looks pretty good apart from the SMPS spurs...

hartytp, 2018-06-12:
Edit: updated figure above after fixing normalisation issue...

hartytp, 2018-06-12:
@gkasprow can you see if you can figure out where those spurs are coming from and how they couple in (power lines? pickup? noise in the ground plane?)

NB I see quite a bit of SMPS noise on the HMC7043 outputs even when they're muted, so I think there is an issue that needs to be looked at there!

hartytp, 2018-06-12:
[image]

HMC7043 GTP_CLK1 (J60 and J61) at 150MHz.

hartytp, 2018-06-12:
Same as before, but now a 1GHz span.

[image]

hartytp, 2018-06-12:
[image]

still looking at GTX_CLK1 TPs, but this time with 150MHz outputs. RBW=1kHz and (hopefully) the correct normalization this time...

hartytp, 2018-06-12:
Now, looking at my 100MHz reference on the MEZ_CLK_IN SMPs.

Wide scan, 1kHz RBW.

[image]

hartytp, 2018-06-12:
5MHz span, 10Hz RBW, still looking at 100MHz reference on SMPs

[image]

hartytp, 2018-06-12:
Same, but putting my synth directly on the spectrum analyzer. Note that only the left section of the measurement completed (the data taking takes an age and I got impatient, the rhs is a previous measurement).

[image]

So, a lot of what I see above is due to the synth I'm using as my clock source, which has worse spurs than I remembered.

hartytp, 2018-06-12:
Anyway, so far, Sayma looks excellent noise-wise. We should put it on a proper phase noise metre at some point to get better data.

hartytp, 2018-06-12:
Looking on the SMPs at 1.89GHz/12=157.5MHz, I don't see anything with or without my clock source applied. Same at 1.89GHz (although, I do obviously see a 1.9GHz harmonic of the squared reference). I'm no longer seeing the 1.89GHz spur on the HMC830 output. Not sure what happened (measurement error? something coming out of lock?)

Anyway, that concludes these measurements for now. 

tk;dr: looks good! But, needs more careful measurements with the proper equipment.

closed 2018-06-12 by hartytp

hartytp, 2018-06-12:
okay, the 1.89GHz is just lab pickup...

hartytp, 2018-06-13:
Posting this here for want of a better place...

Here is the SAWG output at 210MHz, as measured from the SMPs via an AC-coupled balun (TCM2-43X+). Using same clock source as above...

[image]

hartytp, 2018-06-13:
10Hz RBW, narrower span, still in power units.
[sawg_narrow.txt](https://github.com/sinara-hw/sinara/files/2097794/sawg_narrow.txt)

[image]

hartytp, 2018-06-13:
Same, but plotted in PSD units.
[image]