sinara-hw/sinara #561: Sayma: HMC830 loop filter layout

hartytp, 2018-06-12:
I have carefully prepared a diagram showing the layout of the HMC830 loop filter:

[image]

As you can see, this loop filter spans something like 1cm2 of board area, and loops right around the IC. Given that this is one of the most sensitive traces in the design, that's not really good enough.

In particular:
- https://github.com/sinara-hw/sinara/issues/560 shows that the LF phase margin is much lower than simulated. I suspect that this is due to large parasitics in the loop filter layout. In principle, we could empirically optimize the loop filter RCs to compensate for this (to a certain extent at least) but that's a lot of work if the parasitics are this large. As a result, I'm not inclined to spend time optimizing the loop filter on Sayma v1.0, but wait for a better layout on V2.0
- it's a huge area for pickup (and, indeed, we do see a lot of SMPS spurs on the HMC830 output)

So:
- [ ]  Redo the HMC830 layout to keep it as tight as possible (small loop area, short traces, etc
- [ ] ensure that HMC830/HMC7043 are far from any SMPSs and other sources of noise
- [ ] ensure that all sensitive analog signals are kept short and, where appropriate, routed on internal layers, sandwiched between ground plane

jordens, 2018-06-12:
[image]

Hmm. The alternative would be the board top like on the eval board. But then you don't have anything from the shield.
And the loop only looks about twice as large as the one from the eval board.

hartytp, 2018-06-12:
I'd prefer a compact layout to a screening can any day!

hartytp, 2018-06-12:
Having looked at the data more carefully, it seems that the loop filter does match the simulations quite well, so the layout parasitics aren't as bad as I'd thought...

hartytp, 2018-06-12:
Looking again at the layout, the nice thing about having an RTM is that there really aren't any noise sources anywhere near the HMC830. The FPGA and SMPSs are quite a long way away, only the clock mezzanine is near it.

marmeladapk, 2018-12-13:
@hartytp Is this still something you'd like to see in v2.0?

marmeladapk, 2018-12-13:
Then I'm closing this, but including in review list as an example of things to check.

gkasprow, 2019-03-11:
we will do new HMC layout so we can close it now.

closed 2019-03-11 by gkasprow