sinara-hw/sinara #563: Sayma RTM FPGA PUDC_B problems

sbourdeauducq, 2018-06-13:
According to https://www.xilinx.com/support/documentation/user_guides/ug470_7Series_Config.pdf :

* PUDC_B must not be allowed to float during configuration.
* This pin is currently used for DAC2_SPI_SCLK and HW_ID3. I do not know what resistors are populated there.
* The impedance must be less than 1K. It is currently 10K or 20K.
* By connecting this pin to GND, internal pull-ups are enabled after FPGA power-up and during configuration, and can help keep the HMC7043 and other insufferable chips in check.

sbourdeauducq, 2018-06-13:
Note that this pull-up system is not so well-designed:

"When PUDC_B is tied to GND, the activation of internal pull-ups during power-on depends 
on the power sequence because the PUDC_B control signal is forwarded through an input buffer in bank 14 and internal paths to the enables of internal pull-ups at applicable pins in their respective I/O banks. An external pull-up resistor is  recommended between a pin and the pin’s VCCO  power supply when it is critical for the pin to be pulled High immediately as the pin’s VCCO power ramps up."

gkasprow, 2018-06-13:
All resistors are populated. So it means you have 10k pullup at the moment.
So you can replace R258 with 1k to get pull up and swap R253 with R254 to get pull down.

gkasprow, 2018-08-10:
replaced with 1k

closed 2018-08-10 by gkasprow