sinara-hw/sinara #564: LVDS requires 2.5V bank on Artix-7

sbourdeauducq, 2018-06-13:
This breaks the serwb signals.

gkasprow, 2018-06-13:
you can switch the IO standard to SSTL18 with internal termination

sbourdeauducq, 2018-06-13:
No, internal termination is only available on HP banks. Or am I missing out something?

marmeladapk, 2018-06-13:
@sbourdeauducq Yes, DCI standards are only available on HP banks.

gkasprow, 2018-06-13:
so it seems we have to solder the 100Rresistors to the FPGA vias directly and use 1.8V SSTL signalling.

sbourdeauducq, 2018-06-13:
Can we get new Sayma boards soon? The quantity (and fragility) of the current reworks begins to be a significant problem.

gkasprow, 2018-06-13:
Let's find as many as possible HW issues with current boards. There will for sure be ones related with FPGA loading. Soldering of 2 100R resistors is not a big issue, you just need fibreglass pen to scratch the soldmask. vias are separated by 1mm so you will easily solder 402 resistors. Anyway, we won't be able to use 2.5V supply for the RTM FPGA bank so will need to use resistor termination anyway

gkasprow, 2018-08-10:
I added 100R termination resistors on RTM_FPGA_LVDS*  and RTM_FPGA_USR_IO_N lines 
So for input LVDS (in SSTL18 mode) this should work. For output LVDS one needs to use SSTL18 setting as well.

sbourdeauducq, 2018-08-11:
Are RTM_FPGA_LVDS* and RTM_FPGA_USR_IO_N different? If no they should be named the same.
How exactly do you use LVDS with IOs in SSTL18 mode? Who will write the gateware for this, test it and debug it?

gkasprow, 2018-08-11:
I will rename them.
SSTL is ordinary input/output like LVCMOS but centered around 1/2vcc
we cannot really use 2.5V bank supply because there are too many chips which require 1.8V.
If the direction of the LVDS lines is fixed, I can add LVDS - LVCMOS translator. What data rate do you use on these lines? I can also try to add bidir translator since I have two lines on FPGA.

hartytp, 2018-08-11:
NB this also means that we can't use the RTM FPGA to control LVDS peripherals on the AFE mezzanines. I'm not saying that's a problem, but just something we need to remember for v2.0.

sbourdeauducq, 2018-08-11:
The initial plan was to run at 1Gbps, and reducing it is not straightforward due to the limited IODELAY range of FPGAs. The external buffers are an option, the lines are unidirectional, but:
* the buffers typically do not go to 1G
* there are sometimes problems when using certain FPGA I/O features (e.g. phase detector on Spartan-6) with single-ended IOs

I suppose the lowest-cost option is to keep the current slightly dirty 125Mbps single-ended code, and add the external buffers if they improve EMI.

Are there any available differential standards with the I/O bank voltage restrictions anyway?

gkasprow, 2018-08-11:
DIFF SSTL18 should do the job.

gkasprow, 2018-08-11:
O course I will add suitable termination at the FPGA input.

gkasprow, 2018-08-11:
@hartytp the lines that go to AFE are supplied from 3.3V. No FPGA families support LVDS at that voltage.
What we can do?
Change AFE signalling to 2.5V. Most TTL inputs work with 3.3V signalling. We will modify AFEs anyway so can adapt them to 2.5V levels.
Then we will move SPI lines to 1.8V bank and add level converters and move high speed LVDS lines to 2.5V bank to make @sbourdeauducq happy.
In this way AFE will work with LVDS or LVCMOS25, true LVDS communication between FPGAs will be possible.

hartytp, 2018-08-11:
> @hartytp the lines that go to AFE are supplied from 3.3V. No FPGA families support LVDS at that voltage. What we can do?

As I said, I think the current situation is fine. We just need to remember that the RTM can't connect to LVDS peripherals on the AFE, such as ADCs. So long as we leave enough lines free on the AMC<->RTM connector for these LVDS ICs, I don't think this is a problem.

> Change AFE signalling to 2.5V. Most TTL inputs work with 2.5V signalling. We will modify AFEs anyway so can adapt them to 2.5V levels.

I agree that the AFEs should work fine from 2V5 LCMOS. So, if it helps the SER-WB links to move those banks to 2V5 then it might be a good idea. We just need to make sure that this change doesn't break anything else!

> Then we will move SPI lines to 1.8V bank 

You mean the HMC830/HMC7043 SPI? Is the reason to move them just to free up some pins on the 2V5 bank?

gkasprow, 2018-08-11:
@hartytp Yes, I want to move them to free some pins. But anyway we would probably need translators to cope with 2.5V bank supply, at least for inputs. We can also move CAL ADC  (AD7194BCPZ) control lines to 1.8V bank and add translators

gkasprow, 2018-08-12:
Let's try SSTL then. I can solder 100R resistor to the BGA vias and see how from SI it looks like.
Can you instantiate SSTL18 in and out buffers in the code for RTM please?

gkasprow, 2018-08-12:
And route in signal to one of DIO pins.

sbourdeauducq, 2018-09-21:
With the current clocking plans, we can also remove those lines and serwb and run everything over pure DRTIO (with 3Gbps GT lines).

hartytp, 2018-09-21:
So you plan to program the hmc830 over drtio?

sbourdeauducq, 2018-09-21:
Yes.

hartytp, 2018-09-21:
ok, works for me so long as you think you can get RTM drtio up and running quickly

hartytp, 2018-09-22:
@sbourdeauducq 
- in that case, let's put the SYSREF generator (a counter that's reset at the same time as the coarse finestamp) on the RTM as well.
- let's also put the retiming FF and the delay line on the RTM. Might want to put the delay line on a separate SPI bus to the HMC830, since the 830 is such a fussy pita. 
- let's also loop the FF output back to the RTM FPGA for S/H validation
- The sysref output/input after retiming can be any old single-ended standard since they only need to meet S/H at f_rtio up to 150MHz
- however, pending a full audit of signals on the AMC<->RTM connector, it might still be worth routing some LVDS lines directly between the RTM and AMC. They might come in handy for some tasks that one doesn't want to do over DRTIO...

sbourdeauducq, 2018-09-23:
>  let's also loop the FF output back to the RTM FPGA for S/H validation

This may be somewhat complicated to do correctly, and not a requirement as the AD9154 DAC has SYSREF diagnostics which by miracle do not appear to be buggy or have braindamaged design. You  can ask it over SPI if it received a SYSREF pulse.

> it might still be worth routing some LVDS lines directly between the RTM and AMC.

LVDS is not really doable (see above), it's either differential SSTL or some more common single-ended standard. We can also keep those links slow anyway.

hartytp, 2018-09-23:
> LVDS is not really doable (see above), it's either differential SSTL or some more common single-ended standard. We can also keep those links slow anyway.

Yes, sorry, I'd misremembered what the issue was.

So, we can have SSTL but, for fast inputs we have to solder external termination resistors. That means we have to fix the direction of each signal, making them less useful. Probably not worth it then, given how limited we are in terms of AMC<->RTM IO.

hartytp, 2018-09-23:
> This may be somewhat complicated to do correctly, and not a requirement as the AD9154 DAC has SYSREF diagnostics which by miracle do not appear to be buggy or have braindamaged design. You can ask it over SPI if it received a SYSREF pulse.

Using the SYSREF diagnostics in the AD9154 to validate two CDCs sounds a little messy. If you get unlucky and end up right in the transition region of both CDCs then how do you figure out what's going on. Not the hardest thing in the world, but could get a little messy in the firmware. So, worth considering if there are any extra diagnostics we can add to the HW to make life easier in firmware. Anyway, happy to leave this decision to you.

hartytp, 2018-09-23:
Maybe this is a bad idea, but I wonder if it would be worth doing something like latching the FPGA-generated SYSREF signal twice, but with the second FF having a slightly delayed clock. Then put the SYSREF generate in single-shot mode and check for the two FFs having the same output. That gives you a simple way of determining the transition region.

Anyway, if you don't think it's worth it then no point complicating the HW design...

sbourdeauducq, 2018-09-23:
> Using the SYSREF diagnostics in the AD9154 to validate two CDCs sounds a little messy.

We can generate a SYSREF pulse from the FPGA that lasts less than a cycle of the FF clock frequency (e.g. FF clock period 8ns, pulse duration 4ns), scan a delay on this signal while looking at whether the DAC receives a pulse, and then fix the delay in the middle of the region where it does.

hartytp, 2018-09-23:
...I didn't think of doing something like that. That's a better idea.

gkasprow, 2018-11-28:
Since we want to support LVDS on analog mezzanines, what about using 2.5V signalling for mezzanines IOs as well? They could be single ended or differential depending on need. And 2.5V signalling works well with TTL levels. In this way we would solve RTM-AMC communication issue as well.

hartytp, 2018-11-29:
@gkasprow Switching to 2V5 logic on the RTM FPGA sounds like a good idea. When we come to the design review stage, we must make sure to do a thorough audit of everything hooked up to the RTM to ensure that 2V5 LVCMOS won't cause any issues, or add level translators where required.

gkasprow, 2018-11-29:
I meant to switch only AFE-related banks. Some logic will still need 1.8 or 3.3V

hartytp, 2018-11-29:
Sounds good. We'll still need to double check all the components on the AFE mezzanines to check they work with 2V5, but I don't expect issues...

dhslichter, 2018-11-29:
Although 3.3V LVCMOS still works, 2.5V LVCMOS might be pushing it a little.  These are slow signals, though, so putting level translators on the AFEs if necessary seems like a reasonable solution to me.  The range pins A0 and A1 on the AD8253, for example, require 2.8V to register a logic high, and the HMC542 wants +3V to register a logic high.  The HMC349, which needs the fastest logic, only needs +2V to register a logic high, so wouldn't need translators.

gkasprow, 2018-12-16:
We can keep 3V3 for mezzanine IOs. We already have 12 LVDS lines from Sayma AMC FPGA. The plan is to route 10 of them directly and 2 to RTM FPGA (LTC2324 needs 5 f them). I will connect 2.5V instead of 1.8V. DAC IOVDD can handle this.  
But if we want to have more LVDS also on mezzanine IOs, we have to switch to 2.5V and add translators.
I sketched schematics of BaseMod and we need translators in many places.
So I propose 2 scenarios:
1.
- route 32x 3.3V LVCMOS lines from RTM FPGA to each mezzanine
- route 2 LVDS lines from RTM FPGA to each mezzanine. 
- route 5 LVDS lines from AMC FPGA to each mezzanine
2.
- route 16 x 3.3V LVCMOS lines from RTM FPGA to each mezzanine
- route 10 LVDS lines from RTM FPGA to each mezzanine. 
- route 5 LVDS lines from AMC FPGA to each mezzanine

The second scenario seem to be optimal.

hartytp, 2018-12-17:
> The second scenario seem to be optimal.

That only works with a lot of level translators/LVDS->LVCMOS drivers, right? I'd prefer to avoid that unless we really have to.

It would be helpful if someone could do an audit of the components on BaseMod and count the number of signals that need 3V3 logic levels.

Also, don't forget that other mezzanines may need more 3V3 IO (e.g. MixMod will have a PLL).

gkasprow, 2018-12-17:
No, The second scenario uses both 16x 3V3 signals and 16x 2V5 signals assigned as LVDS. With 16x 3.3V signals we will support 99% of use cases including complex PLL chips. For remaining signals like LEDs, 2.5V logic would be fine. In remaining 1% of cases we will add some translators. LVDS are valuable signals.

gkasprow, 2018-12-30:
I connected 2 banks to 2.5V so the issue is solved. The same bank is also used for mezzanine LVDS lines.

closed 2018-12-30 by gkasprow