sinara-hw/sinara #565: [sayma_rtm] IC48 LTM4619 VOUT1/2 of 6V is above abs max rating or 5V

jordens, 2018-06-13:
Don't know whether it's a problem right now, but it's probably not best practice.
Credits to Sven Rotter for noticing.

gkasprow, 2018-06-13:
good catch, we cannot leave it as it is. We can replace it with i.e. [TPSM84824](http://www.ti.com/product/TPSM84824)

hartytp, 2018-06-13:
@gkasprow is this likely to cause problems during testing?

gkasprow, 2018-06-13:
I didn't notice any issues with this converter.
I'm quite surprised why is there such limitation. It's not voltage breakdown of MOSFETs, since they operate at 12V anyway. Could be internal capacitor voltage rating.
I wouldn't expect problems with such operating condition, but we will change the module to make sure we operate within recommended limits.

gkasprow, 2018-11-28:
We will use [TPSM84824MOLR](https://pl.mouser.com/ProductDetail/Texas-Instruments/TPSM84824MOLR?qs=W0yvOO0ixfEo%2FgpT5Em6eQ%3D%3D)
it is much smaller and cheaper than LTM4619

gkasprow, 2018-12-15:
done

marmeladapk, 2019-01-14:
Shouldn't TT resistor be around 70kOhm? After using equation from datasheet with approximated values resistor should be much larger than it is now.

gkasprow, 2019-03-05:
Good catch. Replaced with 30k according to table 8 in Datasheet.
[image]

closed 2019-03-05 by gkasprow