sinara-hw/sinara #566: Sayma AMC: fix FPGA VCCINT supply

sbourdeauducq, 2018-06-28:
c.f. https://github.com/m-labs/artiq/issues/1065

* VCCINT setpoint too low
* VCCINT unstable and dropping under load

Note that we envision an increased FPGA consumption as more ARTIQ/SAWG features are added, so the VCCINT drop should be well understood and properly fixed.

gkasprow, 2018-11-27:
fixed 
[image]

closed 2018-11-27 by gkasprow

marmeladapk, 2018-12-18:
VCCINT is set to ~0,95 V + feedback is from under FPGA. @sbourdeauducq is that fine?

sbourdeauducq, 2018-12-18:
Assuming we use the -1 speed grade, 0.95V is the nominal value from the datasheet.
The PCB layout should be carefully simulated and reviewed, perhaps by at least two independent people. On Sayma v1, Greg's simulation results were wrong (Xilinx tech support ran the simulation again and found issues), the load-dependent voltage drop was excessive, and this contributed to the "SAWG crashes/HMC7043 noise/SDRAM" bug saga.

sbourdeauducq, 2019-02-23:
"
In offline discussion @sbourdeauducq asked...

```
One thing about the power supply: can we run the PCB layout by the 
Xilinx engineers like last time, and have them cross-check the DC drop 
simulation results before PCB production, for both Sayma and Metlino? It 
would be a shame if this problem comes to bite us again.
```

In offline discussion 2/21 @gkasprow responded

`I will publish simulation sources and results so it would take a moment for Xilinx team to review it.`

In offline discussion 2/21 @gkasprow noted

```
I finally found what was a problem with DC drop. It was not really 
caused by the wrong simulation but by the fact that the PCB 
workshop increased clearances around the vias which caused DRC increase 
of the supply plane. Another issue was missing compensation resistor.
```
"