sinara-hw/sinara #567: Sayma AMC 3.3V failure

sbourdeauducq, 2018-07-05:
One of the Sayma cards (F) has an intermittent 3.3V problem. The 3.3V rail fails between <1s and hours after power-up.
Exar dump from one incident when it failed shortly after startup:
```
------------ Exar Dump ----------
GET_HOST_STS 0x2 0x4 0x3
GET_FAULT_STS 0x5 0x0 0x20
PWR_GET_STATUS 0x9 0xd 0x2
PWR_READ_VOLTAGE_CH1 0x10 66 66 V
PWR_READ_VOLTAGE_CH2 0x11 56 56 V
PWR_READ_VOLTAGE_CH3 0x12 120 120 V
PWR_READ_VOLTAGE_CH4 0x13 100 100 V
PWR_READ_VOLTAGE_IN 0x14 160 160 V
```

sbourdeauducq, 2018-07-05:
@gkasprow Should I send you the board or is there something I can try to fix myself?
This is happening very frequently now, and blocking the development of inter-board synchronization.

gkasprow, 2018-07-05:
sure, send it to Technosystem

sbourdeauducq, 2018-07-08:
The other AMC also begins to exhibit this bug now :(

sbourdeauducq, 2018-07-09:
@gkasprow I'm still waiting for your detailed shipping instructions.

hartytp, 2018-07-09:
@sbourdeauducq send it to TechnoSystems via any respectable carrier (FedEx/DHL/etc). I've never had an issue doing that.

sbourdeauducq, 2018-07-09:
Let's see after Brexit :)

sbourdeauducq, 2018-07-09:
Anyway, I received the information and will send it with DHL tomorrow.

jordens, 2018-07-09:
NB for anyone who wants to debug this: looking at Exar's "UnivPMIC" the XR77129 register layout might be more or less the same as the [XRP7724](https://github.com/exarcorp/UnivPMIC/blob/master/XRP7724.cpp) which is also [documented](https://www.exar.com/appnote/anp-44_xrp7724_regdefn_appnote_032715.pdf) and has some [examples](https://www.exar.com/appnote/anp38_r120_011614.pdf)

gkasprow, 2018-07-09:
XR77129 is the same as 7725 but has higher voltage rating. 7725 supports someIntel  management protocol that 7724 does not.

jbqubit, 2018-07-09:
While waiting for Sayma to ship to HK, can M-Labs do testing via remote login to WUT system?

gkasprow, 2018-07-09:
We can make it quickly tomorrow using TeamViewer.

sbourdeauducq, 2018-07-09:
> using TeamViewer

That won't be useful, we need SSH or (better) Mosh. Anyway I don't think it's worth it to access the WUT boards remotely, DHL between Poland and HK is rather fast.

gkasprow, 2018-07-09:
I have 2 VPN accounts so we can do it.

jbqubit, 2018-07-12:
@gkasprow How would @sbourdeauducq disable on-board 3.3V supply and supply 3.3V from bench top PSU? Is this even a good idea?

sbourdeauducq, 2018-07-12:
This sounds like a bad idea, e.g. if the exar chip disabled the channel due to overcurrent.

jbqubit, 2018-07-12:
Set current limit on your benchtop supply.

sbourdeauducq, 2018-07-13:
All the lab power supplies that I have turn into current sources when the max current is reached. And we are typically leaving the board on all the time. So, unless I change the behavior of the PSU to make it work like a circuit breaker, the board might keep receiving its maximum current for days, which does not sound safe. And we do not currently have time or funding for this sort of Sayma hardware debugging.
As I mentioned in another issue, the board that is left in HK isn't strongly affected by this bug yet; it typically behaves itself for 30min-1 day after being turned on. So it's not a huge impediment to development (except for inter-board synchronization, since this is the only board we have left), but since likely this bug will get worse, please investigate it quickly after receiving the board @gkasprow.

jbqubit, 2018-07-13:
> the board might keep receiving its maximum current for days, which does not sound safe.

There's no need for fuse/breaker when using a current limited supply. If board nominal current draw is X set current limit to (1+eps)*X for whatever eps is safe CW. But sounds like your board's supply isn't bad enough yet to warrant using a bench top power supply.

sbourdeauducq, 2018-07-15:
@gkasprow Even though I used DHL as you instructed, the package is again stuck in Polish customs. Can you handle the import this time please?

sbourdeauducq, 2018-07-19:
Still stuck in customs...

hartytp, 2018-07-19:
:(

Do you use cocaine for padding when you ship them? Maybe EU customs are just suspicious of HK

gkasprow, 2018-07-19:
This time it stucked in DHL, not polish post and this is huge difference :)

sbourdeauducq, 2018-07-24:
Still stuck despite new paperwork sent yesterday.

sbourdeauducq, 2018-07-24:
And in my experience, EU customs are suspicious of many things coming from outside EU; it is also a problem to receive most items from e.g. US into Germany or France. "Respectable carriers", as you call them, also exploit the customs mess to make money in pretty shady ways, see the end of http://www.minimachines.net/a-la-une/la-livraison-depuis-lasie-delais-prix-transporteurs-57204.

gkasprow, 2018-07-30:
@sbourdeauducq Yes, I received it a few minutes ago. Will investigate them tomorrow.

gkasprow, 2018-07-31:
@sbourdeauducq the AMC is working stand-alone running ARTIQ together with RTM already 2 hours and nothing happens to the supply...

gkasprow, 2018-07-31:
When I received it, the Exar chip started after a few seconds. I burned recent MMC firmware and so far it works.

gkasprow, 2018-07-31:
The configuration of both Exar chips is also fine.

sbourdeauducq, 2018-07-31:
Try the other board I sent you and which Technosystem received today. It also has this 3.3V bug. Is there a new MMC firmware? I thought I had flashed the latest one already.
Why did it take a few seconds to start the Exar chip?

Sometimes (but more rarely) the 1.5V supply also fails.

gkasprow, 2018-07-31:
It looks like there was old version of  firmware which was waiting for MCH response

gkasprow, 2018-07-31:
I will leave it over night and check tomorrow morning.

gkasprow, 2018-08-01:
@sbourdeauducq I took the RTM you shipped back with annotation that HMC does not lock:

Booting from flash...
Starting firmware.
[     0.000004s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005668s]  INFO(satman): software version 4.0.dev+1219.g4eb26c00
[     0.011930s]  INFO(satman): gateware version 4.0.dev+1214.g729ce58f
[     0.018172s]  INFO(board_artiq::slave_fpga): Loading slave FPGA gateware...
[     0.025120s]  INFO(board_artiq::slave_fpga):   magic: 0x5352544d, length: 0x000c15b4
[     1.038593s]  INFO(board_artiq::slave_fpga):   ...done
[     1.042339s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     1.077067s]  INFO(board_artiq::serwb):  ...done.
[     1.080377s]  INFO(board_artiq::serwb): RTM to AMC link test...
[     2.562678s]  INFO(board_artiq::serwb):   ...passed
[     2.566161s]  INFO(board_artiq::serwb): AMC to RTM link test...
[     4.048469s]  INFO(board_artiq::serwb):   ...passed
[     4.051961s]  INFO(board_artiq::serwb): Wishbone test...
[     5.983985s]  INFO(board_artiq::serwb):   ...passed
[     5.987468s] DEBUG(board_artiq::serwb): AMC serwb settings:
[     5.993025s] DEBUG(board_artiq::serwb):   bitslip: 39
[     5.998062s] DEBUG(board_artiq::serwb):   ready: 1
[     6.002837s] DEBUG(board_artiq::serwb):   error: 0
[     6.007613s] DEBUG(board_artiq::serwb): RTM serwb settings:
[     6.013178s] DEBUG(board_artiq::serwb):   bitslip: 6
[     6.018128s] DEBUG(board_artiq::serwb):   ready: 1
[     6.022904s] DEBUG(board_artiq::serwb):   error: 0
[     6.027912s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1214.g729ce58f
[     6.295633s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     8.726003s]  INFO(board_artiq::si5324):   ...locked
[     8.729686s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     8.735754s]  INFO(board_artiq::hmc830_7043::hmc830): loading HMC830 configuration...
[     8.743797s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[     8.749458s]  INFO(board_artiq::hmc830_7043::hmc830): setting HMC830 dividers...
[     8.756919s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[     8.762743s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for HMC830 lock...
[     8.770161s]  INFO(board_artiq::hmc830_7043::hmc830):   ...locked
[     8.776390s]  INFO(board_artiq::hmc830_7043::hmc7043): enabling HMC7043
[     8.793075s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     8.798051s]  INFO(board_artiq::hmc830_7043::hmc7043): loading configuration...
[     8.806895s]  INFO(board_artiq::hmc830_7043::hmc7043): status=10
[     8.811508s]  INFO(board_artiq::hmc830_7043::hmc7043):   ...done
[     8.817502s]  INFO(board_artiq::hmc542): card 0 channel 0 set to 4 dB
[     8.826008s]  INFO(board_artiq::hmc542): card 0 channel 1 set to 4 dB
[     8.833138s]  INFO(board_artiq::hmc542): card 1 channel 0 set to 4 dB
[     8.840267s]  INFO(board_artiq::hmc542): card 1 channel 1 set to 4 dB
[     8.847398s]  INFO(board_artiq::hmc542): card 2 channel 0 set to 4 dB
[     8.854527s]  INFO(board_artiq::hmc542): card 2 channel 1 set to 4 dB
[     8.861658s]  INFO(board_artiq::hmc542): card 3 channel 0 set to 4 dB
[     8.868787s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB

gkasprow, 2018-08-01:
what power level of 100MHz clock do you use?

gkasprow, 2018-08-01:
And both AMC board you shipped to me (the ones covered with thick layer of dust :) ) don't have Ethernet clock line modification. So it's not surprise that Ethernet does not work.
On of them has sticker saying that there is 1.5V bug another has 3.3V bug sticker.
The one with 1.5V bug has PRBS issues.

But from power supply point of view, they are working fine.
What power supply do you use?

gkasprow, 2018-08-01:
Anyway, I will focus on GTP2 clock on this particular board.

gkasprow, 2018-08-01:
on GTP CLK2 the DC component is 0.7V while on GTP CLK1 it is 0.4V. The DC value is set by FPGA due to capacitive coupling.

hartytp, 2018-08-01:
is this some configuration issue with ARTIQ, or a hardware issue? Can we reproduce that observation with a simple design based on Xilinx IP?

gkasprow, 2018-08-01:
The [datasheet ](https://www.xilinx.com/support/documentation/user_guides/ug476_7Series_Transceivers.pdf) , p33 says
The reference clock input structure is illustrated in Figure 2-1. The input is terminated  internally with 50Ω on each leg to 4/5 MGTAVCC. The reference clock is instantiated in  software with the IBUFDS_GTE2 software primitive. The ports and attributes controlling the reference clock input are tied to the IBUFDS_GTE2 software primitive.

gkasprow, 2018-08-01:
So neither of these values I measured makes sense.
I see only one option - modify my design that tests gigabit transceivers, instantiate clock inputs by Wizard and repeat measurements

hartytp, 2018-08-01:
The data sheet you link to isn't for ultrascale. That's https://www.xilinx.com/support/documentation/user_guides/ug576-ultrascale-gth-transceivers.pdf see figure 2-1

hartytp, 2018-08-01:
Are you sure that all the MGTAVCC pins are connected correctly, that MGTAVCC has the right voltage and that there are no nasty transients on it during startup?

hartytp, 2018-08-01:
@gkasprow if you disconnect the HMC7043 what DC voltages do you measure on these clock inputs?

hartytp, 2018-08-01:
Might also be interesting to stick a scope on those inputs and look at the DC voltage as Sayma boots up

gkasprow, 2018-08-01:
It is still 4/5 AVCC. MGTAVCC is filtered VCCINT which I observed several times.
Plan for today:
- check DC voltages and signal amplitude with the IP core I generated with external 156.25MHz clock from loopback board.
- do the same with RTM board

sbourdeauducq, 2018-08-01:
@gkasprow @hartytp Please keep this issue on the 3.3V power supply failure topic.

sbourdeauducq, 2018-08-01:
> But from power supply point of view, they are working fine.
> What power supply do you use?

ATX.

gkasprow, 2018-08-01:
with Xilinx IP core I get nice looking, 0.75V biased signal on GTP CLK1. Synthesizing for other clock input.

>

sbourdeauducq, 2018-08-01:
@gkasprow This is off topic here.

sbourdeauducq, 2018-08-01:
A third AMC I had and the last board you sent me (without the SDRAM bug) also don't have the 3.3V bug when powered in the same way.

gkasprow, 2018-08-01:
After several power cycles and a few hours running ARTIQ (>30W of power) I still cannot recreate the issue on any of boards.
@sbourdeauducq  What kind of ATX supply do you use? Older designs did not stabilize 12V rail, only 5V. So the voltage may be significantly lower and not stable. New, Gold+ class designs use resonant converter that delivers stable +12V and generates remaining rails using buck converters from 12V.

hartytp, 2018-08-01:
What does the Exar register dump tell you about the cause of the 3V3 failure on @sbourdeauducq's board?

jbqubit, 2018-08-01:
@sbourdeauducq Do you have a laboratory power supply available (instead of ATX)?

gkasprow, 2018-08-01:
It says : 
GET_HOST_STS 0x2 0x4 0x3
GET_FAULT_STS 0x5 0x0 0x20
PWR_GET_STATUS 0x9 0xd 0x2

which means channel 2 OVP event, CH2 fault.
So there could be i.e. sudden voltage burst at the input.
Anyway, I will check if such case can trigger OVP event in Exar chip.
Of course I can switch off the OVP protection but I'd prefer to understand why it gets activated.

hartytp, 2018-08-01:
@gkasprow it seems unlikely that this could be due to @sbourdeauducq's power supply. Doesn't the Exar chip also monitor the input supply, and wouldn't it register an input over voltage event separately? If there is a burst of noise at the input, why does it only affect this one output?

Can you do a step response measurement on the Exar output to see if there are any more issues with loop stability (e.g. if a capacitor has been damaged/degraded).

gkasprow, 2018-08-01:
Exar chip that we use accepts input voltages up to 40V so it's hard to trigger input OVP. But sudden peak can trigger output OVP because the regulation loop works with certain delay. Old-type ATX supplies that regulate only 5V rail, work in non-continuous mode when have only 12V rail loaded. I had issues with such supply and had to apply load to 5V rail to get stable operation. So this may be a reason.

gkasprow, 2018-08-01:
Anyway I will try both approaches - add voltage bursts to the input and do step response test to the output and see what happens.

hartytp, 2018-08-01:
@gkasprow true. I remember having to stick some pretty chunky load resistors on 5V rails before to get the PSUs to behave themselves. Still though, I'm surprised that only a single channel is tripping. Suggests that maybe the loop isn't as stable for that channel as the others.

gkasprow, 2018-08-01:
3.3V rail is the most vulnerable. the OVP is set to react at certain threshold above nominal value. Let's say 100mV. And the higher is the output voltage, the easier is to trigger the OVP.
Let's assume that there is 5V burst so instead of 12V Exar chip gets 17V. For 3.3V rail the PWM was 27.5% so for short moment before the loop reacts, the output capacitor is charged to 4.675V via inductance.
In case of 1.5V the PWM is 12.5% so the output cap is charged to 2.125V. The worst-case over-voltage for 3.3V rail is 1.375 and for 1.5V is 0.625. So the 3.3V will be  first that triggers such protection.
Such case affects all DC/DC converters but Exar is very fragile due to numerous protection.
It's better to keep them on at least during development phase.

hartytp, 2018-08-01:
@gkasprow okay, point taken!

@sbourdeauducq do you have any other boards currently displaying this issue? If so, can you stick a scope on the 12V input and set it to trigger if the voltage goes above, say, 13V? That would allow us to rule out the PSU.

sbourdeauducq, 2018-08-02:
> do you have any other boards currently displaying this issue?

No, but I can still double-check the ATX power supply (I already had a quick look with a multimeter and didn't see an issue).

[38 more replies]

closed 2019-03-06 by hartytp