sinara-hw/sinara #572: Sayma: GTX refclk bias voltage

hartytp, 2018-08-01:
Moving from https://github.com/m-labs/artiq/issues/1080#issuecomment-409607643

gkasprow, 2018-08-01:
I'm compiling example design for IBERT that uses both clock and 4 GTX quads.
Unfortunately I cannot easily load my design to configured HMC7044 because once I load AMC FPGA, it unconfigures RTM FPGA which disables HMC chips and clock muxes... So now I'm using RTM loopback tester that has 2 crystal generators at each of MGT clock inputs.

hartytp, 2018-08-01:
@gkasprow that's interesting. So, I guess that are two ways this could happen:
1. There is some obscure setting for the transceivers that differs between ARTIQ and your test setup
2. There is some hardware bug that makes your test setup differ from the artiq use case (I'm thinking of something like the PI issues that made SDRAM stop working in ARTIQ)

I guess we can double check (1) with Xilinx, although they've already indicated that they didn't think any setting could do this IIRC.

If that's the case, I think this can only be some hardware bug like a PI issue or a missing special function pin

sbourdeauducq, 2018-08-01:
> Unfortunately I cannot easily load my design to configured HMC7044 because once I load AMC FPGA, it unconfigures RTM FPGA which disables HMC chips and clock muxes...

You can just comment out the relevant portions of the firmware, e.g.
https://github.com/m-labs/artiq/blob/master/artiq/firmware/runtime/main.rs#L85-L88
https://github.com/m-labs/artiq/blob/master/artiq/firmware/runtime/main.rs#L108-L110

sbourdeauducq, 2018-08-01:
@gkasprow Or if you're just doing electrical tests, you can prevent the whole firmware from starting by erasing it from the flash.

hartytp, 2018-08-01:
> Unfortunately I cannot easily load my design to configured HMC7044 because once I load AMC FPGA, it > unconfigures RTM FPGA which disables HMC chips and clock muxes... So now I'm using RTM loopback > tester that has 2 crystal generators at each of MGT clock inputs.

@gkasprow what do you want to find out from this measurement? If the question is whether the HMC7043 is the cause of the problems with ARTIQ, then wouldn't it be easiest to connect ARTIQ back up, but to remove the capacitors that connect the HMC7043 to the AMC.

gkasprow, 2018-08-01:
@hartytp I want to makes sure both CLK inputs behave the same with Xilinx design.

gkasprow, 2018-08-01:
@sbourdeauducq So if ARTIQ does not use GTP CLK1, so it means that it is not initialised and its behaviour would be different than CLK2!

gkasprow, 2018-08-01:
but with Xilinx IP, the clock input that is not initialised has DC component close to 0.

sbourdeauducq, 2018-08-01:
> So if ARTIQ does not use GTP CLK1, so it means that it is not initialised and its behaviour would be different than CLK2!

ARTIQ initializes both CLK1 and CLK2 at the HMC7043 in all cases.
The standalone design uses CLK2 only. The masterdac and satellite targets use both CLK1 and CLK2.

gkasprow, 2018-08-01:
with Xilinx IP core and disabled HMC chip, I measure 105Ohm on both GTX clock inputs.

gkasprow, 2018-08-01:
And I measure 0.75V of DC components on both clock inputs

gkasprow, 2018-08-01:
and I get 800mV RMS on both clock inputs

gkasprow, 2018-08-01:
The voltage is different because I use external 2x LVDS oscillators.

gkasprow, 2018-08-01:
@sbourdeauducq I shorted via jumper the INIT_B line of RTM FPGA to 3.3V. But still the RTM FPGA resets the HMC chip after I reload the AMC FPGA... Does it enter reset state when communication between FPGAs breaks?

gkasprow, 2018-08-01:
OK, I shorted the HMC chip reset to GND. So it does not get reset by RTM FPGA. It started generating ~600MHz signal on both outputs.  I loaded the Xilinx IP core. And what I got:
- on both clock inputs I get 0.75V DC
- on both clock inputs I get the same amplitude - ~800mV RMS.
- I measure 85..89Ohm on both clock inputs.

gkasprow, 2018-08-01:
With HMC in reset state, I measure 105Ohm on both inputs

hartytp, 2018-08-01:
So, what's the plan here? Do we wait for Xilinx to confirm whether there are any FPGA settings that can alter the common-mode voltage on the transceiver clock inputs? Or, is there any more data we want.

e.g. would it be useful for me to look at the common-mode voltage on the FPGA clock inputs using ARTIQ, but with the HMC7043 held in reset?

gkasprow, 2018-08-01:
You can look at them before HMC gets initialised. AFAIR I did it already and did not observe any change to the levels but you can do it as well. The easiest place to observe is the RTM connector - the 2 signal pins closest to the board edge, 6-th and 10-th row counting from the alignment pin.
For me it looks like incorrectly initialised transceiver. Where can I have a look at the verilog code generated by ARTIQ?

gkasprow, 2018-08-01:
[image]

hartytp, 2018-08-01:
@gkasprow if you've already done that test then I don't think there is much point in me repeating it. But, let me know if there is anything you would like me to do. Otherwise, let's wait and see what Xilinx think.

I guess that our other option is to try to copy the IBERT transceiver settings into ARTIQ.

hartytp, 2018-08-01:
@gkasprow top.v and top.xdc are here, let me know if you need anything else... https://drive.google.com/drive/folders/19m6mh1qXCTfPNeLMrFk4yKGWKOW8p2F0

gkasprow, 2018-08-01:
Funny thing : ULTRASCALE_IS_AWFUL :)
There is only one clock instantiated, connected to pins P6,  P5 (GTP CLK2)

```
 ## dac_refclk:1.p
set_property LOC P6 [get_ports dac_refclk_p]
 ## dac_refclk:1.n
set_property LOC P5 [get_ports dac_refclk_n]
```

And I observe proper bias and the clock shape on these pins.
No trace of GTP CLK1 (pins V5, V6) so no surprise that I don't see valid DC level on that input.

sbourdeauducq, 2018-08-02:
> For me it looks like incorrectly initialised transceiver. Where can I have a look at the verilog code generated by ARTIQ?

The clock itself is received by a IBUFDS_GTE3, which is a pretty simple component without many settings...

hartytp, 2018-08-02:
Greg,

Can I check I've understood the situation:

- you loaded the IBERT reference design you posted onto the FPGA and measured the DC resistance to ground of the GTP CLK1/CLK2 inputs as well as the common-mode voltage at those inputs.
- the impedance measured between the P/N terminals of each input was approx 100Ohms when input clock was applied, as expected
- the common-mode voltage for the P/N terminals of each input was approx 0.8V as expected (1V*4/5)
- applying a 800mV RMS signal to the clock input shifts the measured DC resistance to 85Ohm (measurement error due to RF rectification in the DMM?) but not the common-mode voltage (which is measured with a high-Z active scope probe)
- so, with the IBERT design everything seems to work as expected indicating that there is no damage to the hardware

with ARTIQ:

- both CLK1 and CLK2 are driven from LVPECL outputs of a HMC7043, so we expect 1.6Vpp differential signal into a 100Ohm load (well, a bit less due to loss in traces etc)
- you measured the common-mode voltages and input resistances CLK1 and CLK2
- the measured common-mode voltages for the CLK1/CLK2 inputs are 0.75V for CLK1 and 0.4V for CLK2. the measured resistances are also lower than in the previous setup
- you verified that the 1V supply is indeed at 1V in this design
- you tried disconnecting the RF signal from the FPGA (by depopulating the AC coupling capacitors), but this did not change the measured common-mode voltage on the FPGA inputs

Does that correctly and fully summarize the measurements you've made so far?

hartytp, 2018-08-02:
edited previous post

hartytp, 2018-08-02:
@sbourdeauducq I'm not sure how up to speed Greg is with the ARTIQ/migen tool-chain. Until @marmeladapk gets back it might be more efficient for us to give him binaries to test than to get him building code.

gkasprow, 2018-08-02:
Did you see my [commen](https://github.com/sinara-hw/sinara/issues/572#issuecomment-409728754)t ?

gkasprow, 2018-08-02:
For me it looks like the compiler synthesized away the input buffer that is not used.

hartytp, 2018-08-02:
@gkasprow 

> And I observe proper bias and the clock shape on these pins.
> No trace of GTP CLK1 (pins V5, V6) so no surprise that I don't see valid DC level on that input.

I'm not sure I follow you.

We are using GTP CLK2 in ARTIQ for the JESD. I thought that when you made measurements earlier on, that it was CLK2 that had problems. Didn't you say that when you measured it, you found that CLK2 had large reflections and the wrong common-mode voltage? So, how is that explained by CLK1 not being used in the HDL?

I do agree that if we're not using CLK1 in the HDL then we should probably disable that HMC7043 output in firmware, but do you think that could be causing our problems?

gkasprow, 2018-08-02:
It looks like there was some misunderstanding. @sbourdeauducq wrote that both inputs are used so I tried to trace it down why they behave differently. And I focused on this aspect. The signals look OK on both clock inputs - only amplitude differs. CLK2[ has](https://github.com/m-labs/artiq/issues/1080#issuecomment-408066703) right DC component, CLK1 don't. This can be easily explained by the fact that CLK1 receiver is not instantiated at all. This also explains why I got much lower signal on CLK2 - and this was OK. The signal on CLK1 was higher because was not terminated. Source termination was enough to damp reflections. 
So it looks like I lost almost two days trying to figure out the issue that did not exist.

sbourdeauducq, 2018-08-03:
@gkasprow In your previous messages, you said that **CLK2** has impedance problems. That one is always used by all bitstreams. (https://github.com/m-labs/artiq/issues/1080#issuecomment-408094486)

The impedances also don't seem right for either input: https://github.com/m-labs/artiq/issues/1080#issuecomment-408097352

Or are those measurement results wrong?

gkasprow, 2018-08-03:
I wrote that CLK2 has problems because it had much lower amplitude. But still enough to drive the clock input. Measurement of input impedance by Ohmmeter with RF applied is not representative. I repeated measurements without RF and they were different. It seems that RF gets rectified on internal circuits and caused measurement errors. XOs driving LVTTL signal is different case than diff LVPECL driving 1V CML input where 2V pp appears without termination and can cause some effects.

sbourdeauducq, 2018-08-03:
> diff LVPECL driving 1V CML input where 2V pp appears without termination and can cause some effects.
@hartytp Please let me know your results with https://github.com/m-labs/artiq/commit/e83ee3a07a3aec91e295ac67d15157c308c412a4.

sbourdeauducq, 2018-08-03:
Hmm, there is still the Si5324 output into the unused transceiver clock input pins though. Maybe instantiate a dummy IBUFDS_GTE3 there. Does Vivado remove IBUFDS_GTE3s with all outputs unconnected, or otherwise do something wrong in that case?

sbourdeauducq, 2018-08-03:
You can also disable the Si5324 in the standalone target if you're not using it to replace a RF generator for the RTM.

hartytp, 2018-08-07:
AFAICT, there is no real issue here.

closed 2018-08-07 by hartytp

hartytp, 2018-08-08:
@gkasprow I see the same DC level on both halves of CLK2 (about 700mV) but one has 1.2Vpp and one has 600mVpp. Still enough signal, but maybe suggests that the termination is damaged on this board?

hartytp, 2018-08-08:
Triggering from my 100MHz reference, looking at CLK2 on the back of the RTM card (at the AMC<->RTM connector pins). I don't have a differential scope probe and am currently grounding a 500MHz probe from the pin that guides the RTM into the AMC, so bad grounding! I can try to improve this by finding a grounded pin on the AMC<->RTM connector, but I'm not sure that's necessary.

@gkasprow do you want me to look at anything else, or just send the boards back to you for investigation?
[image]
[image]

hartytp, 2018-08-08:
Scale is 200mV/div (bottom trace is 1.1Vpp)

gkasprow, 2018-08-08:
It looks like broken pull-down resistor at the HMC output.
Please measure DC at the HMC output directly, before the caps.
I measured signals with 1GHz probe, in your case it looks like poor probe grounding, but this should not avvect the amplitude. Does the issue persist when you swap the probes ?:)

hartytp, 2018-08-08:
> I measured signals with 1GHz probe, in your case it looks like poor probe grounding, but this should not avvect the amplitude. Does the issue persist when you swap the probes ?:)

Yes, the grounding is bad. But, I use the same probe and grounding to look at both halves of CLK2, so the difference between them is unlikely to be due to my scope.

gkasprow, 2018-08-08:
The coupling capacitors are below the HMC chip
[image]

hartytp, 2018-08-08:
Measuring on the bias resistors of HMC7043 pins 21 and 20. Both pins have 2V DC. Pin 21 has 1Vpp, pin 20 has 1.6Vpp

hartytp, 2018-08-08:
I can post photos if that's helpful

gkasprow, 2018-08-08:
OK, this is indeed strange and can be responsible for PRBS issue. Post it to me, I will investigate and check with other boards.

hartytp, 2018-08-08:
Thanks Greg. I'll post it now.

gkasprow, 2018-11-20:
For future reference - it was broken cap.
[image]