sinara-hw/sinara #573: HMC830 does not lock when driven by AMC clock output

gkasprow, 2018-08-07:
The clock output from AMC looks OK:
[image]

gkasprow, 2018-08-07:
But when I connect it to the RTM the signal disappears....
This is a reason why - the input SMA has shorted pin to GND.
When I connect it with AMC, the HMC locks and PRBS errors disappear...

sbourdeauducq, 2018-08-07:
I don't understand that. Where is the short circuit and how do you make it disappear?
What does this have to do with PRBS?

gkasprow, 2018-08-07:
it is at the RTM SMA clock input.
Since it was shorted, the clock signal entering the RTM board had very low amplitude. And that's why in your conditions HMC830 could not lock. In my lab it locks probably because I have very strong 100MHz signal source outside (FM broadcast station a few hundreds meters from me) so the RTM could pickup the signal and together with leaking generator signal the HMC managed to lock to it :)
But the quality of recovered signal was not enough for transcivers

hartytp, 2018-08-07:
hmmm...on my RTM the clock SMA input is a high impedance between the signal pin and ground. Moreover, if you look at the traces I've posted, the HMC830 clock output is very high quality (at least, as far as I can tell from a SA). So, I don't think this was a problem on my board.

hartytp, 2018-08-07:
Also, on my board, CLK1 works reliably and CLK2 doesn't.

gkasprow, 2018-08-07:
So it looks like there is sth else on your board.
If you want, I can post you my set of boards and you will post me your one.

hartytp, 2018-08-07:
@gkasprow after fixing that, does the PRBS work on all of your AMC/RTM combinations? If so, that's great!

gkasprow, 2018-08-07:
I tested with 5 RTM boards and clocking from AMC board works fine.

gkasprow, 2018-08-07:
I have 5 sets of boards so there is quite a lot of combinations to tests :)

hartytp, 2018-08-07:
> I have 5 sets of boards so there is quite a lot of combinations to tests :)

Sure. I'd pick 1 RTM and check it with all AMCs. Then pick one AMC and check with all RTMs. If that looks good, it suggests that the issue only lies with particular boards. If that's the case, the best thing is for me to post you both my AMC and RTM and you can look at this particular pair.

gkasprow, 2018-08-07:
I made 5 sets of random boards and will test them together.

gkasprow, 2018-08-07:
OK, i'm testingt all 25 combinations to be sure.

gkasprow, 2018-08-07:
one combination causes HMC830 lock failure...

sbourdeauducq, 2018-08-07:
> OK, i'm testing all 25 combinations to be sure.
> one combination causes HMC830 lock failure...

I'm glad you are finally starting to get into the Sayma mess...

gkasprow, 2018-08-07:
this AMC is missing rework at the clock buffer IC...

hartytp, 2018-08-07:
> this AMC is missing rework at the clock buffer IC...

What rework is that?

sbourdeauducq, 2018-08-07:
> it is at the RTM SMA clock input.
> Since it was shorted,

Where was that short?

gkasprow, 2018-08-07:
rework around IC40. Pins 7 and 12 must be grounded. IC need to be desoldered, traces to pins 7 and 12 cut and with piece of wire connected to nearby capacitor.

gkasprow, 2018-08-07:
@sbourdeauducq the short was just at the board edge.

sbourdeauducq, 2018-08-07:
So it's a hard short, i.e. something I can check easily with just a DC ohmmeter?

gkasprow, 2018-08-07:
yes

hartytp, 2018-08-07:
> rework around IC40. Pins 7 and 12 must be grounded. IC need to be desoldered, traces to pins 7 and 12 cut and with piece of wire connected to nearby capacitor.

I don't see that on my board. Do you have a photo?

sbourdeauducq, 2018-08-07:
> yes

OK. One of the last RTM you sent me stopped locking after I fiddled with it, so that may be the problem...

gkasprow, 2018-08-07:
it's quite possible that the SMA lost contact with trace

hartytp, 2018-08-07:
Yes, but you can see that clearly with a scope...

sbourdeauducq, 2018-08-07:
The constant stream of problems that Sayma has with just about everything makes my mind boggle. I haven't seen any shorted or broken SMA on any of the DIO and Urukul cards that I've tested. That's almost 200 SMAs and they were all good. What is going on with Sayma?

hartytp, 2018-08-07:
is's a more delicate, more complicated design that has been heavily reworked. Once we stop poking it and put it in a uTCA rack where it belongs, we won't see these kinds of issues.

gkasprow, 2018-08-07:
It was only one shorted connector so far. It was shorted during assembly.
 Sayma is far more complex than any other 3U board. All the EEM boards together have such complexity as Sayma. So if you compare all issues related with EEMs with all Sayma issues the results will be similar :)

hartytp, 2018-08-07:
Also, @sbourdeauducq while people are working hard to fix problems, let's channel our energy towards focused comments/bug hunting rather than general moaning.

gkasprow, 2018-08-07:
OK, so I tested all 25 combinations. All work fine with AMC -> RTM clocking..

gkasprow, 2018-08-07:
The only difference is number of STPL errors that on some boards combinations appear...

hartytp, 2018-08-07:
@gkasprow That's great. So, all boards now lock and pass PRBS tests now apart from my board?

Can you post a picture of the IC40 rework that was missing on one board? If it's missing on mine, I'll do it myself. Otherwise, I'll post the board back to you for investigation.

Also, can you remind me which version of ARTIQ you're testing (maybe post a UART output with the version information)? Make sure you're not testing with the latest ARTIQ as @sbourdeauducq pushed a commit that changed the clock configuration to avoid using the problematic CLK2 input.

gkasprow, 2018-08-07:
I'm testing 
[     0.003883s]  INFO(runtime): software ident 4.0.dev+1286.g015c592a;standalone
[     0.011017s]  INFO(runtime): gateware ident 4.0.dev+1286.g015c592a;standalone

gkasprow, 2018-08-07:
@hartytp 
[image]

hartytp, 2018-08-07:
Okay, that's really quite old, but I think it should be fine. @sbourdeauducq are you happy using this for testing? NB this is before we disabled the unused LVPECL HMC7043 outputs, so it's nice to see that those don't actually cause an issue.

gkasprow, 2018-08-07:
IC40 is locted above FPGA, between FPGA and FMC

hartytp, 2018-08-07:
Oh, you mean on the AMC. Okay, mine does have that rework.

Okay, well let me send you my AMC + RTM back for investigation. I'll leave a version of ARTIQ flashed on them that exhibits this bug.

closed 2018-08-07 by hartytp