sinara-hw/sinara #577: Sayma: connect AMC GTH lane to RTM GTP

sbourdeauducq, 2018-08-11:
One of the AMC backplane GTH lanes can be repurposed - we need just one to talk to Metino - and connected to a GTP lane over the RTM connector. This way we can use the exact same DRTIO as with Sayma-Kasli, and avoid developing a PHY that would be particularly complicated due to how messed-up the Ultrascale IOSERDES is.

sbourdeauducq, 2018-08-11:
If someone likes having the option of multiple AMC backplane lanes, we can support rerouting this lane using capacitors that can be soldered or not.

hartytp, 2018-08-11:
@sbourdeauducq does this suggestion imply that the RTM DRTIO would only be controllable from Metlino, and not from the AMC FPGA? If so, doesn't that kill off use-cases where Sayma is used without Metlino (e.g. an AMC + RTM in a small crate connected to Kasli via fibre)?

sbourdeauducq, 2018-08-11:
No. I meant disconnecting one Ultrascale link from the AMC backplane to free one GTH, and connecting that GTH to the RTM.
The initial plan was to use the IOSERDES on the Ultrascale side but this has two avoidable issues:
1. complications from the different data rates
2. dealing with poorly designed Ultrascale I/Os

The GTH route can be prototyped on the current hardware by using a SATA cable.

hartytp, 2018-08-11:
Okay, that does sound like a good idea. Can we just repurpose one of the ADC JESD links already on the RTM connector?

sbourdeauducq, 2018-08-11:
No.
1. They are RX only.
2. Due to the poor design of Xilinx transceivers, combining different applications in a paired transmitter/receiver is a mess.

sbourdeauducq, 2018-08-11:
> But, I do want to be careful about leaving some lines free on the AMC<->RTM connector for AFE LVDS ADCs.

There are already reserved links for the GTP, which currently go to the Ultrascale IOSERDES. This issue is about making them go the GTH instead.

hartytp, 2018-08-11:
Sorry, yes, fine. well, then I have no objections :) Good plan!

gkasprow, 2018-08-11:
We can make assembly variant and connect GT to RTM by default. Let's use RTM_FPGA_GTP_Rx/Tx0

gkasprow, 2018-08-11:
@sbourdeauducq what clock do you want to use for this link? Both for AMC and RTM FPGA.

sbourdeauducq, 2018-08-12:
RTM: Si5324
AMC: Same clock as the other DRTIO links (SFP and AMC backplane).

dhslichter, 2018-08-14:
> No. I meant disconnecting one Ultrascale link from the AMC backplane to free one GTH, and connecting that GTH to the RTM.

FWIW sounds fine to me.

jbqubit, 2018-08-15:
>One of the AMC backplane GTH lanes can be repurposed - we need just one to talk to Metino

Fine. 

> If someone likes having the option of multiple AMC backplane lanes, we can support rerouting this lane using capacitors that can be soldered or not.

Sounds like a good plan.

marmeladapk, 2018-12-13:
@sbourdeauducq @gkasprow Is this meant to be implemented in Sayma v2.0?

gkasprow, 2018-12-13:
Yes, I implemented it already.

jbqubit, 2018-12-13:
Leave open. Whoever is assigned design review for this item will respond to the Issue. @gkasprow will close the Issue once he confirms that design review changes are implemented.

gkasprow, 2018-12-13:
I already implemented it in schematics. Also, I found and fixed wrong polarity issue.

marmeladapk, 2019-01-25:
@gkasprow Can this be closed then?