sinara-hw/sinara #579: Sayma: move one GT_CLKx to the DRTIO quad

sbourdeauducq, 2018-08-11:
A single GT_CLKx can already clock all the JESD quads without issue, and JESD is a bog-standard transceiver application that is insensitive to skew on that clock.
When clocking from the RTM, the DRTIO quad can currently be only clocked from GTP_CLK1, and we have to cross one quad using extra clocking resources inside the Ultrascale, which add skew that may or may not be predictable.
It can be good idea to move one of those GT_CLKx to the DRTIO quad, so we can see if using it improves synchronization.

gkasprow, 2018-08-11:
Can we swap CDR_CLK_CLEAN with GT_CLK2_IN ?

hartytp, 2018-08-11:
@sbourdeauducq do we need to keep GT_CLK2, or can we scrap it? If we do need it, does it ever need to be different to CLK1? If they are always the same, can we add an LVDS fanout buffer on the AMC to allow us to save an AMC<->RTM line?

sbourdeauducq, 2018-08-12:
> Can we swap CDR_CLK_CLEAN with GT_CLK2_IN ?

Currently, the master is using CDR_CLK_CLEAN (in a non-critical manner), other designs don't (all the Si5324 does is drive the HMC830). I think it should work in the other bank, and we can do this swap. But why swap it instead of moving GT_CLK2 to the other clock input of the DRTIO quad? Are there PCB routing issues?

> @sbourdeauducq do we need to keep GT_CLK2, or can we scrap it?

As demonstrated by the current design, it works with a single clock. But we do need two clocks if we want to try clocking DRTIO from its local quad; the DRTIO quad clock input cannot reach all JESD quads.

> If we do need it, does it ever need to be different to CLK1? If they are always the same, can we add an LVDS fanout buffer on the AMC to allow us to save an AMC<->RTM line?

They are always the same, and we can use such a buffer.

Note that I do not know if local clocking of the quad will actually improve synchronization by a significant amount - but if it's easy to do this experiment, we should try.

hartytp, 2018-08-12:
@sbourdeauducq okay, fine. Well, let's keep this in mind for Sayma v2.0. If we don't need to free up any more AMC<->RTM lines, we can leave it as is. Otherwise, let's add a LVDS fanout.

sbourdeauducq, 2018-08-13:
> Can we swap CDR_CLK_CLEAN with GT_CLK2_IN ?

Due to this https://github.com/sinara-hw/sinara/issues/580#issuecomment-412456090 let's keep the Si5324 output in the DRTIO quad.

sbourdeauducq, 2018-08-17:
Not so relevant after the changes in https://github.com/m-labs/artiq/commit/167e97efd20313a9cfad12bcf3ae2eec45f319fb
If it is very straightforward to do the clock move, then ok, otherwise let's not bother.

marmeladapk, 2018-12-13:
@sbourdeauducq @gkasprow is this still applicable, or can we close it? Or postpone it to later version?

gkasprow, 2018-12-13:
It depends if we want to keep compatibility of v1 and v2 of Sayma AMC and RTM.

hartytp, 2018-12-13:
I would argue strongly that we should **not** prioritize maintaining backward compatibility. The v1.0 boards are quite fragile and mainly don't work anyway and, on top of that, there will not by any ARTIQ support for the v1.0 boards going forwards. 

If maintaining backwards compatibility is "free" then go for it, otherwise prioritize optimal performance of the new hardware over backwards compatibility.

marmeladapk, 2018-12-13:
I'd say, that there's no point in maintaining backward compatibility. Each AMC has a corresponding RTM, there are 10 of them and no more will be produced. 

Should I add this to Sayma v2.0 milestone then?

jbqubit, 2018-12-13:
No need for backward compatibility. There will be no Sayma v1 support after Sayma v2 goes into production.

closed 2019-03-11 by gkasprow