sinara-hw/sinara #580: Sayma v2.0 synchronisation

hartytp, 2018-08-13:
AFAICT, the current synchronisation mechanism for Sayma isn't capable of doing better than 25ps (3.6deg for 400MHz RF). This isn't good enough for most applications, so we need to do something different in v2.0.

hartytp, 2018-08-13:
> I don't see why the 7044 would be any better than the 7043. On the contrary, I suspect it's going to bring a whole set of new bugs.

I actually don't buy that. IME, these chips tend to work really well if you use them in the one way that the designers anticipated and tested. If you do anything else, they get quite buggy. Look at all the app notes to do with synchronising JESD systems, they all use the HMC7044 with a pulse that's sampled from the low-frequency clock. That's what I think we should do here.

> If the problematic drift comes from the Si5324 XO, what about using a better clock source there?

Might do. You'd have to test carefully. A DDMTD/WR PLL is probably a better option however if you want to go down that route.

We also have to ensure that the FW doesn't adjust any analog delays on each boot, as that's obviously death.

sbourdeauducq, 2018-08-13:
> I actually don't buy that.

Fine. You'll be the one debugging the HMC7044 after all :)

hartytp, 2018-08-13:
Sure.

Note that if we had a working 7043 setup I'd feel quite differently. But, we don't, there are still too many inexplicable bugs. I'd rather just give up on that and use the hard-coded mechanism in the HMC7044.

But, let's not be hasty. What's the best "hardware test" on Sayma that we *can* do? How about we use one of the headers to hook the 7044 up to the AMC FPGA. Then I alter the ARTIQ firmware/gateware to initialise it and send it a sync pulse. I'll then check on a scope that it produces RF outputs that are synchronised.

hartytp, 2018-08-13:
I'd also populate the LF and check that I can do the synchronisation with 100MHz in, 2.4GHz out.

sbourdeauducq, 2018-08-13:
> Sure.

Good.

So, the plan with DRTIO is to feed the raw recovered transceiver clock (with jitter) straight into the 7044, and then it'll have deterministic and per-output controllable input-to-output skew?
(not having fine skew control on DCLK channels is acceptable)

If so, that sounds good to me, and fits well within the current synchronization code.

sbourdeauducq, 2018-08-13:
By the way, until the link is initialized, the recovered clock may glitch. I don't know if the 7044 will handle that well (that clock input won't be selected by the firmware at this time, but there might be HMC bugs).

hartytp, 2018-08-13:
> So, the plan with DRTIO is to feed the raw recovered transceiver clock (with jitter) straight into the 7044, and then it'll have deterministic and per-output controllable input-to-output skew?
(not having fine skew control on DCLK channels is acceptable)

My thinking was that for now we would *only* support clocking the HMC7044 from a low-noise external PLL. e.g. use a power splitter on the 100MHz clock source that you're feeding the AMC with.

Then, the AMC uses the current Si-phaser, and the RTM uses the external clock. I think that until/unless the WR PLL lands, that's the best we can do.

hartytp, 2018-08-13:
Note that this means that synchronisation between Sayma cards will be exact (limited by to be measured drifts in the clocking network). But, synchronisation between Sayma and TTL events will be limited by Si-phaser issues. That's fine IMHO. e.g. time critical switching of Sayma cards only involves the DACs, not the TTLs, as it will be done by setting the SAWG amplitude to 0.

hartytp, 2018-08-13:
of course, one could also use the 2-loop PLL on the HMC7044 to implement a jitter cleaner. But, low drift, low noise, low bandwidth PLLs are somewhat tricky. We've put quite a bit of work into developing the DDMTD/DCXO system to do just that, so I'd prefer to use that system than to start again on the HMC7044.

sbourdeauducq, 2018-08-13:
> Then, the AMC uses the current Si-phaser, and the RTM uses the external clock.

OK, we can also do that with the current code. The SYSREF/RTIO align will take care of HMC7043 divider non-determinism.
All that needs to be changed (off the top of my head) is reconnect the DRTIO transceiver clock input and the siphaser input to the Si5324. And fix https://github.com/m-labs/artiq/issues/1129. The setup/hold calibration code should still work when clocking the boards that way. Maybe enlarge the JESD elastic buffers if needed, since the JESD clock and RTIO clocks will have more drift.

sbourdeauducq, 2018-08-13:
Actually, handling the JESD elastic buffers requires (only) a reset after the DRTIO link is established.

hartytp, 2018-08-13:
@sbourdeauducq ack. But, there are a lot of people who use the HMC7044 in the way I've described (generate a 100MHz "trigger" pulse from the FPGA) so we can get help. I'm not aware of anyone ever having done what we're doing with the HMC7043 or similar.

sbourdeauducq, 2018-08-13:
All that we want from the 7043 is dividers, synchronous reset of the dividers, analog delays, and slips. The firmware assumes nothing else from it, it can even glitch as much as it wants while being configured. The 7043 can't even do those basic things right. How can you expect the 7044 to work any better? How do you align SYSREF with the RTIO timestamp counter if the digital slip is also broken?

sbourdeauducq, 2018-08-13:
And the bug might not even be in the 7043. It's just the prime suspect; but we're not sure until we look at SYSREF on a scope...

hartytp, 2018-08-13:
> How can you expect the 7044 to work any better? How do you align SYSREF with the RTIO timestamp counter if the digital slip is also broken?

It's been a while since I looked at this carefully, so I'd have to look again to be sure, but...

The big difference is that we're doing all the phase shift control manually through the register interface, which is quite buggy. The other way of doing things is to send it a single trigger pulse and ask it to synchronise itself to that. Then let it take care of all the delays itself without going through the buggy register interface. That can be done with the HMC7043 as well, but then you need to meet S/H on 2.4GHz, which is nastier.

hartytp, 2018-08-13:
> And the bug might not even be in the 7043. It's just the prime suspect; but we're not sure until we look at SYSREF on a scope...

@marmeladapk / @gkasprow (or send me my board back and I'll do it).

closed 2018-10-02 by hartytp

jbqubit, 2018-11-28:
I inquired with AD a while back about AD9154 temperature response. Here's the response from an AD engineer. 
 
>I do not have a good answer here frankly. To quantify temperature drift, typically more than one part needs to be tested. We have little to no capacity to set this up at the moment.  

>Two AD9154’s (or any other identical generics) would drift in the same direction, and the rate would be similar as well, assuming the same production lot. So the relative phase will likely stay the same, especially with only 1*C increase. The really interesting aspect would be if there is a temp gradient between two devices from two different lots (monte carlo of sorts).  
 
>Temperature is inherently non-linear. So the slope due to a 1*C change will be different depending on the starting temp. Likely more so on the hot end. 
 
>All in all, you would need to collect considerably more data than on a single part.  Going forward, we will have more temperature data on this class of parts.