sinara-hw/sinara #581: Swap DACB _P with _N (or change in AFE)

marmeladapk, 2018-08-13:
It leads to 180 deg phase difference on channels next to each other.

marmeladapk, 2018-12-13:
@gkasprow Should this be implemented on AFE or on RTM?

gkasprow, 2018-12-13:
I wouldn't change AFE since it already exists.

dhslichter, 2018-12-19:
@gkasprow aren't we going to redo the connectors for AFE mezzanine and make new ones without power detectors, etc?  This seems to me like an important bug to fix on the RTM itself, because it would be very subtle and potentially frustrating for people who are trying to make in-phase signals (granted I know that people will be calibrating out cable lengths too, so it's not the end of the world).

gkasprow, 2018-12-19:
I mean that I don't want to change existing FMC-like TestMod .

marmeladapk, 2019-01-22:
This is fixed. I checked chain from DAC to output stage on AFE and lines are not swapped.

closed 2019-01-22 by marmeladapk